TG9541/stm8ef #10: BG Woes

hexagon5un, 2017-01-11:
Having lots of fun with the esp-link + STM8 combo in the ESP-14 chip.  Thanks again.

Just built a two-channel energy monitor device that's going to keep tabs on our washer and dryer.  Was playing around with BG'ing things, and it would hang, sometimes with the BG task running, mostly not.

This works:

    : blink tim 128 and if pa3.on else pa3.off then ;
    ' blink bg !

This doesn't:

    : delay 0 begin  1+ dup  18000 < while repeat drop ;
    : blink pa3.on delay pa3.off delay ;
    ' blink bg !

I tried the same thing with a `do` loop for the delay, and that fails too. I wasn't too surprised, given the return stack tricks that go on with `do`, so I tried this all-parameter-stack code above.  It fails too.

(It's not crucial -- the code will work fine with a pollling main loop too.  I just thought it would be fun to test out...)

TG9541, 2017-01-16:
Sorry for the delay, should teach me to keep an eye on issues, and not use them as a backlog ;-)

The second example won't work because the delay takes longer than the period of time between two background task interrupts: on a 16MHz HSI test board your delay routine runs for 250ms when compiled to RAM, and the background task interval is currently 5ms. 

The assumption is that a background task does its thing in an input-process-output manner in anything up to 4ms (e.g. for closed loop control), and that it's is either stateless and or that it modifies the state from run to run. In this case, the time spent waiting should be state data (e.g. a state machine and a counter), or it should be stateless (using the value of the background ticker).

For the next release I plan to make the background task interval user-definable. An interval of 250ms would require setting the prescaler manually, though.

hexagon5un, 2017-01-19:
Now of course I see that it says all that, and more, on the relevant page in the instructions wiki.  That's what I get for playing first and reading second (if ever!) 

It's all clear to me now. Thanks again.

BTW: building up a real application, I keep running into the memory caps, especially that for RAM.  My old development cycle in Mecrisp on the bigger STM32 chips is usually to hammer at some functions in RAM until they work, then burn them into flash and reset the RAM again.  When I try to do this with any complex functions on these tiny chips, I run out of RAM with horrible/crashy consequences.  

I'm winning in the end, still.  But it's a lot less free and fun than developing on a chip that's got 6K-32K of RAM just sitting there...

Still enjoying, though.  I'll post up when I'm done.  Spoiler: ESP14.

TG9541, 2017-01-19:
I agree, the STM8S003F3 is really short of resources (but then again, constraints drive creativity). I guess that you'll have to move words from RAM to NVM much more frequently, and frankly, the code will have to be much smaller too. I have a little space safer in the pipeline ( http://goo.gl/RznooG ). 

You're working on the ESP14? Nice! Did you try the dual UART feature in the latest release?

TG9541, 2017-01-25:
Cleaning up

commit 0ef6cadd (2017-01-25): TIM2 rate configurable, fixes #10
closed 2017-01-25 by TG9541