TG9541/stm8ef #12: Bug in countdown DO +LOOP

hexagon5un, 2017-01-24:
: foo 1 4 do i . -1 +loop ;

should print `4 3 2 1`.  Instead, it just prints `4`.

TG9541, 2017-01-24:
Before implementing +LOOP I couldn't find any clear requirements. I then compared different implementations and found one the addressed the risk of very long loops due to erroneous upper limit and starting point. That's the one I implemented, since it doesn't require looking at the sign of the increment.
I now understand this is a valid use case, and I'll change the code :-)

TG9541, 2017-01-24:
How should the following code behave?
`: foo 1 4 do i . -2 +loop ;`

hexagon5un, 2017-01-24:
gforth and mecrisp-stellaris say: `4 2`.

My understanding is that the TOS index is increasing the test is run against `<`, and when it is decreasing, it's run against `<=`.  (https://www.forth.com/starting-forth/6-forth-do-loops/)  

(Equivalently, I think it's `begin >= until` and `begin > until` loops under the hood.)

This is crappy, b/c `4 1 do i . 1 +loop` -> `1 2 3`  (doesn't `4` b/c `4<4` is false) and `1 4 do i . -1 +loop` -> `4 3 2 1` (`1` b/c `1 <= 1` is true), so the loops have different lenghts, but that's the way Forth is defined.

https://github.com/gerryjackson/forth2012-test-suite is written for ANS Forth.  This is in core.fs:

    T{ : GD2 DO I -1 +LOOP ; -> }T
    T{ 1 4 GD2 -> 4 3 2 1 }T
    T{ -1 2 GD2 -> 2 1 0 -1 }T

TG9541, 2017-01-24:
Thanks! Good links, too!

I fixed it. If you need binaries right away, I can update the 2.2.6.snapshot.

closed 2017-01-24 by TG9541

hexagon5un, 2017-01-25:
Binaries not necessary, and I can confirm that at least the test cases above are working as they should.

commit cadd43f2 (2017-01-24): Resolves #12: Bug in countdown DO +LOOP