TG9541/stm8ef #13: DNEGATE doesn't in v2.2

RigTig, 2017-02-12:
\ I am having great fun using eForth
\ thank you for a great piece of work
\ In v2.2
3 0 dnegate .s \ results in
FFFD 0 <sp  ok
\ when it should be FFFD FFFF <sp ok

\ regards, RigTig

TG9541, 2017-02-12:
Hi RigTig,
I can conform he problem. It has nothing to do with my meddling with Dr. C.H. Tings code: it's already present in STM8EF V2.1. So far I didn't use double arithmetics much, and it's a good thing that you're testing it.

TG9541, 2017-02-12:
Hi RigTig,
it's fixed in the `develop` branch. 
In case it's needed in the `master` branch before release of STM8EF V2.2.7 please use the following definition as a work-around:

```
: DNEGATE NOT >R NOT 1 UM+ R> + ;
```
Thanks again for reporting this issue :-)

Regards & Greetings from Germany, 
Thomas

RigTig, 2017-02-14:
Thomas,
 Many thanks for fix. All I'd done was figure the high byte was too high by
1, so my temporary fix was just
: DNEGATE DNEGATE 1- ;

anyway, just so you have an integer square root when you need it, here is
one algorithm coded in eForth

\ some maths utility words
\ eForth + is u+
: u+ ( n n -- u )
   +
;

: dnegate ( d -- -d )
 \ fix for bug in v2.2
 dnegate 1-
;

: ud+ ( ud ud -- ud )
  ( l h l h )
  >r rot um+ ( h d ) rot u+ ( d ) r> u+
;
: d- ( d d -- d )
 \ 2do: ensure works on full range
 dnegate ud+
;
: d> ( d d -- f )
  d- swap drop 0< not
;

: isqrt ( d -- n )
 $8000 ( d c ) $8000 ( d c g )
 begin
  dup dup um* ( d c g g^2)
  6 pick 6 pick ( d c g g^2 d )
  d> if ( d c g ) over xor then ( d c g )
  swap 2/ $7fff and ( d g c )
  dup 0= if
    drop rot rot 2drop -1 ( g true )
  else
    swap over ( d c g c ) or 0
  then
 until
;

The only trick here is to do a right bit shift by one bit, I've used the 2/
(which accepts positive and negative integers) and co-opted it by just
dropping the sign bit ( $7fff and ).

During my testing, I had the following line just after the 'begin'
 cr 2dup . . \ g in col1 and c in col 2
Interesting to watch how the result gets refined.

For the record, here is the C program I used as the source of the algorithm
    unsigned int sqrt32(unsigned long n)
    {
        unsigned int c = 0x8000;
        unsigned int g = 0x8000;

        for (;;) {
            if (g*g > n) {
                g ^= c; # Bitwise exclusive OR
            }

            c >>= 1;  # Right bit shift

            if (c == 0) {
                return g;
            }

            g |= c;  # Bitwise inclusive OR
        }
    }

Please keep up the good work
Regards, RigTig

TG9541, 2017-02-19:
Hej RigTig, 

I tested the code you posted: it works great, but `256 0 isqrt .` returns `15`, clearly a rounding error. Adding the line `1 0 ud+` to `isqrt` fixes this :-)

Would it be OK to add the code to a `lib` section in the STM8EF repo under a (permissive) OS license?

RigTig, 2017-02-20:
Thomas, 
 Be my guest. It is the least I can do given that I got the algorithm from the net and your efforts underlie what I did anyway. Well spotted about the shortcoming. 
 I am working on a simple GCODE interpreter for a 3D printer and I think it'll just fit onto STM8. If so, I'll be using an ESP14. I have thoroughly enjoying exploring the inner workings of stm8ef.
 Thank you for your public contributions too. 

Regards, RigTig

TG9541, 2017-02-20:
Thanks :-)
I had never thought that one would actually consider moving the limits of what a tiny micro can do that far! This project sounds like a fun - I'm looking forward to more feedback!

closed 2017-02-20 by TG9541

RigTig, 2017-02-20:
Tested the isqrt and confirmed the bug. A little bit of exploring found the culprit: d> was actually d>=. So here is a new d> that seems to be ok now, and you'd never guess that isqrt actually gives 16 as the answer for 256 0 isqrt .

: d> ( d d -- f )
  d- 2dup or 0= ( d f ) rot rot 
  0< swap drop or not 
;
Thanks Thomas, and please keep up the good work. 
Regards, RigTig

TG9541, 2017-02-20:
@RigTig 

Great, thanks! I had chiefly looked in the same direction but I falsely assumed that `d>=` was intentional :-) 

As the GitHub issues appear to work great as a discussion platform!

Here is something that might be relevant for your G-code project and also for `lib/math`:

trigonometry & fractional arithmetics: 

* Fractional arithmetics and CORDIC source (English/Dutch) http://home.hccnet.nl/anij/fractions.zip
* CORDIC Demo (Dutch) http://forth.hcc.nl/downloads/cordic.html
* another CORDIC source https://www.complang.tuwien.ac.at/forth/programs/cordic.fs

I've got no experience with fractional arithmetics (only fixed point), I guess that I should start experimenting a bit. CORDIC used to be widely used (there is even a MCS51 µC, the XC886, with CORDIC hardware for field oriented motor control).

Another that needs solving is a "uploader" program, ideally one that does includes, and that attempts to be smart (i.e. one that tests whether a library has already been loaded). AmForth has an interesting solution, but I'm skeptical about doing constant replacement in the uploader.

commit 2cd78474 (2017-02-12): Resolves #13 DNEGATE not working