TG9541/stm8ef #17: W1209: problems connecting half duplex to sensor connectors

TabbieSuperb, 2017-04-07:
Hi Thomas!
First of all thank you for your incredible work, you have opened for a newbie like me a whole new world.
Sorry if those are noob questions but I think having some feedback from me can be interesting.
I've successfully flashed W1209 by soldering some pins to the unpopulated swim interface and using the usb dongle, with LED display showing a wonderfull "4th". After that, I've removed the C1 capacitor. Next, I have built with a breadboard the half duplex circuit that you kindly published, and correctly wired it to a working ttl dongle (very happy console user of 4th in the STM8 discovery board), but most of my keyboard presses are translated to zeros and some cursor chars. I think console speed may be too fast for my W1209, because when I change speeds to 2400 I start to see correct chars. However, I suspect W1209 firmware is currenty only working at 9800.
Could you please give me some ideas about what to do/test/build?
Thank you very much.

TG9541, 2017-04-07:
Hi TabbieSuperb!

Great to hear that it almost worked for you! 

Let's check if your half-duplex circuit works with the SWIMCOM image:
1. flash SWIMCOM.ihx
2. connect your circuit to pin2 (PD1/SWIM) and pin4 (GND) of the ICP connector
3. try again with 9600/8-n-1

By the way, which STM8 discovery board do you use?

TabbieSuperb, 2017-04-08:
Thank you Thomas!
For your reference, my ultra cheap serial adapter as identified by linux
[ 2636.719707] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60
[ 2636.719712] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2636.719716] usb 1-1: Product: CP2102 USB to UART Bridge Controller
[ 2636.719718] usb 1-1: Manufacturer: Silicon Labs
[ 2636.719720] usb 1-1: SerialNumber: 0001
[ 2636.728066] cp210x 1-1:1.0: cp210x converter detected
[ 2636.729641] usb 1-1: cp210x converter now attached to ttyUSB0
The "discovery board" that I'm using is this one https://es.aliexpress.com/item/J34-Free-Shipping-ARM-STM8-Development-Board-Minimum-System-Board-STM8S103F3P6-Module-for-Arduino/32351992629.html?spm=2114.13010608.0.0.MPsrwT, which is a STM8S003F3P6. It works great. I love the relay of the cheap W1209 and I wanted a closer to it STM8 version for testing...

Circuit works: I've flashed my W1209 to SWIMCOM and I have a working 9600 console wired to SWIM pin! Thank you!

However, after flashing to the W1209_2400 version still no luck with the sensor's pins. May be it is just broken after a bad removal of C1, who knows. I will continue testing it.

Now for the programmable voltage regulator!!! ;)

TG9541, 2017-04-08:
Hi TabbieSuperb,

the problem with your W1209 is interesting - maybe there is a short circuit. You can test it from "within" by reading the ADC using the SWIMCOM image:

```
: ain   ( -- )   6 ADC! ADC@ . ;
```

I've been thinking about changing the W1209 to a SWIMCOM interface for a while because this would make developing code that uses the analog port much easier, and also modifying the board would be less of a trouble. The W1209 uses PD1/SWIM to a segment of the LED display, but the display is "common anode", and might be able to work around it. I'll reopen this ticket, and give it a try.

Good luck with the DCDC - I'm looking forward to hearing about it!

TabbieSuperb, 2017-04-08:
After running some "ain":
`
: ain ( -- ) 6 ADC! ADC@ . ;
 1023 ok
 1022 okain
 1022 okain
 1023 okain
 1023 okain
 1021 okain
 1023 okain
`
I've tested for shorts, didn't find anything. I've cleaned up some excess solder on the board, no luck with sensors. Stranger things! :)

TG9541, 2017-04-08:
When you connect the NTC sensor the ADC value should be 350-ish (and drop when the sensor gets warmer). That would confirm that the circuit works normally, and make this thing even more mysterious ;-)

TG9541, 2017-04-09:
One more thing about your USB serial interface converter: 
I've never tested the `CP2102` with the half-duplex hack. I often use the `CH340`, but its parameters vary a lot. I found that `PL2303` clones work well. If you have access to a scope it would be great if you could have a closer look at the signals. What you can also try is powering the W1209 through the ICP header using the ST-LINK adapter (without applying 12V to the W1209 power terminals).

TabbieSuperb, 2017-04-09:
As you stated, but with may be 10ºC temperature?
 531 ok
 531 okain
 531 okain
 489 ok
 475 okain
 463 okain

I've myself tested SWIMCOM using +12v in board mains and by the ICP header with CP2102 providing +5V and +3V.
I think the simplest solution is to do as instructed by you from the beginning and buy a CH340, and just try! I will try with the scope if somebody led me one while it arrives... ;)
Thanks for everything!

TG9541, 2017-04-09:
Better get a PL2303 clone. 

It's also possible to do something "completely different" and communicate e.g. through the GPIO of key "-".  In order to make waiting for the new USB adapter easier, I quickly made a W1209 binary with half-duplex communication through PC5 ("-" key) for you:

[W1209km.zip](https://github.com/TG9541/stm8ef/files/908179/W1209km.zip)

Here is the changed part of the `globalconf.inc` file: 

```
        PSIM     = PORTC        ; Port for UART simulation
        HAS_TXSIM        = 1    ; Enable TxD via GPIO/TIM4, word TXGP!
        PNTX             = 5    ; Port GPIO# for HAS_TXDSIM
        HAS_RXSIM        = 1    ; Enable RxD via GPIO/TIM4, word ?RXGP
        PNRX             = 5    ; Port GPIO# for HAS_RXDSIM
```

TabbieSuperb, 2017-04-11:
Totally awesome! It works as you designed on the minus key! Thank you very much! I've made a photo with the rehacked cable... ;) Best wishes! http://es.tinypic.com/r/wvbp5j/9

TG9541, 2017-04-11:
Cool! Great that it works. Maybe I'll uses this as the default method :+1:

closed 2017-04-11 by TG9541

TG9541, 2017-07-23:
Please refer to issue #42 for a different approach.