TG9541/stm8ef #18: Compilation of W1209 fails

hexagon5un, 2017-04-10:
Currently fails to build  w/ an error in the W1209 board.  The rest appear to be fine.

```
sdasstm8 -I. -I./W1209 -plosgffw out/W1209/forth.rel forth.asm
forth.asm:151: Error: <i> .include file error or an .if/.endif mismatch
out/W1209/forth.sym:226: Error: <r> relocation error
out/W1209/forth.sym:278: Error: <r> relocation error
out/W1209/forth.sym:328: Error: <r> relocation error
forth.asm:219: Error: <r> relocation error
```

etc.

TG9541, 2017-04-11:
The target.inc file was missing. Thanks for the heads-up!

closed 2017-04-11 by TG9541

hexagon5un, 2017-04-12:
Aha!  I knew it was easy -- looked like a missing file -- but I didn't know how to debug it.  :)

TG9541, 2017-04-12:
sdasstm8 has it's quirks, but this time it was easy:

```
forth.asm:151: Error: <i> .include file error or an .if/.endif mismatch
```

Line 151 is the `target.inc` include, and it was the part before the `or` that triggered the error :-)