TG9541/stm8ef #22: DO LOOP misbehaviour

RigTig, 2017-04-20:
The Gaussian elimination method for solving a matrix involves closing in the limits of a loop, but I found my code wasn't working properly. I've zeroed in on the following misbehaviour of DO LOOP:
`: tt do i . space loop ;`
`4 1 tt`
` 1  2  3  ok`
`4 4 tt`
` 4  ok`
The last result should be blank. Please can someone check and review this and related looping words, as needed. I'll add a check to my code in the meantime.

TG9541, 2017-04-20:
Hi RigTig, 

I'd like to confirm the issue. I guess that the problem is that I implemented `DO .. LOOP` with the [Marcel Hendrix' extended eForth implementation](http://home.vianetworks.nl/users/mhx/eforth.html) as a reference. 

I already fixed dealing with [negative loop increments](https://github.com/TG9541/stm8ef/issues/12), and this is another gap with respect to ANS Forth compliance.

Until I find a good way to fix it, please use boundary testing as a work-around.

RigTig, 2017-04-21:
There is a long history of implementing looping in Forth, many with inconsistencies. I appreciate your effort in chasing this issue. It seems that my understanding did not match normal forth implementation practice; 4 DUP DO .. LOOP executes until the loop index finally wraps around to 3 and terminates as the index increments to 4. This means the actual loop count is determined by the size of the maximum integer for the Forth implementation! 

Forth-83 describes LOOP (http://forth.sourceforge.net/std/fst83/fst83-12.htm#loop) that says it terminates as the index is incremented by 1 to the limit. For DO, (http://forth.sourceforge.net/std/fst83/fst83-12.htm#do), it is far more specific in that the loop is always executed at least once. For +LOOP (http://forth.sourceforge.net/std/fst83/fst83-12.htm, see page 29), the description is virtually identical to LOOP except the increment can be any number.

Forth200x descriptions are even more definitively written (http://www.forth200x.org/documents/forth16-1.pdf), but really result in the same conclusion for this issue. In particular regarding +LOOP, the description of the run-time behaviour says: 

>  Add n [top stack element] to the loop index. If the loop index did not cross the boundary between the loop limit minus one and the loop limit, continue execution at the beginning of the loop.  Otherwise, discard the current loop control parameters and continue execution immediately following the loop.

On the basis of your checking other systems and these references, I am closing this issue, noting that the protective code for starting with loop parameters being equal is required in the user program.

closed 2017-04-21 by RigTig

TG9541, 2017-04-21:
Hi RigTig, thanks a lot for researching ambiguities of `DO .. LOOP` behavior! 

 When I implemented `DO .. LOOP` I decided to make the behavior of +LOOP more robust with regard to invalid boundary values, at the price of limiting the highest number of iterations to 2^15.

I must admit, that I'm still a Forth newbie, and sometimes I don't see all implications of a design decision. Advice is always appreciated!