TG9541/stm8ef #24: Add support for the HC12

TG9541, 2017-05-01:
al177 forked stm8ef and added initial board support for the HC12 (Si4463 not yet working).

To do:
* reverse engineer the HC12
* enable a wireless Forth console on a HC12 through an unmodified HC12
* write docs

TG9541, 2017-05-01:
Approach: 
* connect logic analyzer to the following signals: `nSEL`, `SCLK`, `SDO`, `SDI`, `nIRQ`, `GPIO0`, and `GPIO1`
* Establish connection to a 2nd HC12, and make recordings of the following situations:
  * power up to capture initialitzation
  * send "enter" and receive " ok" to capture basic communication
  * send "enter" and receive a stream of data (e.g. `WORDS`) 
* make the communication more readable with an  AWK script
* extract Si4463 initialization data by comparing multiple recordings with an AWK script
* analyze initialization and communication sequences

Refer to [`Board HC12`](https://github.com/TG9541/stm8ef/wiki/Board-HC12) for first results.

sergey210982, 2017-08-16:
add string:  cmds+=size;
 in si446x_play_cmds

soyebean, 2020-08-15:
Hello, can you give me a detailed configuration of HC12？ thank you

TG9541, 2020-08-15:
@soyebean I had worked on the HC12 and I still have an experimental setup for analyzing the SPI communication between the radio chip and the STM8 in the basement but I never managed to finish it. I'd consider starting over if someone would be willing to share the effort.

soyebean, 2020-08-17:
I want to communicate with HC-12 based on 51 single chip microcomputer si4438 wireless chip, but I don't know the  preamble pattern and synchronization word of HC-12.I have some HC-12 spi data, but I don't know how to analyze it.

[image]
I don’t know why SCLK is a 4-bit or 5-bit clock.

TG9541, 2020-08-17:
Analyzing that data is one of the problems: the datasheet assumes that one uses "Simplicity Studio" and I don't know of any decoder. The description above describes how I had planned to approach reverse-engineering.