TG9541/stm8ef #34: Incorrect description forth.asm

barewires, 2017-06-12:
;       2/      ( n -- n )      ( TOS STM8: -- Y,Z,N )
;       **Multiply** tos by 2.

RigTig, 2017-06-12:
The perils of copy-and-paste strike again. 2/ is almost the same as 2*, well, in assembler source code if not in effect!

barewires, 2017-06-12:
Real programmers still demand perfection, even in documentation after HAL
9000 and Mr. Bill changed our world for the worst.

https://xkcd.com/378/

Still cannot compile with SDCC on a RasPiZeroW, so I am using the older
MINDEX.ihx on a £0.40 stm8s from China.

Comms with W1209 is one way out, see '4th' and stm8eForth V2.2 but no
input. Is the diode pointing out actually intended?

Anyway, no complaints, a great learning experience once again, and I am so
pleased to get back to my roots. Never forgave my dad for not buying me a
PDP-8 (the price of a house or two) in 1965.

All the best,

Roy

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TG9541, 2017-06-12:
That's a genuine Dr. Ting artifact, I s'pose. Fixed. Comes with the next relase. My all time favorite was the one too many "end call optimization" in `COMPILE`. I found it when I implemented `CREATE...DOES>`, it had a really weird effect on the return stack, and I still don't fully understand why STM8EF worked at all.

@barewires: 
1) could you please tell me which release of SDCC you're using on the Pi? 
Compiling the latest version of SDCC on a PI3 might work (but it would be best to use a fast USB disk since the 700MiB of data compiling SDCC creates amounts to torturing an SD Card).
2) There is nothing wrong with the direction of the diode :-)
3) My wife and I still giggle about the PDP-8

closed 2017-06-12 by TG9541

barewires, 2017-06-13:
Hi Thomas,

I am using SDCC version  3.4.0 #8981
My Pi3 is running off a WD PiDrive 3.14 GB so the SD is only for booting.

This acquisition was my PDP-8 clone and manuals sold in 2004. Was used in a
coal yard belt conveyor scale using loadcells and a belt speed tachometer
to integrate coal quantity. The coal dust sucked in didn't help and the
project was scrapped in '72. The company was then very wary about computers
and I couldn't convince them to get with the 70s and use the 8080 so I left
to the competition.
http://bitsavers.informatik.uni-stuttgart.de/pdf/dcc/

I still have to finish building a PiDP-8 kit. Too many irons in the fire.
http://obsolescence.wixsite.com/obsolescence/pidp-8

Thanks for the help,

Roy

barewires, 2017-06-13:
Hi Thomas,

I am compiling now due to the following error in Makefile

sdcc -mstm8 *-oout*/MINDEV/MINDEV.ihx main.c out/MINDEV/forth.rel

changed to

sdcc -mstm8 -o out/MINDEV/MINDEV.ihx main.c out/MINDEV/forth.rel

Next to see if it flashes.

Roy

TG9541, 2017-06-16:
Hi Roy, 
I've never tested `SDCC 3.4.0` (according the the [SDCC homepage](http://sdcc.sourceforge.net/previous.php) STM8 support was introduced with 3.4.0) but I don't think the assembler has changed much (3.6.0 uses sdasstm8 "`Assembler V02.00 + NoICE + SDCC mods  (STMicroelectronics STM8)`").
The whitespace after `-o` _works_ with `SDCC 3.6.0`, and also with the latest development snapshot `SDCC 3.6.6`. However, the whitespace is inconsistent with the rest of the parameters and it's all but impossible that someone removes it in the future.

larsbrinkhoff, 2017-10-01:
@barewires I have my Forth cross compiler generating code for the PDP-8.  Just saying.

commit bf26d50f (2017-06-15): fixes #34 incorrect description in glossary