TG9541/stm8ef #38: Board ZFC39, Fan speed controller

PSLLSP, 2017-06-26:
Hi Tomas,

just FYI, new board ZFC39 with STM8S003 micro-controller. It can control up to three PC PWM fans.

http://www.ebay.com/itm/PC-CPU-12V-4-Wire-Fan-Temperature-Control-PWM-Speed-Control-Module-W-Alarm/282460745283?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

http://www.formfactors.org/developer/specs/4_wire_pwm_spec.pdf

BTW, I am sorry for new issue but there is no contact email noted.

TG9541, 2017-06-26:
Hi PSLLSP,

Opening an issue is good way for submitting a feature request. Thanks for including the fan specs.
Do you own such a board? If yes, did you trace how the STM8S pins are used?

PSLLSP, 2017-06-27:
Hi Thomas,

I have a sample board. There is 78L05 regulator on the board and STM8S003F (TSSOP20) MCU.

MCU is connected in this way:

1 (BEEP) - 150R - 1uF - BUZZ - GND
2 (AIN) - 1k - 50k NTC - GND; 10k pullup to VCC (10k - 50k NTC is voltage divider)
3 (IN) - DP5 (label BF2)
4 (NRST) - 100nF - GND; there is a pad connected to NRST at the bottom of the PCB.
5 (IN) - DP4 (label BF1)
6 (IN) - DP3 (label TP2)
7 (VSS) - GND
8 (VCAP) - 1uF - GND
9 (VDD) - VCC (+5V); there is a pad connected to VDD at the bottom of the PCB.
10 (IN) - 150R - FAN2-3 (RPM); signal is filtered by pull-up 10k and pull-down 470pF
11 (IN) - NC ??
12 (OUT) - red LED - 10k - VCC
13 (OUT) - 150R - FAN3-4
14 (AIN) - trimmer 10k F3; signal is filtered by pull-down 100nF; voltage divider between VCC and GND
15 (IN) - DP2 (label TP1)
16 (OUT) - 150R - FAN2-4 (PWM)
17 (OUT) - 150R - FAN1-4 (PWM)
18 (IN/SWIM) - DP1 (label TFL)
19 (AIN) - trimmer 10k F2; signal is filtered by pull-down 100nF; voltage divider between VCC and GND
20 (IN) - 150R - FAN1-3 (RPM); signal is filtered by pull-up 10k and pull-down 470pF

NOTE, RPM signal from FAN3 is not connected to MCU
NOTE, DPx switches connect signals to GND
NOTE, buzzer has to be actively driven by MCU, it is not self oscillating type
NOTE, NTC is NTC50K3950 (label on PCB)

images of PCB can be found on internet, one example:
https://images-na.ssl-images-amazon.com/images/I/51Pn9NLqUlL.jpg

Video on YouTube: https://www.youtube.com/watch?v=ZldI9R1zOGE

TG9541, 2017-06-28:
OK, looks good: it's easy to get a console (e.g. PD1 "SWIMCOM", or PD1/PD6). Retrofitting a the FAN3 RPM input isn't hard, and I think that the trimmer inputs can be converted into additional temperature sensor inputs.

I've ordered one of those boards. If you'de like to start experimenting: you can flash the `SWIMCOM` image to the board, and start some experiments:

* getting it to producePWM signals takes some Forth words
* a simple stall detection would is easy (several options)
* if you want to actually __measure__, let alone closed-loop-control, the RPM, writing simple pin-change interrupt handlers is maybe the easiest option.
* Functional dependencies temperature to RPM, time to RPM can be done easily with interpolation code [here](https://github.com/TG9541/stm8ef/wiki/Board-CN2596#code-for-power-supply-control).

TG9541, 2017-08-03:
I now have a board, still no fans, though :-)

PSLLSP, 2017-10-11:
Just note about new board. "DC-DC adjustable Power supply module with LCD display", that can be ordered for about 4USD is build around STM8S003K3T6 MCU. This MCU is in LQFP32 package and is hidden under LCD. I see 4 holes on PCB those could be SWIM interface. I see MP2307 on the board too. And two buttons.

http://www.ebay.com/itm/DC-DC-Adjustable-Step-down-Power-Supply-Module-Voltage-Current-LCD-Display-Shell-/142218535015?hash=item211ce2a067:g:WIMAAOSw2xRYWQNf

TG9541, 2017-10-13:
Thanks, ordered one. Do you know how they controlled the LCD?

PSLLSP, 2017-10-13:
I don't know HW details. It could be a challange to write firmware for this board. Some ideas how to drive LCD display could be found here:

http://www.atmel.com/Images/doc2569.pdf
http://www.ti.com/lit/an/slaa516a/slaa516a.pdf
http://www.st.com/content/ccc/resource/technical/document/user_manual/ca/6a/61/13/08/c4/48/f9/CD00278045.pdf/files/CD00278045.pdf/jcr:content/translations/en.CD00278045.pdf

TG9541, 2017-10-14:
Thanks for the links! I'll be looking into how LCD control works in the power supply once it's there.

TG9541, 2017-11-25:
Since the LCD hides the µC I'll have to disassemble the board I have. A second device is now in the mail. Help with reverse-engineering would be appreciated.

TG9541, 2018-04-02:
Just in the case: if anybody would like to see some progress on this board, just drop me a note here. It shouldn't be too hard to program a fan control application on it provided that your requirements are known!

berrydejager, 2018-09-01:
Does anyone know if it is possible to make the temperature ranges lower?
Let's say, while TP1/TP2=OFF/OFF, the lowest accelerating at 25℃ and Full speed at 35℃

```
TP1 | TP2 | Accelerating temperature | Full speed temperature
-- | -- | -- | --
OFF | OFF | 35℃ | 45℃
ON | OFF | 40℃ | 55℃
OFF | ON | 50℃ | 70℃
ON | ON | 60℃ | 90℃
```

TG9541, 2018-09-02:
Hi @berrydejager, I have no knowledge of the code that defines the behavior of the ZFC39 as-is but you can, of course, program the µC on the board to any temperature to fan speed ratio. Right now my time to do programming and testing is a bit scarce, but I'd be happy to help.

TG9541, 2019-11-16:
For reference: a specification document "4-Wire Pulse Width Modulation (PWM) Controlled Fans" from the Wayback Machine is here: [4_Wire_PWM_Spec.pdf](https://github.com/TG9541/stm8ef/files/3853906/4_Wire_PWM_Spec.pdf)

In chapter 2.1.4 the specification requires the PWM frequency to be between "21 kHz to 28 kHz", typical 25 kHz. At that frequency there are 640 steps between 0% and 100% (16Mhz timer clock). The fan PWM inputs have a pull-up to 3.3V or 5V (older models). µC GPIOs should be in open-drain mode (low side) and should be able to sink 5mA (chapter 2.3.1).

Both requirements shouldn't be a problem for STM8S.

dirkscheffer, 2021-03-23:
Maybe there are some other Folks which like visualizations so I made a quick grapic:
[image]

TG9541, 2021-03-24:
@dirkscheffer thanks for the schema :-) 

I've left this issue open because it appears to be useful to people looking for information on the ZFC39 fan speed controller. Currently this type of fan speed controller is marketed as [NTC3950](https://www.aliexpress.com/item/4000559393167.html) or [DC 12V Pwm PC CPU Fan Temperature Control](https://www.aliexpress.com/item/1005002217601860.html) or maybe HW-585. 

The µC on this board is most likely a Nuvoton N76E003AT20, not an STM8S003F3P6. This means that in order to use the Forth OS in this repository instead of the stock firmware you need to replace the µC.

RaptorSDS, 2024-01-14:
i have now such a board , the CPU has a sticker with new label also my board has i2c pins near MPU and two pins that looks like button ( label SY and TP) did someone know the funktion ?
also when need i can try read the pins  or maybe dump the rom when someone tell me how

RaptorSDS, 2024-01-22:
[image]