TG9541/stm8ef #40: More photos for the Wiki

TG9541, 2017-07-16:
Unlike GitHub Issues the GitHub Wiki doesn't have an image upload feature. 
This issue provides storage for the Wiki.

TG9541, 2017-07-16:
http://solutionoptimist.com/2013/12/28/awesome-github-tricks/

[image]

closed 2017-07-17 by TG9541

TG9541, 2017-07-23:
W1209 full duplex serial interface using test hook clips
[image]

W1209 board with attached serial interface cable
[image]

TG9541, 2017-07-23:
[image]

TG9541, 2017-11-30:
[image]

TG9541, 2018-01-07:
[image]

VK6TT, 2018-02-11:
Some boards pics

[image]
[image]
[image]
[image]
[image]

VK6TT, 2018-02-12:
Some boards pics

[image]

VK6TT, 2018-02-12:
CC pics

[image]
[image]

VK6TT, 2018-02-12:
[image]

TG9541, 2018-05-27:
[image]
[image]

TG9541, 2018-05-27:
[image]

TG9541, 2018-05-27:
[image]

TG9541, 2019-01-12:
[USBaspUSBaspLoader.zip](https://github.com/TG9541/stm8ef/files/2751947/USBaspUSBaspLoader.zip)

TG9541, 2019-09-15:
[image]

TG9541, 2019-09-15:
[image]

TG9541, 2019-09-15:
[image]

TG9541, 2019-09-15:
[image]

TG9541, 2019-09-15:
[image]

TG9541, 2020-01-07:
[image]

[image]

TG9541, 2020-03-29:
Pseudo-W1219
[image]

TG9541, 2020-09-13:
[image]

TG9541, 2020-10-19:
[image]

TG9541, 2020-10-23:
STM8S105x4/6 Pinout and pin description 

Figure 3. LQFP48 pinout
[image]

Figure 4. LQFP44 pinout
[image]

Figure 5. UFQFPN32/LQFP32 pinout
[image]

TG9541, 2020-10-28:
[image]

TG9541, 2020-10-28:
[image]

TG9541, 2020-11-01:
[image]

TG9541, 2020-11-02:
[image]
[image]

TG9541, 2020-11-15:
[ Experiments with the STM8 I2C peripheral ](https://gist.github.com/TG9541/5c3405320794d91ef8129734a4bfc880)

[image]

TG9541, 2020-11-23:
[image]
There is obviously no way for resetting the TXE flag when there is no write to I2C_DR.

TG9541, 2020-12-19:
[image]

[image]

TG9541, 2022-05-02:
[image]
[image]

TG9541, 2023-02-05:
[image]

TG9541, 2023-03-26:
[image]