TG9541/stm8ef #42: Improve W1209 ease of use

TG9541, 2017-07-22:
Several users have had difficulties getting the W1209 serial communication to work, e.g.:
* https://hackaday.io/project/16097-eforth-for-cheap-stm8s-gadgets/discussion-90018
* http://forum.hobbycomponents.com/viewtopic.php?f=73&t=1900&p=5498#p5498
This issue should improve the "user experience". 

Issue #41 enables using `e4thcom` (and regular serial terminal programs) more easily. Consistent documentation, board support, and decisions about "phasing out" sensor header communication shall be taken here.

TG9541, 2017-07-22:
Issue #41 is now closed. e4thcom works fine using PC4/PC5 (keys "+" and "-"), and using the board keys in interactive mode works, too.

TG9541, 2017-07-22:
@barewires: according to the hobbycompents W1209 STM8EF forum thread your experience with the sensor header serial interface were, well, not good. What's your opinion about the new solution?

alfredluca, 2017-07-23:
I tested the W1209-FD.ihx binary and wrote a "how-to":

How-To:
* get a W1209, erase the chip if necessary
* flash the W1209-FD.ihx binary
* connect the serial interface (+key pin near LED display -> TxD, -key -> RxD)
* use terminal program "9600/N/8/1"

It's as easy as using a STM8S103F3 breakout board!

TG9541, 2017-07-23:
Thanks for the feedback so far!

I [updated the Wiki](https://github.com/TG9541/stm8ef/wiki/Board-W1209#serial-communication-through-the-key-pins).

barewires, 2017-07-24:
Great work Thomas! I had put the W1209 on the back burner concentrating on the MINDEV, RCWL-0516 Cat Detector and HC12 comms. My only thought is grounding the TX/RX and debounce. Still a much appreciated update.

TG9541, 2017-07-24:
Hi Barewires, thanks for the feedback! 

I read your latest hobbycomponents forum post - together with Alfredo's post it makes a good account of what can go wrong, and why independent testing is important. I hope that we're on track now. Thanks a lot :-)

I also updated the Wiki (please refer to the link in my previous comment). It would be nice if you could comment on it's usefulness.

You mention debouncing measures for RX/TX. Did you observe communication errors? 
Can you propose a circuit (e.g. RC)? 

PS:  it's great that Picocom works. That's a good starting point for using [e4thcom](https://wiki.forth-ev.de/doku.php/en:projects:e4thcom#e4thcom-061) (it now has a STM8EF plugin, and it should work on the Raspberry Pi). Here is a [small demo](https://gist.github.com/TG9541/321e58dd4c837bb3625d93bf845562f3) what it can do to make an STM8EF programmer's life easier.

TG9541, 2017-07-29:
Let's then use the "full duplex" option as the default. The docs will be changed accordingly which, I guess, will make new users life much easier.

commit 3cac0d5a (2017-07-22): fixes #42 Improve W1209 ease of use
commit 649a0de2 (2017-07-22): fixes #42 Improve W1209 ease of use
closed 2017-07-29 by TG9541