TG9541/stm8ef #48: Question: Why STM8?

larsbrinkhoff, 2017-08-08:
Hello,

I have a question.  Why did you chose to work with the STM8?

I don't have any numbers, but it seems the 8-bit MCU market is pretty much ruled by PIC and AVR devices, and the 8051 also seems to hang in there.  So it's interesting to see something else.

larsbrinkhoff, 2017-08-08:
Found the answer:  
https://hackaday.io/project/16097-eforth-for-cheap-stm8s-gadgets/details

Thank you for an interesting project!

closed 2017-08-08 by larsbrinkhoff

TG9541, 2017-08-08:
Hi Lars, it's good that you found the answer. 

However, I guess that the main reasons were highly personal - here is a list:

* when I was 15 I failed to understand the Forth method, no matter how many colon definitions I wrote 
* at the age of 18 I wrote a compiler to translate a functional language with infix notation to stack machine code, and the interpreter to run it - I knew that something was wrong
*  I did embedded programming for almost two decades - debugging µCs programmed in C was so awkward that I learned to avoid coding errors. However, I always remembered the promise of Forth!
* two years after I finally quit my job, I found Dr. C.H. Tings STM8EF slides, understood what was wrong with my take on Forth 30 years ago, and decided to find out how far one can get by applying 1980s methods to 2016's ultra-low-end µCs
* and finally, in my new job I deal with systems of mind-boggling complexity - STM8EF acts as overpressure valve for a low-level-coding-and-optimizing-things habit ;-)

In the meantime there are of course other reasons, above all learning to know people in the Forth community :-)

larsbrinkhoff, 2017-08-08:
Hello and thanks for the answer,

I think I understand why people are curious about Forth.  However, I was wondering why you chose STM8 over other 8-bit devices like AVR or PIC.  Those two seem most popular, so I was curious when I saw your project.