TG9541/stm8ef #49: Question: Which STM8 board?

larsbrinkhoff, 2017-08-23:
Hello,

I'm considering STM8 as a target for my Forth cross compiler.  Which board would you recommend using for testing?  Ease of use is more important than cost or features.

Thanks!

TG9541, 2017-08-23:
I guess that you don't always need an UART level shifter for testing your cross-compiled Forth. You have a couple of options:

* a STM8S103F3 breakout board and some of the items described [here](https://github.com/TG9541/stm8ef/wiki/Breakout-Boards#getting-started)
* @RigTig experimented with [interactive usage of an ST-LINK V2 adapter](https://github.com/TG9541/stm8ef/issues/32#issuecomment-313638960)
* uCsim has good support for the STM8, and I wrote a bit about [how to use uCsim in the Wiki](https://github.com/TG9541/stm8ef/wiki/STM8S-Programming#sdcc-the-simulator-ucsim)

I would recommend starting with uCsim as it's also a good foundation for automated compiler testing (that's why it's an integral part of the SDCC suite). STM8EF provides a script for compiling Forth code in a uCsim environment.

Please let me know if you have questions about STM8 or uCsim, or if you need someone for discussing VM and coding options.

TG9541, 2017-08-23:
Reference: https://github.com/larsbrinkhoff/xForth/issues/1

larsbrinkhoff, 2017-08-24:
Thank you.  I only require something to indicate success, like blinking a LED.  I'm not so handy with a soldering iron, so an assembled board would be nice.

Yes, I'm planning to use uCsim for STM8.  It was the only suitable simulator I could find.

hexagon5un, 2017-08-24:
Seconding what Thomas said -- if you just need a blinking light to indicate success, the STM8S103F3 boards are the way to go.  No soldering to blinky, and dirt cheap.

larsbrinkhoff, 2017-08-24:
Thank you both.

TG9541, 2017-08-24:
Lars, I can send you a soldered STM8S1033 breakout-board via mail if you send me your address (e.g. via a Hackaday private message). I'd recommend to order a ST-LINK (V2) dongle, and maybe a solderless breadboard, some patchwires, and a serial interface dongle as described in the link at the first comment.

One more thing: I think that it would also be feasible to embed cross compile code into STM8EF, e.g, using something like the interrupt context switch code.

larsbrinkhoff, 2017-08-25:
Thank you @TG9541, that's very helpful!

I'm not sure what you mean with embedded cross compile code?  Would STM8EF be the target of a cross compiler?

larsbrinkhoff, 2017-08-25:
By the way, since you work in automotive.  Do you know if 68HC08 or decentants like RS08 are still in use?  It seems they are or were popular in the automotive industry.

TG9541, 2017-08-25:
I work in a field with very high resource requirements, so I have only limited insight. Today, general purpose application software stacks require at least mid-range 32bit single core controllers ("abstraction costs" are high). 8bit controllers are in use in very cost sensitive sensor/actor applicstions. LIN stacks are frugal, and most 8bit controllers, including the RS08, have the required resouces. Protocols like SENT (SAE2716)  need even less than that.

TG9541, 2017-08-25:
About the cross compiled code: I mean code compiled e.g. on a PC instead of compiled on the target. With embedded I mean "wrapped into an STM8EF word". STM8EF already has two context switch modes (for the background task, and for interrupts coded in Fort,  and I guess that a wrapper for xForth code would look similar).

larsbrinkhoff, 2017-08-26:
Sure, everything xForth does to the target memory image goes through special `c!` and `c@` words.  If we could direct those calls to STM8EF, it would be a perfect target for tethered operation.

TG9541, 2017-08-26:
Tethered operation opens up many new opportunities. Would it also be possible to use STM8EF words in xForth? Features for extracing the dictionary  (including the CFA) already exist. It would be necessary to use  the X register as the data stack pointer, and to align RAM management somehow.

larsbrinkhoff, 2017-08-27:
Yes, the compiler should be able to target different models, including the one used by STM8EF.

I have never done tethering, so there are probably many things lacking.  A facility to command the target to execute a word would also be good.

The host-target interface could be just sending regular text commands to the STM8EF interpreter.  Inefficient, but simple and powerful.

TG9541, 2017-08-27:
Directing the `c!` and `c@` to an STM8EF target shouldn't be a problem. However, for cooperation with the embedded Forth some more facilities might be necessary (e.g. `:`, `[`, `]`, `overt`, `here`, and `c,`). It's also possible to set aside a Flash area for tethered coding but, considering the small amount of memory, a uniform address space would be better.
Executing words remotely is easy: push data on the stack, and call `execute`. However, getting the data back through memory and stack dumps is also required. In [tools/simload.sh](https://github.com/TG9541/stm8ef/blob/master/tools/simload.sh) I used some tricks, like injecting serial I/O words into the target address space.
Speed is certainly a concern. However, the small size of the target Flash ROM, and the inherent "incremental compile" capability of Forth attenuates that.

larsbrinkhoff, 2017-08-31:
Closing this, because the original question has been answered.

closed 2017-08-31 by larsbrinkhoff