TG9541/stm8ef #50: Recoded words

larsbrinkhoff, 2017-09-01:
When porting my Forth to the STM8, I relied heavily on the STM8EF kernel for guidance.  However, I came up with alternative implementations for some words that avoid use of `YTEMP`.  I'm not sure if they're an improvement or not.  >R can be further reduced in size with the use of EXGW.

!

    90 93          ldw Y, X
    90 fe          ldw Y, (Y)
    89             pushw X
    ee 02          ldw X, ($02,X)
    90 ff          ldw (Y),X
    85             popw X
    20 f0          jra 2drop

\>R

    16 01          ldw Y, ($01,SP)
    90 89          pushw Y
    90 93          ldw Y, X
    90 fe          ldw Y, (Y)
    17 03          ldw ($03,SP),Y
    81             ret

R>

    16 03          ldw Y, ($03,SP)
    ad f5          callr pushy
    90 85          popw Y
    5b 02          addw SP, #$02
    90 fc          jp (Y)

TG9541, 2017-09-01:
Thanks for the code!

`!`: I can use it as it is. 

`>R`: you were right about the EXGW but I think that you forgot to DROP TOS
```
      008239 51               [ 1] 1090         EXGW    X,Y
      00823A 1E 01            [ 2] 1091         LDW     X,(1,SP)
      00823C 89               [ 2] 1092         PUSHW   X
      00823D 93               [ 1] 1093         LDW     X,Y
      00823E FE               [ 2] 1094         LDW     X,(X)
      00823F 51               [ 1] 1095         EXGW    X,Y
      008240 17 03            [ 2] 1096         LDW     (3,SP),Y
      008242 20 11            [ 2] 1097         JRA     DROP
```

`R>`: your implementation of doesn't return with TOS in the Y register (as the old implementation does).

larsbrinkhoff, 2017-09-01:
Good catch with the missing DROP.  I have a test for >R, but it didn't over that failure.

commit 1b5486c7 (2017-09-01): fixes #50: recoded words @larsbrinkhoff
commit d9c4bd72 (2017-09-10): fixes #50: code size reduced
closed 2017-09-01 by larsbrinkhoff