TG9541/stm8ef #53: TOS in low memory

larsbrinkhoff, 2017-09-05:
Hello,

Have you considered putting the top data stack element in low memory, where it can be accessed using short memory addressing mode?

larsbrinkhoff, 2017-09-05:
Never mind.  I tried it, and it didn't seem to be a win.

closed 2017-09-05 by larsbrinkhoff

TG9541, 2017-09-05:
Hi Lars,
TOS in a "zero page" cell is an interesting idea, however most 16bit operations demand X or/and Y as an operant. Great that you already did a test implementation!

Using the return stack as data stack with (n,SP)-addressing looks interesting, but this requires a DTC or ITC implementation, and interrupt handlers in Forth would be a bit risky.

larsbrinkhoff, 2017-09-05:
Right, I wanted to use SP-relative addressing many times.  The sdcc compiler use it to access parameters and local variables.  It's a shame a native-code Forth compiler can't.

TG9541, 2017-09-05:
Do you know any source code of a DTC Forth kernel that used the native return stack as the data stack?

larsbrinkhoff, 2017-09-05:
DTC, no.  ITC, yes.  Can't say I know many Forth implementations, though.