TG9541/stm8ef #59: Compilation  fails

timsoer, 2017-09-16:
sdasstm8 -I. -I./MINDEV -plosgffw out/MINDEV/forth.rel forth.asm
forth.asm:1447: Error: <i> .include file error or an .if/.endif mismatch

TG9541, 2017-09-16:
What was your command line?

TG9541, 2017-09-16:
This was a `make BOARD=MINDEV flash` in  89ef113, right?

TG9541, 2017-09-16:
OK, confirmed - thanks for reporting! 
A new release with some improvements was planned for today, it will also fix this bug. Please remove line 1446 for now!

closed 2017-09-16 by TG9541

TG9541, 2017-09-19:
With bed2c73 @larsbrinkhoff fixed the root-cause of this issue once and for all: there is now a travis-ci continuous integration set-up. Thanks a lot!

larsbrinkhoff, 2017-09-19:
That was the idea!  Happy to help.

TG9541, 2017-09-24:
I just enabled "Branch protection for master". I would happily also apply the restrictions also to administrators provided the team is large enough (i.e. time to review is guaranteed). 
[image]

larsbrinkhoff, 2017-09-24:
I think requiring status checks, e.g. passing builds on Travis CI, also makes sense.

TG9541, 2017-09-24:
You're right about the status checks. I didn't realize that Travis-CI acts as a status-check service.

I made some progress with containerizing the tool chain but there's still a problem with a recent change in uCsim that stops the simulator after injecting UART code. Automated eForth core tests will be good to have.

PR #60 (2017-09-16): Develop
This also fixes #59 (the merge prior to release v2.2.14 went totally wrong).