TG9541/stm8ef #66: Contribution guidelines

larsbrinkhoff, 2017-09-22:
I suggest it might be good to add some brief contribution guidelines.  Such as, what is the branching strategy involving `master` and `develop`?

Note that if it's named exactly `CONTRIBUTING.md`, GitHub will take notice and offer it for display when users make pull requests.

TG9541, 2017-09-22:
I've had that on my todo list for a long time. Do you know of a good&simple example?

Of course feature branches, merging to develop, and then merging to master is a good idea. Also guidelines or suggestions for other forms of contributions might be worth thinking of (Wiki, Gists, application code).

larsbrinkhoff, 2017-09-22:
I only ever wrote one, but it's not very applicable.  Feel free to pick anything you like.  
https://github.com/PDP-10/its/blob/master/CONTRIBUTING.md

Browse files on GitHub: https://github.com/search?q=filename%3ACONTRIBUTING.md

Maybe these can be some inspiration:  
https://github.com/sodabrew/puppet-dashboard/blob/master/CONTRIBUTING.md  
https://github.com/drj11/sixtyforth/blob/master/CONTRIBUTING.md  
https://github.com/cysys/stratum/blob/master/CONTRIBUTING.md

TG9541, 2017-09-23:
Thanks, that's great! 

* the [Puppet rules](https://github.com/sodabrew/puppet-dashboard/blob/master/CONTRIBUTING.md) are more than a bit over the top for this project (but still very instructive).
* the [stratum rules](https://github.com/cysys/stratum/blob/master/CONTRIBUTING.md) fit the bill 
* I like the [sixtyforth](https://github.com/drj11/sixtyforth/blob/master/CONTRIBUTING.md) style guide (with an assembly supplement)
* I'd like to add some guidance on Open Source contributions (ISC style?)  

I made a [Gist with a draft](https://gist.github.com/TG9541/d694d38275369a5eea93364b9f006ef5). I'm looking forward to comments and improvements!

commit 141b4a36 (2017-09-29): Merge pull request #79 from TG9541/contributing

closes #66
commit b357b918 (2017-09-29): closes #66
closed 2017-09-29 by TG9541