TG9541/stm8ef #68: Minor version number

barewires, 2017-09-23:
May I suggest having the minor version number appended to the header:
stm8eForth v2.2.xx

TG9541, 2017-09-23:
The revision number should indeed be visible (the 2nd group of digits is usually called the minor).  
But what should we do with development versions, and when should be increment the revision number?
Maybe we need to be more generous with the minor.
Any thoughts?

barewires, 2017-09-25:
I suppose with all of the targets and devices that it could easily become unmanageable. Best if I put it in eeprom for my own records. Yes, a complex issue but worth thinking about. Thanks, no longer an issue.

barewires, 2017-09-25:
Just blasted the latest version 2.2.16 and as a major release can you reconsider adding .16
/home/pi/Downloads/2216/out/MINDEV/MINDEV.ihx

hi
stm8eForth v2.2.16

TG9541, 2017-09-26:
I'll be looking into how to pass the revision number, and the release status, into the build chain.

TG9541, 2017-10-22:
This issue will be resolved in #67: the revision number will be generated from a "package.json" file

commit 09127b88 (2017-10-28): fixes #68
closed 2017-10-22 by TG9541