TG9541/stm8ef #113: "It's Like Déjà Vu All Over Again"

barewires, 2017-11-12:
Hi Thomas,
Sorry to be a pest but could we have the sub-sub-minor version implemented?
STM8eForth 2.2.20.pre

TG9541, 2017-11-12:
The purpose of the pre-release is the same as that of a [Maven snapshot](https://stackoverflow.com/questions/5901378/what-exactly-is-a-maven-snapshot-and-why-do-we-need-it). Its purpose is test (regression, integration, ...), and the assumption is that users of pre-releases (or snapshots) know that the paint is too fresh to stick a tag on it.

It might be possible to replace the string, but it's not easy:
* the build gets triggered, and the binary is deployed by setting a release tag (`git tag 2.2.20.pre`, then `git push origin --tag`). 
* Travis-CI is triggered by the tag, and uses the tag name for deployment

The problem is that the source code doesn't change at all. Changing the source code in the build system is possible, but the result would be different if there is no tag. 

I'm of course open to suggestions :-)

TG9541, 2017-11-15:
I'll close this issue for now. I'll reopen it if someone has a solution in the context described.

closed 2017-11-15 by TG9541