TG9541/stm8ef #124: codeload.py does not support half duplex

rlourette, 2017-11-27:
Just an FYI, that codeload.py does not support half duplex. The description doesn't advertise this, so it really is not a bug. I still need a way to use my W1219 board with a windows platform.

Rich

rlourette, 2017-11-27:
Apparently, it does not handle a reset command in the file as the board returns "STM8eForth 2.2.20" instead of 'OK'.

TG9541, 2017-11-27:
Hi Rich,
thanks for the feedback on using Windows!

`codeload.py` should actually support half duplex (I just tried with my W1219 board on Linux, and it works fine). It's likely that there are differences in the python serial line support for Windows, and I would appreciate some assistance in getting it working. The serial protocol is rather robust, though. It uses the OK response as a handshake, and it even tolerates the CPU core "stalling" during Flash write operations.

The second thing you mention is something I worked around by having `'BOOT` point to `.OK` (like [here](https://github.com/TG9541/stm8ef/blob/014fcae7a52ec5bc1414e47319983012fff40a91/test/board.fs#L73)). 

However, it would be much easier if STM8 eForth would just print `ok` at the end of `HI`. What do you think?

TG9541, 2017-11-27:
Please also have a look at #125

TG9541, 2017-12-29:
Closing for now. Please re-open if new information is available!

closed 2017-12-29 by TG9541