TG9541/stm8ef #135: Wiki: wrong/contradictury statement about code execution from EEPROM

alfredluca, 2017-12-28:
In the Wiki there is are [two statements about code execution from the EEPROM](https://github.com/TG9541/stm8ef/wiki/STM8S-Programming#code-execution-from-different-memory-areas):
1. it doesn't work (with demonstration)
2. it works (without demonstration)
I guess that there was a wrong statement, which was only partially corrected!

TG9541, 2017-12-28:
Fixed. Code execution from RAM is a feature that's not implemented in EEPROM, and I don't know why I ever claimed that it would work.

TG9541, 2017-12-28:
There were also wrong log entries on HackaDay, which I edited.

closed 2017-12-28 by TG9541

TG9541, 2018-03-25:
#178 introduces a way to still use the EEPROM for Forth: put ALIAS dictionary entries in it, and save Flash memory for the executable part.