TG9541/stm8ef #136: Ultrasonic HC-SR04, W1219

RandoSY, 2017-12-29:
Do you have some insight where it would be best to pull of a couple of GPIO pins, (or possibly just 1, as both output for trigger, and input for echo) to drive a low cost ultrasonic ranger sensor? I guess the pins on the SWIM make sense, but I would love to have the console up for testing/programming.

Struggling to find schematic for this board, as well ...

TG9541, 2017-12-29:
This sounds like a fun! Which kind of application do you have in mind? I'd be interested to hear about use-case and features :-)

Unfortunately, I don't know of any full schematic of the W1209, but I think that GPIO list will be sufficient to get the ultrasonics sensor working.

Besides PD1 (not the best choice) I see the following options:

* one (or two) of the keys PA1, PA2, and PD6
* PD4, the GPIO that's connected to the blue LED

Using a key usually works fine, since operating the keys won't happen too often, and it's likely that errors caused by pressing a key can be filtered out. The only thing I'm a bit uncertain of is interrupt latency. Using a pin change interrupt on a key, and a free running TIM1 might be an option. You should take into account the error caused by interrupt latency which can be in the order of 1cm (most likely that's not really a problem).

PD4 offers the option of using a TIM2 capture: it's most likely possible to run the distance measurement in the background task. A minor modifications of the TIM2 ticker routine might be required. I think that the LED can still be mostly operational (a bit dimmer when lit, a little flicker when off).

TG9541, 2017-12-30:
I did a quick check: the blue LED is indeed cathode controlled (i.e. activated by driving PD4 low). Based on what I know about the HC-SR04 this means that the LED must be either removed, or some extra circuitry must be installed.

Removing the LED is maybe the best option. It would make it possible to install a 3-pin sensor header (the 3rd pin can be soldered to the GND copper area). Replacing the LED series resistor R11 (10k) with a bridge (or e.g. 4R7) brings the 5V supply to a pin.

RandoSY, 2017-12-30:
Hi Tom,

 

Thanks so much for the consideration of your reply.  I got shipping notification on my unit, wait slow boat literally from China, and then I can try what you advise.

 

I will keep you posted on progress, thanks for all your hard work to bring this swell project to the community!

 

Best,

 

Randy

RandoSY, 2017-12-30:
Thomas,

 

I want to extent the thermostat function for a couple of simple thermodynamic mixing experiments by being able to provide physical measurement of the containers I use. 

 

Standards and measurement activities in the public schools. 

 

Needs to be DIRT CHEAP.

 

Best,

 

Randy

TG9541, 2017-12-30:
Thanks for the insight! 

I have a rough idea of the public school system in the USA, and I've had opportunities to discuss projects with middle school teachers. The educational aspects of classroom experiments require considerable didactic skills. There is nothing as motivating as an inspired teacher, of course!

Out of curiosity, can you share some requirements, e.g.

* who will build the devices (skills, safety)
* measurement: units, range, and accuracy
* rate of measurement (1/s)
* user interface: LED number groups/digits, values, keys, serial console

Maybe I know of an alternative solution.

RandoSY, 2018-01-02:
Hi Thomas,

 

First off, best wishes for a healthy and prosperous New Year, you and yours!

 

Right now I am targeting pre-service teacher and of course later their students. I have a Science Workshop series that provides educational services for the local community. These kids a K-10 usually.

I use SI units, English units for comparison, and teach the difference and conversion as part of the curricula. SAFETY IS A PRIORITY.

 

The rate of measurement is around 1hz. Plenty fast enough.

 

Working hard on all sorts of user interfaces, esp. format to provide an opportunity for base number conversion, binary, hex, and ASCHII encoding.  OLED seems to be economical and high performance.

 

I really like the idea of having a cheap thermometer, that I con “conscript” into being an ultrasonic  measuring  tool. 

 

Best,

 

Randy

TG9541, 2018-01-02:
Hi Randy,
thanks for the greetings! The same goes back to you across the Big Pond! 

I think I've now got an idea of what you're up to. For building instruments Forth powered low cost thermostats are a good choice. Of course it depends how much "control" plays a role in the experiment.

If you intend to use metric units a display range of 0..99 should be sufficient for many experiments, and  the W1401 board with 3 x 2 digits LED should also be usable. The low-cost champion is, or course, the W1209 (1 x 3 digits) is also a good candidate. Here, the HC-SR04 can be connected to the "set" key PC3/TIM1_CH3. Using the TIM1 capture feature a very good low-noise/hi-res distance measurement should be possible.

By the way, I recently made a [data logging thermostat](https://github.com/TG9541/W1209#w1209-data-logging-thermostat) demonstrator. You might find some of the code interesting.

RandoSY, 2018-01-03:
Thomas,

 

Just wondering, you’re UK based, yes? Are you an academic person?

 

I do a lot of research (when I get the time,  which is getting more and more a valuable, rare commodity) into international education curricula. 

 

Best,

 

Randy

TG9541, 2018-01-03:
Hi Randy! 

No, I don't have an academic background. I'm just an R&D engineer with functions related to QA-Methods&Tools innovation. I'm German - British friends used to say that I speak like an "old bat", which is most likely due to the age bracket my English teachers at high school were in. Working with Indians on a daily basis ("please do the needful"), or Australians ("bat" again, for obvious reasons) didn't help. To make things worse, my exposure to American English is dominated by the Washington Post, and scholars like Daniel Kahneman or Nancy Leveson. The occasional game of Cards Against Humanity fails to counterbalance the British influence.

I got interested in curricula lately because it's useless to assume that all people are self-directed learners, and understanding how people learn is important if one wants to have any success with methods innovation. Building curricula is a very difficult task, especially if one wants to build cross-domain skills.

Best,
Thomas

RandoSY, 2018-01-03:
Thomas,

 

That’s exciting, you’re the second correspondent I’ve established this year who is German, and coincidently is very much interested in the STM8. I forwarded him your excellent work, and he was able to apply it to immediate advantage. His English skills are equally phenomenal, and leave me with nothing but a little sense of awe… and more than a little deficit.  Sometimes I wished I spent more time having a rounded educational experience, rather than chasing dollars, which is I’m quite afraid to say, an American weakness.

 

https://github.com/gicking/STM8_templates

 

Georg’s coding style is quite spare, and elegant, as is his prose.

 

Since today is still holiday for me, and I’ve pretty much recovered from my 1 beer, and 1 glass of Champagne hang over, I will look at German education content today. I’m sure much to learn.

 

Thanks again for taking the time to write,

 

Best,

 

Randy

VK6TT, 2018-01-04:
Thomas
 
your prose can only be improved by speaking with Australians. We understand
how butchered the English language is by accent. It's why we have our own
slang.
 
Gotta go, the dead horse with sauce (meat pie and tomato sauce) is ready.
Afterwards I'll be as full as a goog. *grin*
 
Cheers
Richard

  _____  

y4X6-Rgks5tG67pgaJpZM4RPRbG> the thread.
<https://github.com/notifications/beacon/AT7ya3thYaAAz-MpGF8mrQmDEpL-LR_Jks5
tG67pgaJpZM4RPRbG.gif>

TG9541, 2018-01-04:
Richard, I'm sure I'll find use for both idioms, soon (other Australian slang for meat pie is even more colourful)

TG9541, 2018-01-21:
@RandoSY :
I received a HC-SR04 and connected it to a W1209 through PC3/TIM1_CH3 (the `set` key input). With the STM8S Timer1 puls-width measurement feature it works surprisingly well (up to about 3m). 

The W1219 doesn't have an input that's equally well suited, but I'm sure that something can be hacked with PD4/TIM2_CH1 (although the background task might have to use TIM1 then). 

Check out the code here: https://gist.github.com/TG9541/617c9c719eee3c9c2345d69a505491a7

RandoSY, 2018-01-31:
Thanks so much for posting this! Just what I needed, and certainly much appreciated... :)

closed 2018-01-31 by RandoSY