TG9541/stm8ef #142: uCsim simulator stalls ... Ubuntu 16.04

RandoSY, 2018-01-02:
Howdy!

Just running a quick test of the uCsim example here ...

./sstm8 -g -w -tS103 -Suart=1,port=10000 out/MINDEV/MINDEV.ihx and telnet localhost 10000 which seem to start and run OK no error. Control-X on the terminal side and type "g" to starts the simulator, but no response after that.  

Any thoughts? I feel I must be close...

[image]

TG9541, 2018-01-02:
Hi Randy,

`uCsim 0.6-pre30` works for me, but, however, the ihx-file doesn't appear to load. Could you please check of the path `out/MINDEV/MINDEV.ihx` contains a valid file?

A normal uCsim start should look like this:

```
thomas@w500:~/source/stm8s/stm8ef$ sstm8 -g -w -tS103 -Suart=1,port=10000 out/MINDEV/MINDEV.ihx
uCsim 0.6-pre33, Copyright (C) 1997 Daniel Drotos.
uCsim comes with ABSOLUTELY NO WARRANTY; for details type `show w'.
This is free software, and you are welcome to redistribute it
under certain conditions; type `show c' for details.
0> Loading from out/MINDEV/MINDEV.ihx
4808 words read from out/MINDEV/MINDEV.ih
```

Note:  I had a lot of exchange with the author of uCsim to get the scripted STM8 eForth working. Since the main use case for uCsim in the SDCC suite isn't integration testing (rather unit-testing) already the next revision of uCsim broke the brittle chain of interactions uCsim and interactive test in `simload.sh`.
If you want an environment that's guaranteed to work, please try using the containerized build- and test environment:

First terminal:
```
thomas@w500:~/source/stm8s/stm8ef$ docker run -ti -v `pwd`:/home tg9541/docker-sdcc:V1.0.1 /bin/sh -c "cd /home && sstm8 -g -w -tS103 -Suart=1,port=10000 out/MINDEV/MINDEV.ihx" -expose=22
uCsim 0.6-pre32, Copyright (C) 1997 Daniel Drotos.
uCsim comes with ABSOLUTELY NO WARRANTY; for details type `show w'.
This is free software, and you are welcome to redistribute it
under certain conditions; type `show c' for details.
0> Loading from out/MINDEV/MINDEV.ihx
4808 words read from out/MINDEV/MINDEV.ihx
```

2nd terminal:
```
thomas@w500:~/source/stm8s/stm8ef$ telnet 172.17.0.2 10000
Trying 172.17.0.2...
Connected to 172.17.0.2.
Escape character is '^]'.
uart[1] terminal display, press ^x to access control menu
 ok
words
 IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT OUT! COLD ok
```

You can also use that for running a make target in a cloned GitHub repository:
```
docker run -v `pwd`:/home tg9541/docker-sdcc:V1.0.1 /bin/sh -c "cd /home && make BOARD=MINDEV"
```

If you put a file `board.fs` or `main.fs` in a board folder you can also run Forth code, e.g. for automated testing. This even works for binaries with simulated UART:
```
docker run -v `pwd`:/home tg9541/docker-sdcc:V1.0.1 /bin/sh -c "cd /home && make BOARD=test forth"
```
If everything

RandoSY, 2018-01-03:
Thank you for this helpful, considered summary. Time to learn some "docker"

closed 2018-01-18 by TG9541