TG9541/stm8ef #152: Word Digit appears to be missing from 2.2.21

VK6TT, 2018-01-25:
Tried the latest binary today for my WD1209-FD testing. Found that the word "digit" is no longer in the pre-compiled binary. Which is odd because the forth.asm file in the master shows the entry. (Line 2860)

VK6TT, 2018-01-25:
Maybe this is related to a possible vocabulary issue alluded to in #153?

TG9541, 2018-01-25:
Please check your `target` path and try using `#require digit`.

<https://github.com/TG9541/stm8ef/issues/153>?

<https://github.com/TG9541/stm8ef/issues/152#issuecomment-360483633>, or mute
the thread

VK6TT, 2018-01-27:
Hi Thomas,

if I look at forth.asm  I see in line https://github.com/TG9541/stm8ef/blob/c420dcf9fa5c9dfc190f08574680cf07cad76a7f/forth.asm#L2178 the definition of digit. eg
;       DIGIT   ( u -- c )      ( TOS STM8: -- Y,Z,N )

I had assumed that all the words in the file forth.asm, and also mentioned in words.md, would simply be there in the W1209-FD binary distribution. It's no big deal, in the case of the word digit, to define my own version of word if it's not in the vocabulary of the binary distribution.

But given that I couldn't understand what the following line in the file \stm8ef-bin.V2.2.21pre2\out\W1209-FD\target\digit meant I'm hoping I don't have to do this for other words.
: DIGIT [ $CC C, $88F1 , OVERT

Kind regards
Richard

TG9541, 2018-01-27:
Hi Richard, 

first of all, thanks for pointing out that the whole Alias stuff needs  a better explanation. 

I updated the [Using Alias Words](https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Alias-Words) Wiki in the hope of making this more accessible. 

Could you please check if the concept is sufficiently understandable? I would also appreciate a recommendation on where to point users to the "missing words", and hear your opinion on e4thcom and codeload.py.

Kind regards
Thomas

VK6TT, 2018-01-28:
Hi Thomas,
 
I'll read through this over the next few days and comment as I notice
something. But to point my future comments in context I'd point out that I
need very little flash available for my applications. I just can't imagine
writing that many new forth words for a W1209 board for one of my projects.
So while applaud the efforts yourself and others go to to minimise the size
of the forth binary, it comes at a cost for less knowledgeable users like
myself in the time it takes to work out what is happening. 
 
My suspicion is that Amforth suffered when the 'recognizer' concept was
introduced. It obviously meant something very powerful to the architects.
But when I struck problems there was no material or help that could be used
to start out on a learning curve. So while my AVR assembly background
allowed me to work out how Amforth was internally constructed and compiled,
the issue of getting recognisers to work left me confused. In the end I
wrote my own solution but I suspect it could have been so much better if I
had understood this recogniser implementation.
 
I'm going to need more time than I have today to read Using Alias Words.
It's on my list for tomorrow.
 
I don't use e4thcom and coadload.py because I haven't found the motivation
and hardware for a Linux box. Maybe later this year. 
 
I simply like forth. All the other languages to me are at best a necessary
evil!
 
Kind regards
Richard

  _____  

(#152)

pbyAckcks5tOx4qgaJpZM4RstVv> the thread.
<https://github.com/notifications/beacon/AT7ya8c67jr8qhcpu3UqVTy7LqBSIt7Kks5
tOx4qgaJpZM4RstVv.gif>

TG9541, 2018-01-28:
I tried adding full dictionary as aliases (it was actually a 5 minutes hack) - the binaries are here: [W1209-fulldict.zip](https://github.com/TG9541/stm8ef/files/1671568/W1209-fulldict.zip)

The aliases require 550 bytes memory, but they can be removed by `RESET` as shown below. By using `PERSIST` from the `/lib` folder you can make them permanent. Of course, creating subset is also easy.

```
words
 CP NUMBER? do$ dm+ QUIT $"| docon $COMPILE CUPPER DOXCODE _TYPE PRESET (+loop) QUERY ?STACK ?UNIQUE doVAR LOCKF NAME? PAD str TAP DIGIT? EXTRACT WORD $," PACK$ .OK 1 0 dodoes EVAL find ?branch COMPILE? branch PARSE >CHAR .ID ABORT TOKEN DIGIT ^H SAME? -1 $INTERPRET kTAP ULOCKF ACCEPT ."| $,n IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok
reset
STM8eForth 2.2.21 ok
words
 IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok
```

VK6TT, 2018-01-30:
Hi Thomas,
 
I got my head around what was achieved with the Alias words. As a result I
made a few edits to the wiki which I hope you find useful. Fortunately I
doubt I will ever run out of flash now.
 
Kind regards
Richard
  _____  

(#152)

<https://github.com/notifications/beacon/AT7ya8c67jr8qhcpu3UqVTy7LqBSIt7Kks5

TG9541, 2018-01-30:
Hi Richards,

I had a look at the changes - I like a lot that you write about ALIAS from the user perspective!

Thanks!

closed 2018-02-05 by TG9541