TG9541/stm8ef #153: NVM appears to be broken

VK6TT, 2018-01-25:
When I load my forth definitions via the console on the W1209-FD they are not being saved in the flash.

eg after a clean flash of W1209-FD binary

STM8eForth 2.2.21 ok
words
 IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok
  ok

\ now I define "a"
 NVM ok
 : a ." test" cr ; ok
 atest
 ok
 words
 a IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok

\ now I power off and back on
STM8eForth 2.2.21 ok
words
 IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok

\ and a is missing.

Maybe my board has developed a fault?

VK6TT, 2018-01-25:
It appears to be a hardware fault. Same behaviour on 2.2.13. Closed and new board required.

VK6TT, 2018-01-25:
Nope. Definitely something amiss. With a new board and 2.2.21 flashed I can confirm the flash has been written to by using :
' iret 2000 dump 
And my new word vectored into boot works as expected.

However, all my words are there but they are not shown when I type "words" at the prompt. Since I can only use those words from the terminal are those shown by the command "words" I have no way of manipulating the search order or vocabularies. 

These words are, to void repeating them here, what was shown in my earlier message.

TG9541, 2018-01-25:
Isn't switching back to RAM missing?

TG9541, 2018-01-26:
Hi Richard,
just to be totally sure:

```
\ now I define "a"
NVM ok
: a ." test" cr ; ok
atest
ok
words
a IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE [COMPILE] LITERAL CALL, C, , ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACES SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG / /MOD M * UM / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = DNEGATE 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX 'BOOT BKEY OUT! COLD ok
```

Now, here I would have expected `RAM` before the restart. However, you do a reset without it:

```
\ now I power off and back on
STM8eForth 2.2.21 ok
words
IRET SAVEC RESET RAM NVM LOCK ULOCK ADC@ ADC! ?KEYB P7S E7S WORDS .S DUMP I
```
The result is that the new words won't be linked. They remain in the flash memory, and they can be used of the address is known (e.g. by writing the address of a word to `'BOOT`) , but they will be overwritten by new word definitions!

Could you please run the following:

```
RESET
NVM
: a ;
RAM
COLD   \ or do a power cycle
WORDS  \ a should be there
```

/Thomas

VK6TT, 2018-01-27:
Hi Thomas
 
thanks for your help. I hadn't realised that the word RAM must be called
before a power cycle or reset. It was not something I recall ever doing in
the past so I'm unsure if that's an omission on my part or something that
has changed.
 
I can happily confirm that if I type the commands below everything works as
expected.
 
Once again, many thanks.
 
Regards
Richard

  _____  

BhGvDdCks5tOiUKgaJpZM4RstvJ> the thread.
<https://github.com/notifications/beacon/AT7ya5mto4i7SmmLA7rZAKvr2RPIM4nZks5
tOiUKgaJpZM4RstvJ.gif>

TG9541, 2018-01-27:
Hi Richard,

the behavior of `NVM` and `RAM` is there by design. It's been done that way in order not to "wear out" the Flash memory cells used for initializing memory pointers like `CONTEXT` and `LAST` (the code is [here](https://github.com/TG9541/stm8ef/blob/c420dcf9fa5c9dfc190f08574680cf07cad76a7f/forth.asm#L4422)). Sometimes `RAM` is hidden in constructs like the `#include STARTTEMP ... TARGET ... ENDTEMP`. 

Maybe one of the following solutions would be more user friendly (because more obvious):

* introduce a word `KEEP` (similar to `PERSIST`, which, by the way, copies the initialization values to a second list that's used by `RESET`)
* the initialization values now updated by `RAM` could also be stored in EEPROM when completing a column definition in `NVM` mode

Your input is, as always, highly appreciated!
/Thomas

PS: please also have a look at issue #152 :-)

VK6TT, 2018-01-27:
Hi Thomas,
 
maybe all that is needed is an addition to the docs to make it clear that
the user must execute RAM before a power cycle or cold restart to keep the
new NVM words available. Perhaps in the readme.md where the text is:
 
Words `NVM` and `RAM` switch between volatile (RAM) and non volatile (NVM)
modes
Add a new line "REMEMBER execute 'RAM' before a power recycle or executing
COLD if you want the words added to NVM to be available in future terminal
sessions."
 
My approach would simply be to remember this requirement. Another word would
still be something I have to remember to execute and I already have a word
that fixes this issue called RAM. From my limited understanding of what is
happening KEEP would therefore be unnecessary.
 
I understand the concern over wearing out the flash since 100 cycles could
be consumed very quickly. I assume the cycle limit prevents us, when
compiling in NVM, from redefining the word ';' to also call RAM then NVM
again to remove the requirement to manually execute RAM prior to a power
cycle / cold restart. 
 
Keep up the good work though. I'm having a blast playing with these W1209
boards and I'm about to start putting the bare chip into my projects when I
don't need the display and buttons. eg I have a 1296MHz beacon ready to
construct which uses a STM8S003F3 to load up the PLL in the RF synthesiser
then gate the transmission in morse code to satisfy the identification
requirements of my ham license. I will be writing this up in due course.
 
Kind regards
Richard
 
ps I'll respond separately about #152. 
 
 

  _____  

<https://github.com/TG9541/stm8ef/blob/c420dcf9fa5c9dfc190f08574680cf07cad76
a7f/forth.asm#L4422> ). Sometimes RAM is hidden in constructs like the

<https://github.com/TG9541/stm8ef/issues/152>  :-)

-
hKbg4sBks5tOtBKgaJpZM4RstvJ> the thread.
<https://github.com/notifications/beacon/AT7ya6XOuKYjlzYrH8cnCLn_M1RfBQwhks5
tOtBKgaJpZM4RstvJ.gif>

TG9541, 2018-01-27:
Hi Richard, 

thanks for the encouragement, and for the suggestion! 

I added the line you proposed to readme.md, and I'm now thinking of adding a FAQ to the Wiki (some questions were indeed recurrent).

Regarding the number of write cycles: I never experienced the consequences of exceeding the STM8S00x limit of 100 cycles. Some of my STM8S003F3 boards were re-flashed much more often than that, but I never tried to operate those at "55°C for 20 years". My recommendation is therefore to use some µC boards for development, and others for deployment. It's likely that the low spec is a tool for lowering the quality assurance costs in batches of low-cost devices, and maybe ST even package STM8S103 production batches that may be slightly out of spec (and may thus not qualify for "55°C for 20 years after 10000 write cycles" ) as the low cost variant.

Kind regards,
Thomas

VK6TT, 2018-01-30:
Hi Thomas,
 
an alarming development today. I had a collection of W1209 boards I was
flashing since I had the hardware all set up. Imagine my surprise when I
discovered approx. 1/3 could not be flashed. Close examination revealed the
STM8 chip has been replaced with something made by Nuvoton. Same pin out but
it does not talk via SWIM. It is probably a masked ROM.
 
I tried updating the wiki to alert people but the real issue is the vendors
have/or pretend not to have and idea about the difference between a ST and
Nuvoton chip.
 
Fortunately I can swap the dud chip out myself but many people are going to
be disappointed. And I also flashed about 15 mindev boards I had accumulated
without issue.
 
Have you had any similar experience with non-flashing STM8's?
 
Regards
Rihcard

TG9541, 2018-01-30:
Hi Richard, 

I knew that Nuveton offers an 8051-like µCs which are pin compatible with STM8S003F3P6, but I certainly didn't see that coming. Lately, someone got a STM8S003 based W1209 with a 5k reference resistor (instead of 20k). The W1209 platform is now getting unstable, and people should know that.

STM8 eForth has a basis as long as (less popular) devices like W1219, or W1401, are cheap enough.

Thermostat boards are rarely advertise to be build with a specific µC. However, I don't think that breakout boards are going to have "fake" chips (except maybe STM8S003F3 chips labeled as STM8S103F3).

zcsahok, 2018-02-04:
FYI I have a W1209 similar to the one in Edit2. It can be flashed, but it has a common anode LED display (p.n. TUSO-02803-U) as opposed to the common cathode display on the original board. Pin assignment is also a bit different and the result is garbled display. Did not test the serial IF, though.

TG9541, 2018-02-05:
Zoltan, thanks, that's really interesting!

Now we have 4 variants (in the order of appearance):

* the garden variety W1209 as described in the Wiki
* one with improved sensor sensitivity in the 40°C range (5k reference resistor)
* and one with a Nuveton 8051-style µC
* one with a common anode LED display (yours)

Is your common-anode variant branded in any way (e.g. TENSTAR)?
Do you know of a way to identify it before buying?

I guess I'll have to think for a solution to this problem. If you like you can flash the W1209-FD binary, and try to connect through the PC4/PC5 key pins. If that works, we'll have to test the hypothesis that the PCB is the same. In that case patching the binary through the Forth console might be sufficient.

TG9541, 2018-02-05:
@zcsahok described the hardware changes quite well. Of course the PCB had to be changed to accommodate common-anode LED displays https://github.com/mister-grumbler/w1209-firmware/issues/19

Based on your hardware description, here is a version for CA displays: [W1209-FD-CA.zip](https://github.com/TG9541/stm8ef/files/1696758/W1209-FD-CA.zip).

Please let me know if it works!

zcsahok, 2018-02-05:
It's definitely better: now it shows

```
| |      |      |
---    ---    ---
  |    |      | |
       ---
```

Swapping B and F would give the expected 4th.
I does not react on key presses, but maybe that's intentional.

I'll upload HW details tomorrow.

TG9541, 2018-02-05:
Oops, of course! 

Here is the new file: [W1209-FD-CA.zip](https://github.com/TG9541/stm8ef/files/1696885/W1209-FD-CA.zip)

Yes, not reacting on keys is intentional as this was the "naked" Forth binary (you need to connect a terminal to program it - the compiler actually runs on the board). 

I made a thermostat file for you - that one should work just fine: [W1209-CA.zip](https://github.com/TG9541/stm8ef/files/1696909/W1209-CA.zip)

zcsahok, 2018-02-05:
Wow! `W1209-CA.ihx` works as expected.

TG9541, 2018-02-05:
Great :-) You'll find that it isn't functionally equivalent to the orignal firmware. If you connect a terminal you can give the logger a try!

zcsahok, 2018-02-06:
I can confirm that serial console works also fine. No issues so far.

Here is the board I have: [Link](https://www.aliexpress.com/item/50-110-Digital-thermostat-Temperature-Control-Switch-12V-waterproof-sensor/32279039523.html). Its "brand name" is _diymore_ and on the back of the board it shows "XH-W1209". It has a uC marked as 8S003F3P6.

[image]
(programming connector was populated by me)

The two marked vias are placed differently to the "normal" board.
[image]

Here's a normal board for reference (pic from net):
[image]

TG9541, 2018-02-06:
The traces and vias you highlighted is a way to tell boards apart. The air gap isolation slots are also different. The pictures on the vendor's page show these details, too.
The µC markings are standard for STM8S003F3P6.

zcsahok, 2018-02-07:
Patch is fine as long it's easy to apply. I mean it shall be somehow automatized.

As far I could see in this case the only affected place is the `LED_MPX` routine (same pins are used, no change in `BOARDINIT`). Can one ensure that it's always compiled to the same address? Could it be linked from of library? I don't know the possibilities of the tooling.

(btw looking at the code I had the impression the at the fourth phase of multiplexing the display stays blank. Did I miss something?)

TG9541, 2018-02-09:
Out of the cheapest offers, only [this one with a green LED display](https://www.aliexpress.com/item/10PCS-NEW-IC-NSC-LM3914-DIP-18/1852521499.html) looks like it were of the original kind!

The W1209 [here](https://www.aliexpress.com/item/W1209-DC-12V-LED-Digital-Thermostat-Temperature-Control-Thermometer-Thermo-Controller-Switch-Module-NTC-Sensor/32804123816.html), [here](https://www.aliexpress.com/item/W1209-DC-12V-LED-Digital-Thermostat-Temperature-Control-Thermometer-Thermo-Controller-Switch-Module-Waterproof-NTC-Sensor/32801327249.html), [here](https://www.aliexpress.com/item/W1209-Digital-LED-Heat-Cool-Temp-Thermostat-Temperature-Control-Switch-Module-On-Off-Controller-Board-DC/32793202333.html) and [here](https://www.aliexpress.com/item/50-110-Digital-thermostat-Temperature-Control-Switch-12V-waterproof-sensor/32258765300.html) look like they were of the "common anode" type.

The ones [here](https://www.aliexpress.com/item/1pcs-W1209-Mini-thermostat-Temperature-controller-Incubation-thermostat-temperature-control-switch/32842583291.html), [here](https://www.aliexpress.com/item/W1209-Digital-LED-Heat-Cool-Temp-Thermostat-Temperature-Control-Switch-Module-On-Off-Controller-Board-DC/32844542093.html), and [here](https://www.aliexpress.com/item/1pcs-W1209-Mini-thermostat-Temperature-controller-Incubation-thermostat-temperature-control-switch/32840996598.html) are (likely) also (likely) non-standard.

VK6TT, 2018-02-11:
I had three of the boards as marked below. (Or again here for certainty:
https://www.aliexpress.com/item/1pcs-W1209-Mini-thermostat-Temperature-contr
oller-Incubation-thermostat-temperature-control-switch/32842583291.html )
 
I've just flashed them and they appear to work fine. 4th was displayed but I
didn't get a chance to check out the serial comms.
 
Maybe we need a library of pics on the wiki to help identify useful boards?
I started to write a page but got stuck with images. Seems I can't get git
to run on my legacy hardware so I'm putting this on the backburner. One of
these days I'll stump up for another PC. But not today.
 
 

  _____  

(#153)

<https://www.aliexpress.com/item/10PCS-NEW-IC-NSC-LM3914-DIP-18/1852521499.h
tml>  looks like it were of the original kind!

The W1209 here
<https://www.aliexpress.com/item/W1209-DC-12V-LED-Digital-Thermostat-Tempera
ture-Control-Thermometer-Thermo-Controller-Switch-Module-NTC-Sensor/32804123
816.html> , here
<https://www.aliexpress.com/item/W1209-DC-12V-LED-Digital-Thermostat-Tempera
ture-Control-Thermometer-Thermo-Controller-Switch-Module-Waterproof-NTC-Sens
or/32801327249.html> , here
<https://www.aliexpress.com/item/W1209-Digital-LED-Heat-Cool-Temp-Thermostat
-Temperature-Control-Switch-Module-On-Off-Controller-Board-DC/32793202333.ht
ml>  and here
<https://www.aliexpress.com/item/50-110-Digital-thermostat-Temperature-Contr
ol-Switch-12V-waterproof-sensor/32258765300.html>  look like they were of

<https://www.aliexpress.com/item/1pcs-W1209-Mini-thermostat-Temperature-cont
roller-Incubation-thermostat-temperature-control-switch/32842583291.html> ,
<https://www.aliexpress.com/item/W1209-Digital-LED-Heat-Cool-Temp-Thermostat
-Temperature-Control-Switch-Module-On-Off-Controller-Board-DC/32844542093.ht
ml> , and here
<https://www.aliexpress.com/item/1pcs-W1209-Mini-thermostat-Temperature-cont
roller-Incubation-thermostat-temperature-control-switch/32840996598.html>

HVk6r2jks5tTJ2agaJpZM4RstvJ> the thread.
<https://github.com/notifications/beacon/AT7ya-pUaKYitSaEUXu-C46oMl-A4w4Aks5
tTJ2agaJpZM4RstvJ.gif>

TG9541, 2018-02-11:
Hi Richard,
thanks for the info! 

The vendor's picture looks like one of the 3rd kind in my last comment (one of those that I ordered for inspection).

It's a very good idea to provide an identification aid in the Wiki. There is no need to install git for uploading pictures: these can be dragged & dropped into comments of the "wiki photo non-issue" #40.

TG9541, 2018-02-11:
One open question is whether the "communication through the sensor input" variant should be removed. The "W1209-FD" variant allows using mini clamps while the first "W1209" requires removing the capacitor from the sensor input which is a rather "invasive" procedure.

VK6TT, 2018-02-11:
I support removing the sensor input version. While I hate mini clamps, once
you tack on some leads they work much better. And this allows you to
debug/develop while reading through the sensor input. W1209-FD is the way to
go in my opinion.

  _____  

(#153)

IcWRllmks5tTrZHgaJpZM4RstvJ> the thread.
<https://github.com/notifications/beacon/AT7ya7uqTTetjwzLCP93ob5cD7ZE3VGmks5
tTrZHgaJpZM4RstvJ.gif>

TG9541, 2018-02-11:
OK, that simplifies things when variants are to be supported. 

We now have knowledge of the following variants: 

1. µC variants (STM8S003F3P6, and 8051).
2. reference resistor variants (20k and 5k), and 
3. LED variants (CC, CA), 

For obvious reasons STM8 eForth won't support the 805, and it will be up to the users to avoid them. The resistor variants can must be handled in the application software (or fixed by replacing resistors). The variants that matter are CC and CA LED displays.

zcsahok, 2018-02-11:
How about this approach for the variants

- board name would be `<baseboard>-<var1>-...-<varN>`
- includes come from `<baseboard>` dir, as now
- tooling dynamically creates `variant.h` in `<baseboard>` dir that contains just defines for var1..N that control the included features,
e.g.

```
.define var1
...
.define varN
```
- board output files go to the corresponding dir in `out`, as now. 

This would give quite a flexibility, even to define different baud rate. (say 1200 for the sensor pin so that cap can stay in circuit)
And it would of course allow a `W1209-CA` board.

TG9541, 2018-02-11:
Zoltan, I also had ideas along these lines but creating variants is a messy business ;-)

The following things need to be considered:

* variant points in multiple source files (e.g. boardcore.inc)
* variant files and subdirectories in the out folder(<board>.ihx, forth.rst, target/)

What do you think of a two level configuration folder structure? Generating files would be possible but it makes the debugging process even more complicated (I had to work around limitations of the outdated and heavily patched version of sdasstm8 that comes with SDCC).

It's Forth though. Some things can be patched easily on-line (e.g. the baud rate). As think about it... the baud rate timer reload value should have been exposed long ago ;-)

TG9541, 2018-02-17:
Thanks a lot @VK6TT for providing [great docs in the Wiki](https://github.com/TG9541/stm8ef/wiki/W1209---Identifying-'compliant'-boards) and to @zcsahok for providing code for the [W1209-CA](https://github.com/TG9541/stm8ef/pull/157) (Common Anode) variant!

closed 2018-02-17 by TG9541

VK6TT, 2018-02-18:
Your welcome. I'm happy to update as necessary. But the best outcome is newcomers not being caught out. I wrote to vendors asking if the w1209 board had the St Chip. All I got was responses that were unhelpful. I might try a different approach when time permits.
Regards
Richard

TG9541, 2018-03-05:
I've now got a W1209 with a Nuveton chip on it of my own, and the original PCB layout. However, I opened a dispute on the basis of misleading pictures, e.g. components on the board, 30cm instead of 50cm sensor cable, and "wrong material" (which it clearly is). I've got a refund, but I couldn't test the 3rd W1209 PCB layout variant as [planned](https://github.com/TG9541/stm8ef/issues/153#issuecomment-364736057).

g7bbg, 2019-10-21:
Hi I've found what apperas to be yet another variant , it has a processor with no markings on the top, C5 missing but on removing the chip ismaked E7573071F on the bottom. I've tried connectint toa Nuvolink programmer but it's not reconised . So I assume it's a ROM based chip of some sort.

TG9541, 2019-10-23:
@g7bbg that's an interesting bit of information! It's possible that there now is more than one "second source" of "STM8S Low Density TSSOP look-alike" chips. Pin-compatibility as a product and marketing strategy often works (if you can produce cheaper).

g7bbg, 2019-10-26:
Hi, it looks like I've found another variation of the Common Anode boards (bought on Ebay in 2016) 
 which has totally different pin mappings, and a different layout please see the
 Readme.txt in the attached zipfile.

[W1209-CA-V2.zip](https://github.com/TG9541/stm8ef/files/3775299/W1209-CA-V2.zip)