TG9541/stm8ef #156: New WS1209 = Nuvoton N76E003AT20 (8051)

e0929364, 2018-02-16:
When I wanted to program a recently bought WS1209, i noticed unlocking the device didn't work even under STVP.   

The board and MCU look identical, only on close inspection you will notice it's not a STM8S003F3P6 buta a Nuvoton N76E003AT20.   The N76E003AT20 is pinout-compatible to the STM8S003F3P6, has more Flash, the same amount of SRAM, but obviously not code-compatible.   There is no dedicated EEPROM, only IAP Flash.

Really sucks :-1:

TG9541, 2018-02-16:
Please check #153 - you'll find some more on known clones. Also please compare your board to the ones [here](https://hackaday.io/project/26258/log/86362-which-w1209-boards-not-to-buy-if-you-want-to-run-the-code-here).

e0929364, 2018-02-16:
Thanks, but the problem is, my board even had a STM8S003F3 on the photo, but I was sent a Nuvoton version.  You can't be sure by the seller's product photo. I will make  a photo of the bad boards.    All my good STM8 boards have "HW-557" written on them.

The bad Nuvoton boards came from here:
https://www.aliexpress.com/item/50-110-Digital-thermostat-Temperature-Control-Switch-12V-waterproof-sensor/32279039523.html  (Store name: ElectronicFans)
https://www.aliexpress.com/item/50-110-Digital-thermostat-Temperature-Control-Switch-12V-waterproof-sensor/32258765300.html? (Store name: ModuleFans)

The good STM8 board (weak green LED, but at least STM8) came from here:
https://www.aliexpress.com/item/W1209-DC-12V-Green-LED-Digital-Thermostat-Temperature-Control-Thermometer-Thermo-Controller-Switch-Module-Waterproof-NTC/32836219816.html
(Store name: Worldchips)

So far I have 6 useless boards, for a total of 7 EUR wasted.   I think I will use STMS103F3P6 breakout boards as donor and replace MCUs.

e0929364, 2018-02-16:
I now soldered an STM8S103F3P6 from the common 0.50 EUR breakout boards and a VCAP capacitor to the Nuvoton board and it works 👍

TG9541, 2018-02-17:
That's really interesting, and of course frustrating.

The 1st photo shows the notorious common anode LED variant, and the 2nd looks like the "original PCB" (most likely also a knock-off). Your PCB is of the same type.

I guess the best way to deal with this is to give the sellers less-than-favorable ratings. Transparency is crucial, and it should hurt to sell something else than they advertise.

STM8S003F3P6 chips used to be available for below $0.25@10pcs. I think that the market has changed:

* the best offer I found is $0.35@10pcs, the new normal is at $0.50@10pcs, and the best price I found for 1000pcs is twice as much as it used to be
* STM8S103F3P6 are often cheaper than STM8S003F3P6
* sellers [try to promote](https://www.aliexpress.com/item/10PCS-N76E003AT20-instead-of-STM8S003F3P6-new-original/32846394292.html) the N76E003AT20 as a STM8S003F3P6 replacement at hobby quantities!

I think that for now the W1209 "platform" has received a blow. Possibly it will recover after the apparent shortage of STM8S003F3P6 chips has ended. However, it might also well be that ST has stopped flooding the Chinese market with chips, and they're trying to get better prices. Which would of course be exactly what Nuvoton waited for.

closed 2018-02-27 by TG9541

g7wbd, 2021-05-02:
e0929364  are you still about I am facing similar problems ?