TG9541/stm8ef #158: W1209 low pass filter behaving erratically

zcsahok, 2018-02-18:
I have a strange issue with the W1209 thermostat code.
Not sure if it's bug or just a missing step from my side, so this is just a question.

Summary: after uploading W1209 code display shows "random" values but entering a digit on the serial console brings back "correct" temperature display.

Here are the steps I did.

* created `W1209-CA.ihx` the official way, using `docker-sdcc`.
* flashed it: 4th on display, all normal on serial console.
* made a serial upload of thermostat code from W1209 repo.
* run `cold` (or bounced power, same effect)
* seemingly random (positive and negative) numbers are shown in quick succession, relay led flashes.
* now: if on console a digit value is entered then display "calms down" and converges to a value that looks correct.
* this (working) state remains even if the digit is deleted using ^H
* different digits cause minor differences in the value shown (so it's only "correct")
* other characters cause other effects. e.g. on lower case letters -9.9 (or -10.0) is shown.

I have checked the internal states

```
1: cold ok
2: lpf.adc ? 11584 ok
3: lpf.tem ? -3169 ok
4: 6 . lpf.tem ? 6 7199 ok
5: lpf.tem2 ? -198 ok
6: 6 . lpf.tem2 ? 6 448 ok
7: 6 . lpf.adc ? 6 11715 ok
```

The `6 .` is just to have a digit in the first position of the serial buffer.
ADC does work normally, even though on line 2 the display actually shows -9.9.
The output of @inter+LPF (`lpf.tem`) is definitely wrong on line 3.
As soon as the first char is a digit (line 4) the display and `lpf.tem` are normal.
On line 5 it's again got broken just after entering the first char ('l').
Lines 6 and 7 are normal again due to the 6 on first position.

It seems that the first char of the serial buffer interferes with the interpolation
and/or low pass filtering.

I made a double check with the fw from issue #153: it's a fully working thermostat
and after running `reset` I uploaded the code the same way as above.
In this case all was OK, display values quickly converged to the correct temp reading.

So there must be something with my build. But what?

zcsahok, 2018-02-18:
Thanks for the good tip! Magic has happened. Now I followed more closely the Makefile of `W1209` repo and  executed the `target` step that creates the symlinks. After uploading all worked fine, no more strange effects.

It seems upload found wrong word definitions when `target` was missing.

This was with my private build, I'll give the official build a try also.

The `target` goal must be executed explicitly. Wouldn't it make sense to add it as a dependency to load/simload? 
The other file symlinked (FORTH.efr) is missing from my zip file. Not sure what it is for.

zcsahok, 2018-02-18:
It works also with the official build. The commands were (after changing the board name)

```
make clean
make target
make
make load
```

TG9541, 2018-02-18:
Hi Zoltan, thanks for the feedback - it's always good when problems find a simple solution!

I'll have to think of a way of sanity-checking aliases. Using the wrong target folder can lead to unpleasant effects. 

The `target` goal is a bit of a hack: it should be possible to check in a Makefile whether the symlinks are correct, but I couldn't find a solution for this use case.

zcsahok, 2018-02-20:
Thanks for the great project and your support!

closed 2018-02-20 by zcsahok

TG9541, 2018-02-20:
Thanks for your contribution to making this project great!

TG9541, 2018-03-05:
Using the right set of auto-generated aliases became a lot easier in STM8 eForth 2.2.21: as of 6f09234 the TG9541/stm8ef Makefile target `main.ihx` maintains the ./target symlink for e4thcom and codeload.py.