TG9541/stm8ef #159: MINDEV strange bliky example bahaviour

4yerba, 2018-02-19:
I am running stm8ef ver 2.2.19 on the MINDEV board. Just starting out with forth. When running blinky I came across this strange behaviour.
The code :
: blinky TIM $40 AND 0= OUT! ;
' blinky BG !
runs as expected. 
But the modified code (just an example, have found many more values behaving strangely):
: blinky TIM $24 AND 0= OUT! ;
' blinky BG !
does not. It flashes modulated pulses (blinks within blink). $20 value is ok again as is $4000 as well. Is this to  be expected or qualifies as an issue ?

TG9541, 2018-02-19:
Hi @4yerba ,
The behavior you observe is to be expected. The value you selected produce a 320ms period modulated by a 40ms period. 
If you want to change the frequency try the following:

```
VARIABLE interval
: blinky TIM interval @ 2DUP 2* MOD ROT ROT MOD = OUT! ;
50 interval !
' blinky BG !
```
The period is now `10ms * interval`, and you can change it by setting the variable interval (e.g. `100 interval !` results in 1Hz pulses). Unless `interval` is a power of 2 (2, 4, 8, 16, ...) there will be an odd pulse once in about 5.5 minutes (TIM returns a 16 bit value).

4yerba, 2018-02-19:
Hi Thomas,
Thank you. That was super quick. Now, your code works like a charm (and that indeed was to be expected). It is so much better to actually understand the example code than just to play with it. 
Forth is fun, forth is rewarding and forth is addictive. And the community (that would be you this time) is superb. I am yet to discover more stm8eforth goodies very soon. You got me truly hooked on.

closed 2018-02-19 by 4yerba

TG9541, 2018-02-20:
Thanks for the encouragement! 

Forth is indeed fun to play with - if you get stuck don't hesitate to open  an issue :-)