TG9541/stm8ef #180: Stack Underflow with SPI library

VK6TT, 2018-04-03:
This could be a product of my own shortcomings but I found the following today when trying to use the SPI functionality.

NVM
 : SPI_CR1    $5200 ; IMMEDIATE \ ($00)
 : SPI_CR2    $5201 ; IMMEDIATE \ ($00)

: SPIon ( -- )
  $3C SPI_CR1 C! 
  $01 SPI_CR2 C!  \ no NSS, FD, no CRC
  $7C SPI_CR1 C!  \ enable SPI 
;
RAM
Executing SPIon after a powercycle resulted in a stack underflow error. However, manually entering the three lines did not give an error.

STM8eForth 2.2.21 ok
spion  underflow?
 $14 SPI_CR1 C!  \ master, CLK/8, CPOL=CPHA=0, MSB first ok
   $01 SPI_CR2 C!  \ no NSS, FD, no CRC ok
   $54 SPI_CR1 C!  \ enable SPI  ok

Then I redefined SPIon as follows:
 spion  underflow?
  ok
   $14 SPI_CR1 C!  \ master, CLK/8, CPOL=CPHA=0, MSB first ok
   $01 SPI_CR2 C!  \ no NSS, FD, no CRC ok
   $54 SPI_CR1 C!  \ enable SPI  ok
SPIon  underflow?
  ok
 : spion reDef spion OK
   $14 SPI_CR1 C!  \ master, CLK/8, CPOL=CPHA=0, MSB first OK
   $01 SPI_CR2 C!  \ no NSS, FD, no CRC OK
   $54 SPI_CR1 C!  \ enable SPI  OK
 ; ok
 spion ok
STM8eForth 2.2.21 ok
spion  underflow?

However, when I reset the micro and used:
$5200 constant SPI_CR1
$5201 constant SPI_CR2
: SPIon ( -- ) OK
   $14 SPI_CR1 C!  \ master, CLK/8, CPOL=CPHA=0, MSB first OK
   $01 SPI_CR2 C!  \ no NSS, FD, no CRC OK
   $54 SPI_CR1 C!  \ enable SPI  OK
 ; ok

SPIon now returns no errors before or after a power cycle. eg

STM8eForth 2.2.21 ok
 spion ok

So I think it is something about the immediate nature of SPI_CR1 and SPI_CR2 that is the issue. However, as I stated at the outset, it could be my ignorance *grin*

And one other minor matter: what is the purpose of RAMMARK in the SPI.fs file? I couldn't understand why it was required. Perhaps this is related to the issue since I never used it.

Regards
Richard

TG9541, 2018-04-03:
Hi Richard, I'm glad you unearthed that outdated piece of hackery. Of course the IMMEDIADE isn't worth a bit of dingo poop, and I'm happy that it's been deleted from the master branch some time ago. CONSTANT has long been added to the STM8 eForth core (Manfred Mahlow nagged me into adding it, and the resulting binary size increase has long been over-compensated by several steps of optimization). 

RAMMARK used to store the current CP/LAST on the stack, and restore it afterwards. The idea is to use the RAM for "scaffolding" (a scratchpad for temporay dictionary elements). It turned out that doing that is a bad idea since eforthcom and codeload.py rely on testing for words by checking the console reaction (`?` means that a #required word needs to be defined). It first replaced by MARKER (which works in RAM and Flash, many markers possible), and WIPE (narrower use-case: like COLD, but preserves the stack).

By the way, if I can't nag you into using e4thcom, maybe I should point out that codeload.py can interpret #require and \res directives, and output a text file :-)

The bottom line is that SPI.fs needs to be re-factored before it's usable again.

VK6TT, 2018-04-04:
Hi Thomas,
 
thanks for the fast reply. I'm not sure how I managed to reference an old
file when I was coding my nRF24L01 stuff. Perhaps the same reason I soldered
an ESP8266 into my project instead of the nRF24L01 board. They both had 8
pins! So I know what Dingo poop is.
 
I looked at MARKER but the wording in \stm8ef-bin\lib\MARKER suggested it
only works for words in RAM. So I didn't use it. I use the word in other
Forths a lot so I'll give it a try.
 
Please nag away on e4thcom. I am looking for a small usb based distro to run
Linux so I can utilise e4thcom. It's just a time constraint that I haven't
proceeded. But python? Probably won't happen. I never say never but it would
be hard to devote the time for the learning curve just to avoid the #require
and \res directives.
 
But on a positive note I had what might be an enlightening thought after
finishing with my SPI coding. The nRF24L01 chip can send a maximum payload
of 32 bytes. I was kicking around how I might structure this to accommodate
data exchange and it occurred to me that this is actually a useable tib. So
my thinking is to send the payload from one device to the other, have the
recipient move the payload to the tib and interpret it.
 
Using a ficticious weather station example - to send a temperature reading I
might fill the payload with
    24 Temp ! 
assuming Temp is the relevant variable the recipient uses. If I had multiple
pieces of data the payload might look like
    24 Temp !  56 Humidity ! 12 WSpeed ! 
I wouldn't have to remember what order to send the data and subsets of data
could be sent without remembering the data format
e.g. it would be valid to send: 12 WSpeed ! 24 Temp ! 
 
I need to consider how I might send a string like "ENE" for a direction.
Perhaps a new word like $direction! which moves the 3 characters to an area
of ram defined with
    variable $direction 3 allot
I dismissed a word defined as ENE that achieved the same thing because if I
was sending direction in degrees I'd need 360 words. And I might want to
send text strings in the future so that new words could be promulgated onto
networked devices.
 
Anyway, I am rambling on. Thanks for your response and I need to crack on.
After I have soldered the correct device onto my board I will get back to
coding.
 
Kind regards
Richard
 
 
 
  _____  

<https://github.com/TG9541/stm8ef/issues/180#issuecomment-378333309> it on
GitHub, or mute
Zq5BLmLks5tk7KqgaJpZM4TE7K6> the thread.
<https://github.com/notifications/beacon/AT7ya9_0JCH0_d2G9bUZqdo9xG3By1Csks5
tk7KqgaJpZM4TE7K6.gif>

TG9541, 2018-04-04:
Hi Richard,

I had a first look at the SPI code before work, but I'd like to test it first, and also have a look at the LCD code. 

Your idea of injecting data into the TIB for interpreting data is nice. Something similar, if not nearly as sophisticated, has been demonstrated [here](http://blog.teknikelektrolinks.com/index.php/2017/10/12/berkenalan-dengan-board-stm8s103f3p6-yang-mungil/).

It might be an option to use I/O words instead of directly using the TIB.

TG9541, 2018-04-06:
@VK6TT Richard, I cleaned up the code a bit and added an example.

closed 2018-04-06 by TG9541

VK6TT, 2018-04-07:
Hi Thomas
 
thanks again for your efforts. I was about to send you my high level Forth
words which I used to successfully communicate with a device. However, your
email arrived while I was cleaning up the file so it's redundant now.
 
I couldn't find ]C! to use spi.fs but it's no trouble. I'm muddling away on
other matters and will use spi.fs down the track.
 
Many thanks
Richard

  _____  

pQe05lAks5tl8ctgaJpZM4TE7K6> the thread.
<https://github.com/notifications/beacon/AT7yawlno4qX1iwFJXR0XGX4m1Aw_-Fiks5
tl8ctgaJpZM4TE7K6.gif>

TG9541, 2018-04-07:
Hi Richard, 

sorry that it took so long - some muddling came in its way here as well. 

`]C!` is [here](https://github.com/TG9541/stm8ef/blob/master/lib/%5DC!) in the library, it should be no problem to add the definition to your code. I'm looking forward to NRF24L01 communication with a Forth console! 

What do you use on the other side for communication testing?

Best Regards,
Thomas

VK6TT, 2018-04-07:
Hi Thomas
 
at present I just have one breadboard up and running. I was able to test the
SPI function by:

*	

	reading the config register, changing the bit to power up,
re-reading config register to confirm it was powered up
*	

	powering down using same test methodology
*	

	read the 5 byte default address, change it to my own 5 byte address,
then read it back to verify it worked. 

All good so far. I'll solder another bareboard and nRF24L01 up today or
tomorrow and then try an on-air test.
 
The high level words I used to achieve this are shown below.
 
Kind regards
Richard
 
NVM
\ ******************************** SPI words
 
$5200 constant SPI_CR1
$5201 constant SPI_CR2
$5204 constant SPI_DR
$5203 constant SPI_SR
 

\ Init and enable SPI 
\ 76543210
\ 0         MSB first
\  0        SPI disabled
\   111     Baud rate control 111=fmaster/256
\      1    STM8 is master
\       0   SCK to 0 when idle
\        0  First clock transition is when data is read
\ 00111100  = $3C
 

\ Init and enable SPI 
: SPIon ( -- )
  $3C SPI_CR1 C! 
  $03 SPI_CR2 C!  \ no NSS, FD, no CRC, software slave, master
  $7C SPI_CR1 C!  \ enable SPI 
;
\ disable SPI 
: SPIoff ( -- )
  0 SPI_CR1 C!    \ disable SPI
;
 
: TXE?   \ Wait till TXempty flag set
   Begin
      SPI_SR c@
      $02 AND
   Until
   ;
: RXNE? \ wait till RX not empty flag set
   Begin
      SPI_SR c@
      $01 AND
   Until
   ;
: SPI ( c --- c )
   TXE?
   SPI_DR c!
   TXE?
   RXNE?
   SPI_DR c@
;
RAM

 
 
 

  _____  

Rq_gPi6ks5tmEsdgaJpZM4TE7K6> the thread.
<https://github.com/notifications/beacon/AT7ya2jP2arSxDvprhAxcO0QNCjdpUjZks5
tmEsdgaJpZM4TE7K6.gif>

TG9541, 2018-04-07:
Hi Richard,

great that it works! I'm currently lazily repairing the PD8544 code, while improving it a bit.

`]B!` and `]C!` can help to improve code density, and maybe I'll add inline assembly words for spin-locks or loops. However, Mandred Mahlow recently ported Lars Brinkhoff's STM8 assembler to 4ethcom, and that's a much more powerful approach. Since I'd like to be the command line tool codeload.py to be compatible with source code for e4thcom, I'll have to figure out how to run a Forth style assembler in Python.

/Thomas

commit 1ae82c26 (2018-04-06): Merge pull request #181 from TG9541/fix-spi

fixes #180 SPI register constants
commit 354b3531 (2018-04-06): fixes #180 SPI register constants