TG9541/stm8ef #197: build issues with gawk on OSX.

hydra, 2018-04-23:
gawk isn't in `/usr/bin/gawk` on OSX by default, `brew install gawk` installs it to `usr/local/bin/gawk` so the scripts don't work.

TG9541, 2018-04-23:
An interesting  problem.

I tried using / the following approaches, without success:
* use `#!/usr/bin/env gawk -f` (env doesn't like -f), and
* a [polyglott](https://unix.stackexchange.com/questions/97141/distributing-a-script-should-i-use-bin-gawk-or-usr-bin-gawk-for-shebang?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa)  approach (the script hangs, or exec doesn't like `-a`)

Do you have any suggestions?

TG9541, 2018-04-23:
The problem is even more interesting: https://groups.google.com/forum/#!topic/comp.unix.shell/msq-6mTHJRo

hydra, 2018-04-23:
didnt realise the -f option wouldnt work on linux though.

perhaps look at some other projects on github and see what they do?

TG9541, 2018-04-23:
Awk is hardly the most popular scripting language these days ;-)
One option might be patching the scripts when they run on OSX. It wouldn't be a nuisance if done automatically.

TG9541, 2018-04-25:
@hydra I tried some more variants of the "polyglot" method, but I had no success passing in command line parameters and gawk options at the same time. Some remaining option might be passing options in a file, through STDIN, or with environment variables.

David-Gould, 2018-07-22:
Can you use awk instead or do you depend on gawk extensions?

TG9541, 2018-07-22:
I think I didn't use any non-POSIX gawk features.

By the way, there is an obvious way to approaching the OSX problem.

In the makefile's main.ihx target the following sidesteps the shell script problem: 
 
```
	awk -f tools/genalias.awk -v target="out/$(BOARD)/target/" out/$(BOARD)/forth.rst
	awk -f tools/genconst.awk -v target="out/$(BOARD)/target/" out/$(BOARD)/forth.rst
```

@hydra FYI

commit 55de25e7 (2018-07-22): fixes #197 use POSIX awk, don't rely on linux script processor path
commit 7a21e888 (2018-07-22): fixes #197 use POSIX awk, don't rely on linux script processor path
commit 7c96cfc3 (2018-07-22): Merge pull request #222 from TG9541/fix-197

fixes #197 use POSIX awk, don't rely on linux script processor path
closed 2018-07-22 by TG9541