TG9541/stm8ef #219: Licensing

bholroyd, 2018-06-07:
Sorry If this is the wrong place to ask but I was wanting to make use of https://github.com/TG9541/stm8ef/blob/master/inc/stm8ldevice.inc in a project. 
Now that doesn't include a licence, but the license file https://github.com/TG9541/stm8ef/blob/master/LICENSE.md says that some parts are public domain and some are permisively (gpl v2?) licensed.

I'm just wondering which you consider this to be?

Regards

closed 2018-06-08 by TG9541

bholroyd, 2018-06-08:
Ok thanks.
Thought that might be the case but didn't want to assume. :)

TG9541, 2018-06-09:
When I started this project the license conditions for the STM8EF base unclear, and now, since I'm not an OSS expert, it's difficult to get copyright texts consistent. I don't even know if GitHub license tagging works when one doesn't use an unchanged GitHub template.

VK6TT, 2018-06-09:
Hi Thomas
I'm no lawyer but for any license to be meaningful it has to be enforced. So if you are not going to enforce the license conditions, whatever they might be, then who would? It's a rhetorical question because in practice no one will. 
My recollection is that anyone who uses stm8ef does so at their own risk. So it is hard to imagine a scenario where any liability could fall back on you. You have made the disclaimer and any party using stm8ef takes all the risk that the software is fit for purpose.
I can't see any pleasure arising from this thread. How about everyone moving on? I'd rather spend time trying to get e4thcom working.
Regards
Richard

TG9541, 2018-06-09:
Hi Richard,
of course you're right - there is no way to enforce a strong FOSS license like the GPL in a project like this, but this isn't what the issue is about. It's about if someone using this software is likely to receive a letter by a lawyer, e.g. from ST, because their licensing conditions are all but friendly. A lot of people decided not to use the ST header files because of the unusual conditions they contain.