TG9541/stm8ef #220: incremental stm8ef upgrade

4yerba, 2018-06-11:
Just a thought, a question really. Since we now have a possibility to store words in eeprom (limited yes, but still) - would it be possible to perform incremental upgrade to the stm8ef itself, thus eliminating the need to replace the whole firmware via the flash programmer ? No need to do it all at once - just an incremental self upgrade would suffice. May be viable for larger (more flash/eeprom) micros only if possible at all.

TG9541, 2018-06-11:
Hi 4yerba,
replacing a Forth image from user code should be possible, but unless there is a built-in boot ROM, or enough free flash memory it's a bit tricky to implement, unless a binary format for the data bytes is used, and the STM8 UART is used for programming. Not interrupting the download process would be critical. 

For larger STM8 variants (e.g. 16K Flash) it's not a big problem (e.g. using a hexfile reader in Forth). One thing to keep in mind is that Flash memory is relatively slow, and a line delay, or some kind of handshake, is required.

4yerba, 2018-06-12:
Hi Thomas,
Thank you for the swift response. To be honest, I strongly suspected it will be possible. Tricky but doable. I just wanted the final confirmation from the expert before taking up the challenge. The old trusty stm8s discovery board I have laying around (dusted off for this purpose) seems like a good target for experimenting with the idea.

TG9541, 2018-06-12:
I'm curious how it works out. The diff tool kit will certainly be interesting. The most tricky thing will be identifying all target addresses in the word, and link them to the addresses in a different environment. It might be possible to use the linker for this task. In some cases relative addresses are used. Solutions for these might require "telescope jumps". 
I guess that patching only works up to a certain level, from where on replacing the complete image is a good alternative.
Some time ago I pondered the idea of using "token threaded code". That would make replacing core words trivial.

TG9541, 2018-06-13:
Bye the way, do you use e4thcom? If no, which one do you use?

4yerba, 2018-06-13:
Good points. The challenge is very tempting but when I'll think about it cool-headed I might reconsider. The result, even if successful, might not be worth the effort (lots of hours will be burned with this work). When it comes to the terminal program, I use e4thcom for linux and picocom for freebsd as there is no e4thcom freebsd port I'm aware of.

TG9541, 2018-06-13:
FreeBSD sounds interesting. In Picocom, using a script (e.g. codeload.py) instead of ascii-xfr might be a good solution but I've yet to understand how picocom hands over the COM device to an external program.

TG9541, 2018-06-17:
I'll be closing this for now. If someone decides to write The task to write an auto-updater later on, it can still be reopened :-)

closed 2018-06-17 by TG9541