TG9541/stm8ef #223: New STM8 boards available from ST

4yerba, 2018-08-25:
Hi,
This is just a short note - ST has released two brand new STM8 Nucleo boards - one with the stm8l152r8 chip and the second one with the stm8s208RB. Both of these can make excellent targets for stm8ef exeriments. Not ultra cheap but well worth the price, I think (~10 USD). For those who care - manufactured in China :).

TG9541, 2018-08-25:
Hi, nice boards. I tried to order a set from the ST site, but I gave up when I checked the shipping costs: $37. Do you know of a better source?

4yerba, 2018-08-25:
Yes, I do. Try mouser.de. I'll be ordering one myself sometime next week. Free shipping when you spend at least 50 EUR, which is not that difficult to reach once you brose their website ;). Have boght from them many times already with no issues at all.

TG9541, 2018-08-25:
That's very very nice of you, but it's likely that I'll be able to support the boards even without getting my hands on one of them: 

* STM8S high density devices, like STM8S208RB, are already supported (I soldered an [STM8S207RB](https://github.com/TG9541/stm8ef/tree/master/STM8S207RB) on a breakout board), and 
* @plumbum wrote support code for the [STM8L Discovery](https://github.com/TG9541/stm8ef/tree/master/STM8L-DISCOVERY) which is based on the STM8L152C6

In the meantime I've also started making my own boards - my next project is maybe an STM8S103K3 based control board, one of which will control the solar water heating on the roof of our home.

RandoSY, 2018-08-29:
Love to see  your new STM8S103K3 board when done?  Good luck on project!

TG9541, 2018-09-25:
I'll be closing this for now - please re-open the issue if the binaries mentioned above don't work.

closed 2018-09-25 by TG9541