TG9541/stm8ef #232: sstm8 with tools/simload.sh or socat

ivpri, 2018-10-24:
Hi, stm8ef in sstm8 simulator works perfectly for me using telnet connection, However, I need also #include feature there so I've tried simload.sh but it fails:

```
ivo@ip:~/Projects/forth/stm8ef/src$ tools/simload.sh W1209
simload.sh: run STM8EF in uCsim with breakpoint at HI
FLASH reset
FLASH write-iapsr 40
FLASH write-cr2r 00
FLASH write-ncr2r ff
FLASH set_mode 00
FLASH reset
FLASH write-iapsr 40
FLASH write-cr2r 00
FLASH set_mode 00
FLASH write-ncr2r ff
FLASH set_mode 00
load "out/W1209/W1209.ihx"
Loading from out/W1209/W1209.ihx
5231 words read from out/W1209/W1209.ihx
break 0x008EBD
Breakpoint 1 at 0x008ebd: call  0x8b79
simload.sh: inject UART code into RAM and continue execution
����uCsim 0.6-pre34, Copyright (C) 1997 Daniel Drotos.
uCsim comes with ABSOLUTELY NO WARRANTY; for details type `show w'.
This is free software, and you are welcome to redistribute it
under certain conditions; type `show c' for details.
3> download
:04006000100010205C
:1010000090AE680390CF5232350C523535100061e6
:0C1010005CF65C720F5230FBC75231815d
:101020004F720B523003C652314D2704AD024F436d
:08103000905F90975A5AFF816e
:00000001FF
run
56 words loaded
3> 3> Simulation started, PC=0x008ebd
local main.fs not found ...
... trying W1209/board.fs
simload.sh: transfer W1209/board.fs
Uploading W1209/board.fs
Uploading /home/ivo/Projects/forth/stm8ef/stm8ef-2.2.22/lib/PERSIST
TX: \ STM8EF dictionary management
TX: \ refer to github.com/TG9541/stm8ef/blob/master/LICENSE.md
TX: 
TX: \ Set RESET defaults to include newly defined NVM words
TX: : PERSIST ( -- )
error  :PRIT(-?

Error file W1209/board.fs line 1: could not upload file
>>>  #require PERSIST
```
So, next try was with simulated serial port using socat:
```
sstm8 -wgtS105 -Suart=2,port=10000 src/out/STM8S105K4/STM8S105K4.ihx
uCsim 0.6-pre34, Copyright (C) 1997 Daniel Drotos.
uCsim comes with ABSOLUTELY NO WARRANTY; for details type `show w'.
This is free software, and you are welcome to redistribute it
under certain conditions; type `show c' for details.
0> bind: Address already in use
FLASH reset
FLASH write-iapsr 40
FLASH write-cr2r 00
FLASH write-ncr2r ff
FLASH set_mode 00
FLASH reset
FLASH write-iapsr 40
FLASH write-cr2r 00
FLASH set_mode 00
FLASH write-ncr2r ff
FLASH set_mode 00
Loading from src/out/STM8S105K4/STM8S105K4.ihx
4793 words read from src/out/STM8S105K4/STM8S105K4.ihx
```
Redirect telnet to pty:
```
sudo socat pty,user=ivo,link=/dev/virtualcom0,rawer tcp:localhost:10000
```
However, the same issue here with both using e4thcom or e.g. echo words > /dev/virtualcom0
In all these cases each second character is dropped, when I type wwoorrddss, the word is accepted:
```
SIO : /dev/virtualcom0 open  hdl=3 B9600 8N1 full-duplex  
      Type '\ [Enter]' to close the Terminal

����uart[2] terminal display, press ^x to access control menu
 ok
wrs wrs?
wordsrddss
  IRET SAVEC RESET RAM NVM LOCK ULOCK WORDS .S DUMP IMMEDIATE ALLOT VARIABLE CONSTANT CREATE DOES> ] : ; OVERT ." $" ABORT" AFT REPEAT WHILE AHEAD ELSE THEN IF AGAIN UNTIL BEGIN +LOOP LOOP DO NEXT FOR COMPILE LITERAL CALL, C, , [COMPILE] ' hi CR [ NAME> \ ( .( ? . U. TYPE U.R .R SPACE KEY DECIMAL HEX <# SIGN HOLD #S # #> ERASE FILL CMOVE HERE COUNT +! DEPTH PICK 0= ABS NEGATE NOT 1+ 1- 2+ 2- 2* 2/ EXG */ */MOD M* * UM* / MOD /MOD M/MOD UM/MOD WITHIN MIN MAX < U< = 2DUP ROT ?DUP BG TIM BL OUT last '?KEY 'EMIT BASE - 0< OR AND XOR + UM+ I OVER SWAP DUP 2DROP DROP NIP >R R@ R> C! C@ ! @ B! 2C@ 2C! 2@ 2! EXIT EXECUTE LEAVE EMIT ?KEY TX! ?RX ADC@ ADC! OUT! COLD 'BOOT ok
```

Any Idea what's wrong?

TG9541, 2018-10-24:
I never tried using e4thcom through a uCsim telnet connection - the idea is good but the uCsim telnet interface is far from easy to deal with, and the author of uCsim didn't fully support the interactive scripting use case. When I worked on `simload.sh` I already had to work around a range of problems, and I guess that race conditions can be one of them. e4thcom might also contribute to problems since, as far as I know, recent versions have only been tested with serial interfaces.The only way i could make scripting work was with the Python Telnet client library. Maybe Python code can be used as a glue between uCsim and e4thcom?

TG9541, 2018-10-24:
On a second thought, did you try using the uCsim version [tg9541/sdcc-docker](https://hub.docker.com/r/tg9541/docker-sdcc/)?

TG9541, 2018-11-07:
I'm closing this for now. Please use uCsim 0.6-pre32 as shown [here](https://hub.docker.com/r/tg9541/docker-sdcc/~/dockerfile/) or raise an issue upstream (which I had tried).

closed 2018-11-07 by TG9541