TG9541/stm8ef #238: [Q] Modbus and push button on C0135

bademux, 2018-12-18:
Hello,
I'm looking for easy way control relay (220V load) with modbus and input pins (push buttons) on C0135 board.
As far as I understand I need to implement modbus and handling push buttons with GPIO? 
Is it possible\worth with stm8ef?

Thanks for great project.

TG9541, 2018-12-18:
Hi :-) 

There are plans to implement the MODBUS RTU protocol. My problem has been creating a test setup.

Is my understanding right that you're talking about a MODBUS endpoint based on the C0135 board and not a MODBUS master?

Otherwise, what's the role of Forth (e.g. local logic, fallback-behavior)?

bademux, 2018-12-18:
Thanks for your response!

Lets me describe the problem I want to solve:
```
Got: 
- Openhab Server  with USB-to-RS485 dobgle  (Modebus master plugin).
- 6x bunch of push buttons and twisted pair cables nearby.
- 6x 220V loads to be on\off
Goal:
- to be able to change the state of relay with Openhab Server and push button simultaiously, so the server can track curret state.
```
From what I know INT1-4 are read only by default on C0135 (I have no C0135 yet).
My question is how much should be done to solve my problem using C0135.
From my understanding it is: 
1. Modenbus slave
2. Handling INT1-4

TG9541, 2018-12-18:
I browsed the [openHAB MODBUS docs](https://www.openhab.org/addons/bindings/modbus/): what surprised me is that there doesn't seem to be the notion of a bus, just that of a MODBUS device connected to an RS485 interface with specific parameters (otherwise it looks like a nice piece of software!).

If that's the case (**please check!**) two C0135 devices would have to be connected to two independent RS485 buses. 

Otherwise a solution might be to build something using cheap Chinese modules (e.g. [8 relay board](https://www.aliexpress.com/item/1-2-4-8-Channel-DC-5V-Relay-Module-with-Optocoupler-Low-Level-Trigger-Expansion-Board/32905958885.html), [RS485 converter](https://www.aliexpress.com/item/MCIGICM-RS-485-TTL-to-RS485-MAX485CSA-Converter-Module-Integrated-Circuits-Products/32906230192.html), [STM8S103F3P6 module](https://www.aliexpress.com/item/1pcs-STM8S103F3P6-system-board-STM8S-STM8-development-board-minimum-core-board/32786729870.htm)), a couple of resistors and diodes for the inputs, some perfboard, and wires.

Now we come to the software: I've not yet done it but I guess that the implementation of the MODBUS RTU protocol isn't too hard if one has an example and a cheap protocol analyzer (the latter I have). I wanted to implement a UART interrupt handler anyway. What I don't know is if the MODBUS implementation on the openHAB side tolerates some delay when sending RTU replies.

bademux, 2018-12-18:
wow, thanks for such great analisis!
It is not a problem to update Openhab binding implementation, but hardware\firmware here is a great unknown to me :)

TG9541, 2018-12-18:
Ok then - give me some time - maybe I can hack a MODBUS RTU implementation during the Christmas break :-)

TG9541, 2018-12-19:
Ok, the [Modicon Modbus Protocol Reference Guide PI–MBUS–300 Rev. J](http://modbus.org/docs/PI_MBUS_300.pdf) contains the most useless description of a binary encoding I've ever seen:

```
When controllers are setup to communicate on a Modbus network using RTU
(Remote Terminal Unit) mode, each 8–bit byte in a message contains two 4–bit
hexadecimal characters.
```

I double checked: no, the MODBUS RTU protocol doesn't use "two hexadecimal characters" for "each 8-bit byte in a message":

[image]

This information about timing constraints is relevant:
```
Each message must be transmitted in a continuous stream.
```

Regarding endianness I found more information about magic of hexadecimal numbers. However, I infer that RTU is big-endian:

```
The CRC field is appended to the message as the last field in the message.
When this is done, the low–order byte of the field is appended first, followed by the
high–order byte.  The CRC high–order byte is the last byte to be sent in the
message.
```

sbridger, 2018-12-19:
FYI, [Realterm](https://realterm.i2cchip.com) can encode messages with modbus16 crc, on the "send" tab This might help you verify.

TG9541, 2018-12-19:
Thanks, that's cool

I'll have to "steal" my wife's laptop computer, though - maybe I should somehow get a Windows10 machine for such tests.

sbridger, 2018-12-19:
Here is the module source, for what its worth:

```
{MODBUS CRC-16 ObjectPascal (Delphi) Routine by Panu-Kristian Poiksalo 2003
This function calculates a checksum STRING for a message STRING for maximum
usability for me... example call for this function would be something like...
Examples
$47 03 00 01 00 08 CRC= 1B6A
1 3 0 1 0 1  CRC=D5CA

var message:string;
begin
message:=#1#5#0#0#255#0;
ModbusSerialPort.Output:= message + crc16string (message);
end;

happy coding! send me a greeting at bancho@microdim.net when you get it to
work and share this code freely with everyone at any web site!}
//sjb 12/5/15 change CHR() at end to fix errors.
// 1 3 0 $7E 0 10 = A5 D5 -good (old and new)
// 1 3 0 $81 0 10 = 95 E5 - good ; 3F E5 - bad

unit ModbusCRC;

interface
function ModbusCrc16String(msg:string):string;

implementation

function ModbusCrc16String(msg: string): string;
var
  crc: word;
  n, i: integer;
  b: byte;
begin
  crc := $FFFF;
  for i := 1 to length(msg) do
  begin
    b := ord(msg[i]);
    crc := crc xor b;
    for n := 1 to 8 do
    begin
      if (crc and 1) <> 0 then
        crc := (crc shr 1) xor $A001
      else
        crc := crc shr 1;
    end;
  end;
  result := ansichar(lo(crc)) + ansichar(hi(crc));
end;
end.
```

TG9541, 2018-12-19:
I'm using `lib/CRC16` and the following simple test.

```Forth
HEX
\ All light: 01 06 00 01 ff ff d9 ba (vendor's MODBUS RTU docs )
-1 $01 CRC16 $06 CRC16 $00 CRC16 $01 CRC16 $FF CRC16 $FF CRC16 . BAD9 ok
\ 1 3 0 $7E 0 10 = A5 D5 -good (old and new)
-1 $01 CRC16 $03 CRC16 $00 CRC16 $7E CRC16 $00 CRC16 $0A CRC16 . D5A5 ok
\ 1 3 0 $81 0 10 = 95 E5 - good ; 3F E5 - bad
-1 $01 CRC16 $03 CRC16 $00 CRC16 $81 CRC16 $00 CRC16 $0A CRC16 . E595 ok
```

**Looks good :-)**

Oh, it's been a while since I coded in Turbo Pascal 6 or Borland Pascal 7 - I used to earn my salary with that :-D

My humble routine looks like this:

```Forth
  : CRC16 ( n c -- n )
  \ CRC-16-ANSI (seed with -1 for Modbus CRC)
    XOR DOXCODE [
      $A608 ,  \         LD      A,#8
      $54 C,   \ 1$:     SRLW    X
      $2407 ,  \         JRNC    2$
      $01 C,   \         RRWA    X   ; XOR X,#0xA001
      $A801 ,  \         XOR     A,#0x01  
      $01 C,   \         RRWA    X
      $A8A0 ,  \         XOR     A,#0xA0
      $01 C,   \         RRWA    X
      $4A C,   \ 2$:     DEC     A
      $26F3 ,  \         JRNE    1$
   ] ;
```
43 bytes including header, no variables :-)

TG9541, 2018-12-20:
Interrupt driven RxD was easy. I also have the "3.5 character time" trigger for processing in the slave working. The processing runs in the idle task - not sure if that's the right place to put it but it's very convenient for debugging.

Let's see how the next steps will look like.

TG9541, 2018-12-21:
I ran into a problem: the CRC16 test with various examples was successful (see above). 

Now I tested processing of received messages with the command line tools [mbpoll](https://github.com/epsilonrt/mbpoll) and [modbus-cli](https://github.com/favalex/modbus-cli). Both programs send a different CRC value than I expected:

```
 A 10 64 1 2 2A 3D0F 5C9B

  80   A 10 64  1  2 2A 5C 9B  0  0  0  0  0  0  0  0  __d__*\_________ ok
```
`A 10 64 1 2 2A` is the data, `5C 9B` is the CRC (little endien, by the way!). 
`3D0F` is what  CRC16 gives me if I seed with -1:
 
`DECIMAL -1 HEX A CRC16 10 CRC16 64 CRC16  1 CRC16  2 CRC16 2A CRC16 . 3D0F ok`

Something isn't quite right.

TG9541, 2018-12-21:
The current code is here: https://gist.github.com/TG9541/25a0d8e2e68e42d08ecec13c7e9a9981

I'm using a STM8S103F3P6 breakout board and the DOUBLECOM binary.

TG9541, 2018-12-21:
The examples [here](http://www.simplymodbus.ca/enron_FC05.htm) are mostly correct, except that for me the CRC high and low bytes are consistently swapped:

```
DECIMAL -1 HEX 11 r 05 r 04 r  31 r FF r 00 r . 55DE ok
```

This is the example: 

```
Request

This command is writing the contents of boolean variable  # 1073 to ON
to the slave device with address 17.

11 05 0431 FF00 DE55

11: The Slave Address (11 hex = address17 )
05: The Function Code (Force Single Boolean)
0431: The Data Address of the boolean variable. (0431 hex = 1073 )
FF00: The status to write ( FF00 = ON,  0000 = OFF ) 
DE55: The CRC (cyclic redundancy check) for error checking.
```

TG9541, 2018-12-22:
[This thread](https://modbus.control.com/thread/1026168695) warrants social studies.

bademux, 2018-12-23:
@TG9541 thanks for research.
From what I understand Openhab modebus plugin can do swap on his side ([int64_swap](https://www.openhab.org/addons/bindings/modbus/#int64_swap)).
Looks like it is indeed 2 versions of protocol ASCII and RTU, [jamod](https://github.com/openhab/jamod) supports [both: "There exist two transmission modes, which differ in encoding, framing and checksum"](http://jamod.sourceforge.net/kbase/protocol.html). Threre is  "encoding" configuration parameter in openhab modbus plugin.

TG9541, 2018-12-23:
@bademux : I've assumed that we use MODBUS RTU as it appears to be much more popular than MODBUS ASCII. The jamod link you provided describes what kind of problems are to be expected when using that implementation (the 1.5 Character Time upper bound for serial transmission inter character delay is maybe problematic for anything that's JAVA based).

The best way to proceed, I guess, is to actually use openHAB to produce MODBUS RTU requests.

TG9541, 2018-12-27:
@bademux helped me to locate the problem: there is nothing wrong with openHAB/JAMOD or Python libmodbus (mbpoll and modbus-cli): the problem was that `?RX` uses the STM8 eForth `atokey` which doen't take 0x00 for an input... my bad.

The [MODBUS reveive Gist](https://gist.github.com/TG9541/25a0d8e2e68e42d08ecec13c7e9a9981) has been updated!

TG9541, 2018-12-30:
I've got biderectional communication with ISR chains working. STM8 UART TX handlers are a bit peculiar. The gist is [here](https://gist.github.com/TG9541/7f340dea90538f654eb6faa0f32093ac).

TG9541, 2019-01-01:
I tested the timing of the `UARTISR` code at 115200 baud by transmitting the 4 char sequence `ABBB` with `start-tx` and looping UART_TX back to UART_RX with a patch wire:

[image]

The diagram shows nicely how the TX ISR chain works (double load at spin-up 30µs, spin-down and end-of-transmission 20µs. The RX ISR also takes about 30µs for filling the receive buffer and for writing the receive time stamp.

My conclusion is that buffered communication with ISRs in STM8 eForth is good up to 115200 baud in full-duplex mode and up to 230400 baud in half duplex mode (e.g. MODBUS). Higher baud rates require optimizations, e.g. assembly instead of the Forth VM for control structures and data flow.

TG9541, 2019-01-01:
@bademux: that's good progress but we're not there yet :-)

Which MODBUS features will we need (e.g. supported MODBUS function codes, MODBUS registers in EEPROM)?

I can see that for the on/off-switches we'll need some kind of "fail-safe properties" (e.g. how to detect when the MODBUS master is down, how manipulate states locally). 

When there are local states we also need to establish a protocol for state information synchronization by the MODBUS master when the connection is active again.

bademux, 2019-01-01:
Yes it is good progress.
From Modbus it will be register (Descrete Output Coil?).
From Manual Switch (a User) I see the scenario:
when our uC "listens" Input port for a state changes (puch button clicked)
then the state of the Relay and state of the Register.

In case of faliure Modbus master reread register one again.
The only state in our system is "an variable" inside the uC, so I hope that modbus poll process will keep the state of our device in sync with master.
If for some reasone our device will be rebooted, then "OFF" state is good default one.

TG9541, 2019-01-01:
This sounds good! Users will observe an immediate reaction, and automation use cases can be handled by the MODBUS master nevertheless. States can even be stored in local EEPROM.

bademux, 2019-01-01:
Maybe it is better to store states in RAM?
Full turn off smiply resets all and device will start in OFF state.

TG9541, 2019-01-01:
First light with MBTEST:

```
thomas@w500:~/source/stm8s/stm8ef$ mbpoll -a10 -b 9600 -t4:int /dev/ttyUSB1 -- -1
mbpoll 1.4-8 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2018 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU
Slave configuration...: address = [10]
                        start reference = 1, count = 1
Communication.........: /dev/ttyUSB1,       9600-8E1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 32-bit integer (little endian), output (holding) register table

Written 1 references.
```

My test code [MBTEST](https://gist.github.com/TG9541/c12e81eb22288c0de4723811a9bf85fe) outputs the following:

```
CRC: -1
SLAVE:  10
FC:  16
R0:  0
R#:  2

 rxbuf: 13
  80   A 10  0  0  0  2  4 FF FF FF FF D7 1F  0  0  0  ________________
 txbuf: 8
  98   A 10  0  0  0  2 40 B3  0  0  0  0  0  0  0  0  ______@_________
```
This only works for FC 16 but at least it works :-)

The test code runs in an `'IDLE` task, which means that it processes stuff when the user doesn't enter commands with the Forth REPL.

I also updated [UARTISR](https://gist.github.com/TG9541/7f340dea90538f654eb6faa0f32093ac).

TG9541, 2019-01-01:
Now it also works with the JAMOD SerialDITest:

```
thomas@w500:~/source/jamod$ java -cp "target/jamod-1.2.3.OH-SNAPSHOT.jar:target/dependency/*" net.wimpi.modbus.cmd.SerialDITest /dev/ttyUSB1 43 13 11
net.wimpi.modbus.debug set to: null
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
2b 02 00 0d 00 0b af c4 
Response: 2b 02 02 0c 14 
Bit 0 = false
Bit 1 = false
Bit 2 = true
Bit 3 = true
Bit 4 = false
Bit 5 = false
Bit 6 = false
Bit 7 = false
Bit 8 = false
Bit 9 = false
Bit 10 = true
```

The STM8 eForth console outout is this:
```
SLAVE:  43
FC:  2
R0:  13
R#:  11

 rxbuf: 8
  80  2B  2  0  D  0  B AF C4  0  0  0  0  0  0  0  0  +_______________
 txbuf: 7
  98  2B  2  2  C 14 A5 71 4E  0  0  0  0  0  0  0  0  +_____qN________
``` 

I updated [MBTEST](https://gist.github.com/TG9541/c12e81eb22288c0de4723811a9bf85fe).

MODBUS has a way to keep a programmers busy by using 1 and 0 as index for the first element in a seemingly random way.

TG9541, 2019-01-01:
[image]

TG9541, 2019-01-02:
Here is something I'd like to come back to:

> If that's the case (**please check!**) two C0135 devices would have to be connected to two independent RS485 buses.

I think it's possible to create a distributed MODBUS slave, that is a number of nodes on the same bus that interpret the register address as an extended node identification. These would only answer to input, coil and holding r/w which they are responsible for. The shared bus would allow for a distributed error reaction when none of the nodes that share a MODBUS ID sends a regular response.

TG9541, 2019-01-02:
JAMOD didn't provide a CLI test for MODBUS RTU FC01 "Read Coils" and FC05 "Write Single Coils" so I did some copy-and-paste programming:

* FC01 [SerialCoilsTest.java](https://gist.github.com/TG9541/16d698c648812b1ce9935c41f1c85cf0)
* FC05 [SerialDOTest.java](https://gist.github.com/TG9541/15a6a45424490e1ed11cfa7299313129)

I updated [MBTEST](https://gist.github.com/TG9541/c12e81eb22288c0de4723811a9bf85fe) and implemented FC01 (Read Coils) and FC05 (Write Single Coil).

Here is the test result:

```
thomas@w500:~/source/jamod$ java -cp "target/jamod-1.2.3.OH-SNAPSHOT.jar:target/dependency/*" net.wimpi.modbus.cmd.SerialCoilsTest /dev/ttyUSB1 43 0 8
net.wimpi.modbus.debug set to: null
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
2b 01 00 00 00 08 3a 06 
Response: 2b 01 01 00 
Bit 0 = false
Bit 1 = false
Bit 2 = false
Bit 3 = false
Bit 4 = false
Bit 5 = false
Bit 6 = false
Bit 7 = false
thomas@w500:~/source/jamod$ java -cp "target/jamod-1.2.3.OH-SNAPSHOT.jar:target/dependency/*" net.wimpi.modbus.cmd.SerialDOTest /dev/ttyUSB1 43 1 true
net.wimpi.modbus.debug set to: null
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
2b 05 00 01 ff 00 da 30 
Response: 2b 05 00 01 ff 00 
thomas@w500:~/source/jamod$ java -cp "target/jamod-1.2.3.OH-SNAPSHOT.jar:target/dependency/*" net.wimpi.modbus.cmd.SerialCoilsTest /dev/ttyUSB1 43 0 8
net.wimpi.modbus.debug set to: null
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
2b 01 00 00 00 08 3a 06 
Response: 2b 01 01 01 
Bit 0 = true
Bit 1 = false
Bit 2 = false
Bit 3 = false
Bit 4 = false
Bit 5 = false
Bit 6 = false
Bit 7 = false
```

TG9541, 2019-01-05:
I did a test with the original relay board firmware. The programmer didn't care about the timing requirements, or didn't bother reading the protocol specification:   
[image]
To use the board with that firmware a MODBUS client must switch from TX to RX in about 150µs - that's sporty: even at 115.2kbaud the "3.5 char times" requirement equals 300µs.

TG9541, 2019-01-05:
Here is a general [MODBUS protocol handler](https://gist.github.com/TG9541/f3a3ce83c9a04e743353ddf46ab745a8) with the following features:
* client command detection
* CRC checking
* FC handler look-up (limited to FC01 .. FC16)
* FC parameter access
* answer envelope generation
* exception code code generation

By the way: there must be a bug in JAMOD: it doesn't take "no" for an answer. MBPOLL handles exception codes correctly, so I don't think that the problem is at my side.

TG9541, 2019-01-06:
Next step, here is a (very simple) [MODBUS server implementation](https://gist.github.com/TG9541/bbd0819ea46a3ffcfe73d52d15397c2d) that handles FC01 and FC05 (and that has dummies for FC02 and FC16).

The implementation works but some more factoring would be beneficial, and exception codes, e.g. for out-of-range parameters, should be implemented.

That shouldn't be a big problem, though: the complete MODBUS server (including protocol and buffered UART) currently fits in 1341 bytes Flash memory: in SWIMCOM there is still 2007 bytes free.

TG9541, 2019-01-06:
I updated [UARTISR](https://gist.github.com/TG9541/7f340dea90538f654eb6faa0f32093ac): the baud rate is now selectable from 2400 to 230400 baud.
An update of the STM8 eForth core will follow since it's advantageous to export some more constants that describe a target architecture (a.k.a. BOARD).

TG9541, 2019-01-12:
The [main functionality is now implemented ](https://gist.github.com/TG9541/bbd0819ea46a3ffcfe73d52d15397c2d) with working handlers for FC01, FC02, FC03, FC04, FC05, FC15 and FC16.

The code is really dense but, as I think, very maintainable. The MODBUS server (including buffered UART ISR routines) requires less than 400 lines of code and 1365 bytes of Flash memory (with 1983 bytes free). I used SWIMCOM, which is a "full featured" STM8 eForth binary - a custom binary for MODBUS could be smaller by at least 600 bytes.

What's still missing is range checks (and the evaluating the MODBUS server ID ;-) ). Not a biggie, maybe 20 lines of code and less than 200 bytes of Flash memory.

TG9541, 2019-02-05:
There is now a GitHub repository for MODBUS with STM8 eForth!
https://github.com/TG9541/stm8ef-modbus

closed 2019-02-05 by TG9541