TG9541/stm8ef #247: W1219 with Nuvoton Chip

suskitech, 2019-01-13:
In August of 2018 I ordered 3 XH-W1219 boards from an eBay vendor single2000 and they came with the Nuvoton N76E003AT20 chip. There were also some changes in the power diode which is an smd and resistor placement in that area.  Otherwise looks very close.  I assume that the codebase is completely different for the Nuvoton vs the STM8S.  But the question comes up is the pin out close enough that I can remove the Nuvoton and put in an STM8S?
Thanks in advance, Allen
PS I found a nifty little device on eBay titled "Li-ion Lithium Lead-acid Battery Capacity Meter Discharge Tester Analyzer" that is based on the STM8S003F3P6 going for around $3.30 each.

VK6TT, 2019-01-13:
Hi Allen,
 
yes, the codebase is completely different. However I've never had any issues
changing the microcontroller out. Just remember there is a capacitor usually
missing which needs to be also re-instated. The wiki shows where it is.
 
Regards
Richard
 
Was this the tester you were referring too? My initial reaction was it was
too well made with de-bounce capacitors across the switches *grin*
  <http://ae01.alicdn.com/kf/HTB1vw84ofuSBuNkHFqDq6xfhVXaP.jpg> 

  _____  

<https://github.com/TG9541/stm8ef/issues/247> GitHub, or mute
qQ_wr7bks5vCn-7gaJpZM4Z863y> the thread.
<https://github.com/notifications/beacon/AT7ya5njl07atCZKwrJKNUmK8bYx3MSMks5
vCn-7gaJpZM4Z863y.gif>

suskitech, 2019-01-14:
To VK6TT: Yes the link you showed is the exact tester.  I have 3 of them and use them to validate battery capacity of NiMH as well as my 18650's.  I see this as a useful tool 3 buttons, 4 digit LED plus current sense. After seeing this forum, I desoldered the display hiding underneath it was the stm8s along with a TM1650 LED controller.  Also hidden by the picture is what appears to be swim connections.  I have not had time to further analyse it, but may do that in the future.  I just don't have enough time and to many hobbies.

sbridger, 2019-01-15:
I also ordered thermostats from 3 different suppliers, and got all non-stm8 boards.

I suggest having a page where we can list the supplier URL and date of order, and STM8/NOT so we can have a better chance of getting these.

TG9541, 2019-01-25:
As there is now a hint in the STM8 eForth Wiki I'll be closing this issue.

closed 2019-01-25 by TG9541