TG9541/stm8ef #248: ZB2L3 Battery Capacity Tester

TG9541, 2019-01-14:
Yes the link you showed is the exact tester.  I have 3 of them and use them to validate battery capacity of NiMH as well as my 18650's.  I see this as a useful tool 3 buttons, 4 digit LED plus current sense. After seeing this forum, I desoldered the display hiding underneath it was the stm8s along with a TM1650 LED controller.  Also hidden by the picture is what appears to be swim connections.  I have not had time to further analyse it, but may do that in the future.  I just don't have enough time and to many hobbies.  Added a pic and noticed that silkscreen on the board states ZB2L3 on my other two that I purchased.
[image]
  @VK6TT

_Originally posted by @suskitech in https://github.com/TG9541/stm8ef/issues/247#issuecomment-453892683_

TG9541, 2019-01-14:
@suskitech:
This board looks interesting - thanks for letting us (and others) having a peek under the LED display!

I guess that the 321 on the silkscreen near the SOT23-5 chip is a generic [single OP-AMP](http://www.ti.com/lit/ds/symlink/lm321.pdf). The 8205 might be a [Dual N-Channel logic enhancement mode power FET](https://www.ait-ic.com/uploads/2011-01/05/_1294200000_5amsl7.pdf). For the TM1650 I couldn't find a datasheet - I found one e for the [TM1651](http://aitendo3.sakura.ne.jp/aitendo_data/product_img/ic/LED-driver/TM1651_%20V1.2/TM1651_V1.1_EN.pdf) and there is example code on TM1650 on GitHub.

I'll do the same surgery with my board. Let's see what else it's good for.

VK6TT, 2019-01-14:
Here's a non-English one which is surprisingly clear on what is happening.
It appears to be quite different than the TM1650. I couldn't find another
source for this part to see if that manufacturer had an easier to read
datasheet. 
 
Thomas, I think you're right regarding the other parts. The only thing to
watch out for is the regulator. I tested 10 of the 5 volt versions of these
on W1209 boards recently. Half let out smoke at 15.6 volts applied to the
power connector. 
 
It still looks very appealing. 
 
Regards
Richard

  _____  

<http://aitendo3.sakura.ne.jp/aitendo_data/product_img/ic/LED-driver/TM1651_
%20V1.2/TM1651_V1.1_EN.pdf>  and there is example code on TM1650 on GitHub.

<https://github.com/TG9541/stm8ef/issues/248#issuecomment-453908065> it on
GitHub, or mute
2FX-7fJks5vDBysgaJpZM4Z9o3J> the thread.
<https://github.com/notifications/beacon/AT7ya3NC8FBNdU5M0Wwris2NLVQWwhfAks5
vDBysgaJpZM4Z9o3J.gif>

suskitech, 2019-02-03:
Man that is to bad.  I ordered 4 more battery testers in hopes that it would still have the STM8 in them.

I have traced everything on the board, but did not have a chance to put it in nice format.  Here is what I do have:
STM8S003F3P6 Signals
Pin	STM	Connected to
1	PD4	<-> TM1650 3 SDA
2	PD5	
3	PD6
4	NRST <- (ICP header pin 3)
5	PA1
6	PA2
7	VSS	GND
8	VCAP 105 capacitor to GND
9	VDD
10	PA3	-> LT8205 N-Channel MOSFET pins 6 and 4 (gates) 
11	PB5	-> A LED
12	PB4	-> Ah LED
13	PC3	-> V LED
14	PC4/AIN4	<- V+ in 10k/51K ratio plus .06V
15	PC5	<- OK switch closes to GND 104 cap debounce
16	PC6	<- + switch closes to GND 104 cap debounce
17	PC7	<- - switch closes to GND 104 cap debounce
18	PD1/SWIM <-> (ICP header pin 2)
19	PD2/AIN3 <- LM 321 OpAmp
20	PD3/AIN4 -> TM1650 2 SCL

I also see that the version you received came with an ET6226M for the display driver instead the Titan Micro Electronics TM1650A, which has drivers in the Arduino environment.

I have to admit that the research I did on the Nuvoton showed qty 1000 pricing of $.25US vs around $.75US so with that kind of cost difference we will see more and more of this transition. There are a few tutorials that discuss programming these, but I just don't have time to learn another CPU.

TG9541, 2019-02-03:
The LED driver chip on the new board is an [ET6226M](http://archive.espec.ws/files/ET6226.pdf)

The chip is pin compatible with the TM1650 

Here is a Google translation of the functional register description:
<hr>
ET6226 Control Program Format The ET6226 control program format is shown in the figure below.

The instruction shown below is SYSON.
Start 0 1 0 0 1 0 0 0 Ack 0 0 0 0 0 0 0 1 Ack Stop

The specific control instructions are shown in the table below.

High 8 bits |low 8 bits |function
-|-|-
48| 00| Off display 
48| 01| On display (eight-segment mode)
48| 09| Turn on display (seven-segment mode)
48| 04| Turn on sleep
48| X1| Turn on the display, keyboard, X-level brightness X=1, first-level brightness; X=4, four-level brightness; X=0, eight-level brightness, the rest
68| XX| digital tube position 0 display, XX is 8 bits of data, the highest position is punctuation
6A| XX| digital tube position 1 display, XX is 8 bits of data, the highest position is punctuation
6C| XX| digital tube position 2 display, XX is 8 bits of data, the highest position is punctuation
6E| XX| digital tube position 3 display, XX is 8 bits of data, the highest position is punctuation
4F| —| Get button, return button code
<hr>

The Control Register description is sketchy but it appears to be consistent with the more detailed description in the TM1650 datasheet:

```
  MSB LSB  
  B7 B6 B5 B4 B3 B2 B1 B0 Function Description
  ×  0  0  0  -  ×  ×  -  Brightness setting 8 levels of brightness
  ×  0  0  1  -  ×  ×  -  1 level brightness
  ×  0  1  0  -  ×  ×  -  2 brightness
  ×  0  1  1  -  ×  ×  -  3 levels of brightness
  ×  1  0  0  -  ×  ×  -  4 brightness
  ×  1  0  1  -  ×  ×  -  5 brightness
  ×  1  1  0  -  ×  ×  -  6 brightness
  ×  1  1  1  -  ×  ×  -  7 brightness
  ×  -  -  -  0  ×  ×  -  7/8 segment display control bit 8-segment display mode
  ×  -  -  -  1  ×  ×  -  7-segment display mode
  ×  -  -  -  -  ×  ×  0  Turns the display bit on/off off
  ×  -  -  -  -  ×  ×  1  on display
```

TG9541, 2019-02-03:
@suskitech thanks for the pin description! I inserted some Markdown formatting, I hope you don't mind 
:-).

The good news is that PD1/SWIM can be used as a communications interface just like that. An interesting observation: these guys didn't even bother to use the I2C GPIOs PB4 and PB5.

suskitech, 2019-02-03:
I appreciate the reformatting and it was after midnight my time and did not want you to duplicate effort.  Thanks for the clean up feel free anytime.  I am new to GitHub and formatting with markdown code is another thing I need to look into.
We need a cheap source of STM8S103 chips.  I can get them for about $1 each on eBay, but most of the China and HK vendors are on vacation until Feb 11 celebrating their New Year.

TG9541, 2019-02-03:
I made a start by flashing the SWIMCOM binary to a ZB2L3 board - I'm trying to use M.Mahlows VOC-I2C demo in the examples folder. So far, I still get the error message `I2C: ACK missing?`. Maybe there are no pull-up resistors and I need to set internal pull-ups and the proper GPIO mode?

VK6TT, 2019-02-04:
Hi Thomas,
 
I had huge issues getting I2C working. My solution is shown below and it
might help you.
 
: I.tx ( byte -- )
\ Send a byte to the I2C Bus.
  DUP 7 FOR 
     $80 AND 
     IF SDA1 
     ELSE SDA0 
     THEN SCL1 
     2* DUP SCL0 
  NEXT 2DROP 
  SDA1 \ **** guess fix by earlier SDA1
  ?I.ACK 
;

Not so much as a guess, since the logic analyser showed there was a timing
issue. The guess was where to pull SDA high.
 
Regards
Richard 

  _____  

4Wn9spMks5vJzlBgaJpZM4Z9o3J> the thread.
<https://github.com/notifications/beacon/AT7yawGAwLH4t37H5Tavo9j6Je3toOoeks5
vJzlBgaJpZM4Z9o3J.gif>

VK6TT, 2019-02-04:
I gave up on flea market style sourcing when I found this
https://lcsc.com/product-detail/ST-Microelectronics_STMicroelectronics_STM8S
003F3P6_STM8S003F3P6_C18615.html
 
I use LCSC for some of my business sourcing and I have been very happy. But
I still watch the flea market wit great interest. Can't help myself!
 
 
  _____  

pDVfuXaks5vJxXKgaJpZM4Z9o3J> the thread.
<https://github.com/notifications/beacon/AT7ya-DMI14FSCN0HGitqKaN9iRMqCUnks5
vJxXKgaJpZM4Z9o3J.gif>

paulcrawford, 2019-02-04:
I ordered 10 pieces last October and paid CDN$5.56 including shipping. The same vendor is only CDN$4.42 today including shipping https://www.aliexpress.com/item/Free-Shipping-10pcs-lots-8S003F3P6-STM8S003F3P6-TSSOP-20-New-original-IC-In-stock/32919745220.html?spm=a2g0s.9042311.0.0.99794c4dgWvW62. The site Thomas quoted above is right now CDN$8.51 including shipping. The shipping is slow, 70-90 days to Canada, but if you are not in a rush...

TG9541, 2019-02-04:
@VK6TT Richard, thanks for the code! In the far past I wrote several bit-banged I2C interfaces that were used at industrial scale, and I also learned quickly that timing is critical.

LCSC looks interesting (e.g. STM8S003F3P6 for 0.285 @10pcs) but they add $3 for handling and at least $5 for shipping... I agree that if one's business depends on the delivery that's a small price to pay.

ildar, 2019-10-29:
Greetings!
1st, I found this great [video on how it works](https://www.youtube.com/watch?v=7SWExcGAnOQ) (analog part).
2nd, TM1650 datasheet is found on the [vendor site](http://www.titanmec.com/index.php/en/product/view/id/540/typeid/88.html).

TG9541, 2019-10-29:
@ildar thanks for reviving the discussion! 
The video is indeed very useful. 
Maybe I'll find some time working on this during the winter break - if you'd be willing to push support for this board I'd be willing to provide support.

ildar, 2019-10-29:
Unfortunately my device came with some other MCU, I'll post photos later

DrSaad77, 2020-04-16:
Can somebody  provide the soft to calibrate this tester please??
it seams that the IC lost the stuff, so when I plug it now, the 3 led blink and device turn off, so maybe I need to re upload the soft/code to the IC?? Thx for any help

ildar, 2020-04-16:
https://lmgtfy.com/?q=ZB2L3+calibration

TG9541, 2020-04-16:
@ildar I didn't quite manage to get to it during the winter break, sorry

ddodge40, 2022-05-13:
Has anyone worked out how to read data from the SWIM port to log the data from the ZB2L3?

TG9541, 2022-05-14:
@ddodge40 I'm still waiting for someone to write some specs:
* what do you expect the board to do so that a replacement software can be designed
* be inventive, e.g., remember that you don't need the buttons to control all of the behavior if it can be done with a script 
* what kind of data do you expect to receive

ddodge40, 2022-05-14:
I would like to grab the data as it comes in and also the final reading. Adding time to a file is easy. I saw on a blog that it is a ST processor, but would need to know where the readings are stored. 
There was a post about some software being available, but no info as to where to get it
Thanks 

Don Dodge
Cell 714-271-7888
W6SNX

TG9541, 2022-05-14:
@ddodge40 what you've described is the use case, not the specs of a software so that it can be designed. Unfortunately, I can't do the reverse engineering of the original firmware. Someone else will have to do that :-)

ddodge40, 2022-05-14:
I have asked HiLetgo and other sellers for the software and only HiLetgo have responded that they are looking for itThanks 

Sent from the all new AOL app for iOS

FBMinis, 2023-01-17:
> Yes the link you showed is the exact tester. I have 3 of them and use them to validate battery capacity of NiMH as well as my 18650's. I see this as a useful tool 3 buttons, 4 digit LED plus current sense. After seeing this forum, I desoldered the display hiding underneath it was the stm8s along with a TM1650 LED controller. Also hidden by the picture is what appears to be swim connections. I have not had time to further analyse it, but may do that in the future. I just don't have enough time and to many hobbies. Added a pic and noticed that silkscreen on the board states ZB2L3 on my other two that I purchased. [image] @VK6TT
> 
> _Originally posted by @suskitech in [#247 (comment)](https://github.com/TG9541/stm8ef/issues/247#issuecomment-453892683)_

Which load do you use to test your NiMH?

wolf-t-me, 2023-06-24:
I bought this board recently but he micro USB port broke off after very limited use and I was unable to re-attach it.
I identified tat I could power the board by soldering a severed USB cable's + and - wires directly to two exposed solder points on the rear of the board.
It works fine but it will not allow me to enter calibration mode, which would normally be achieved by pressing and holding the three micro-switches while powering up the board.
It would seem that the physical insertion of a micro USB male to the original port provides an additional trigger to initialise the calibration function.
Can anyone suggest a way of entering calibration while the unit is being powered in my unconventional way?

selevo, 2023-10-11:
it's now others mcu
ft61f135-rb
[image]

ddodge40, 2023-10-11:
Have one and checked the calibration and it is within 5%. Not bad for an inexpensive tester.  I use it for a few 12 volt batteries removed from FIOS service. Even 5 year old text at about 50% of original Don DodgeCell 714-271-7888W6SNX

TG9541, 2023-10-11:
> it's now others mcu and other pinout
> ft61f135-rb

Here is [FT61F13X datasheet](https://www.lcsc.com/product-detail/Microcontroller-Units-MCUs-MPUs-SOCs_FMD-Fremont-Micro-Devices-FT61F135-TRB_C708789.html) - the device appears to be PIC16-like and it has just 3K words ROM and 256 bytes. That's what I call "old school" and "frugal". 

> The era of quick and easy modification of ready-made devices is history

Yes, that appears to be the case.

dumaster, 2024-01-18:
> it's now others mcu and other pinout ft61f135-rb The era of quick and easy modification of ready-made devices is history
> 
> ![IMG_20231007_210807_edit_331079661148959](https://private-user-images.githubusercontent.com/13694921/274307779-2c10cfe9-addb-4bb9-9c92-fb40b10e70c2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1NjE1MTMsIm5iZiI6MTcwNTU2MTIxMywicGF0aCI6Ii8xMzY5NDkyMS8yNzQzMDc3NzktMmMxMGNmZTktYWRkYi00YmI5LTljOTItZmI0MGIxMGU3MGMyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDA3MDAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0YmZiODQxNTY3MjczMzc3YTlmMGJkNjUyNWVmNWU1ODM1NTY1Zjc4ZjU5MTExOWRhZjY0NzU2ODNiYTNjNjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SVarStqhyZNaJL0NtEBJhBnu_XqZl8MUw0jrYsQHSg0)

I have this new version 3.0 of the HW-586, it does not enter calibration mode using the method used in older versions, which is to press the three buttons simultaneously and connect the power to USB, has anyone found out if they changed the method for entering calibration mode Or if in this v3.0 there is no way to enter calibration mode ???

I will be very happy and grateful if anyone can help !!!

Thanks
Edward

selevo, 2024-01-22:
> I have this new version 3.0 of the HW-586, it does not enter calibration mode using the method used in older versions, which is to press the three buttons simultaneously and connect the power to [...]

Almost all of them now have no calibration.
It's just a terrible copy of what came before.
Programmers with crooked hands and brains did not add a calibration mode; it is also impossible to view the cut-off voltage in operating mode, only once after turning off the power, by re-installing it.
The modules also do not remember settings after being disconnected.
I think there are still older versions somewhere, but they are more expensive and do not compete.
And sellers on Aliexpress don’t understand anything about this, they buy what’s cheaper, even if it’s 0.1 cheaper

dumaster, 2024-01-23:
> > > it's now others mcu and other pinout ft61f135-rb The era of quick and easy modification of ready-made devices is history
> > > ![IMG_20231007_210807_edit_331079661148959](https://private-user-images.githubusercontent.com/13694921/274307779-2c10cfe9-addb-4bb9-9c92-fb40b10e70c2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1NjE1MTMsIm5iZiI6MTcwNTU2MTIxMywicGF0aCI6Ii8xMzY5NDkyMS8yNzQzMDc3NzktMmMxMGNmZTktYWRkYi00YmI5LTljOTItZmI0MGIxMGU3MGMyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDA3MDAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0YmZiODQxNTY3MjczMzc3YTlmMGJkNjUyNWVmNWU1ODM1NTY1Zjc4ZjU5MTExOWRhZjY0NzU2ODNiYTNjNjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SVarStqhyZNaJL0NtEBJhBnu_XqZl8MUw0jrYsQHSg0)
> > 
> > 
> > I have this new version 3.0 of the HW-586, it does not enter calibration mode using the method used in older versions, which is to press the three buttons simultaneously and connect the power to USB, has anyone found out if they changed the method for entering calibration mode Or if in this v3.0 there is no way to enter calibration mode ???
> > I will be very happy and grateful if anyone can help !!!
> > Thanks Edward
> 
> Almost all of them now have no calibration. It's just a terrible copy of what came before. Programmers with crooked hands and brains did not add a calibration mode; it is also impossible to view the cut-off voltage in operating mode, only once after turning off the power, by re-installing it. The modules also do not remember settings after being disconnected. I think there are still older versions somewhere, but they are more expensive and do not compete. And sellers on Aliexpress don’t understand anything about this, they buy what’s cheaper, even if it’s 0.01 cheaper
> 
> Perhaps in the future this will be done on disposable programmable microcontrollers. This will be the next stage of optimization.

Is it not possible to create firmware for these ft61f135-rb MCUs to include these essential features?

Thanks
Edward

selevo, 2024-01-23:
> Is it not possible to create firmware for these ft61f135-rb MCUs to include these essential features?
>
> Thanks Edward

You can do it yourself if you have a lot of time and desire.
To do this, you need to download the development environment from the manufacturer’s website,
 Study the documentation for the chip and buy a programmer.
Most likely, no one will do this except you, if only because tomorrow these “blanks” could be made on another microcontroller and all the work will be useless for the community and
 almost useless for this person in the future.

dumaster, 2024-01-24:
> > Is it not possible to create firmware for these ft61f135-rb MCUs to include these essential features?
> > Thanks Edward
> 
> You can do it yourself if you have a lot of time and desire. To do this, you need to download the development environment from the manufacturer’s website, Study the documentation for the chip and buy a programmer.
> 
> Most likely, no one will do this except you, if only because tomorrow these “blanks” could be made on another microcontroller and all the work will be useless for the community and almost useless for this person in the future.
> 
> You can also write about this on the product review page on Aliexpress. If no one buys the 'garbage', they will stop producing and resell 'garbage'.

So there remains one last attempt, hardware calibration, perhaps to modify the circuits that generate the signals at the MCU's analog inputs, for example by modifying the SHUNT resistor, or place a trimpot to change the values that arrive at the MCU!!!

But thank you very much for your attention and if I have any news I will send it here in this thread!!!

selevo, 2024-01-24:
> So there remains one last attempt, hardware calibration, perhaps to modify the circuits that generate the signals at the MCU's analog inputs, for example by modifying the SHUNT resistor, or place a [...]

There is one more point besides adjusting the resistors....
This is the linearity of the ADC of these microcontrollers and the accuracy of the timer.
If, for example, it ticks faster, then in an hour it will calculate more capacity than it is, and if it ticks slower, then less capacity.
If the ADC correctly shows a voltage of 1 volt, then it can show 9.2 volts at a voltage of 10 volts.
You can check this if you have a multimeter and compare.
The accuracy of the timer can be checked by connecting this tester to the power supply and setting any voltage, for example 10 volts. If there are two 7.5 Ohm resistors in series, then the current will be 10\15 = 0.667
After an hour the tester should show 667

satyajitbej903-hue, 2025-08-02:
> Man that is to bad. I ordered 4 more battery testers in hopes that it would still have the STM8 in them.
> 
> I have traced everything on the board, but did not have a chance to put it in nice format. Here is what I do have:
> 
> ```
> STM8S003F3P6 Signals
> Pin	STM	Connected to
> 1	PD4	<-> TM1650 3 SDA
> 2	PD5	
> 3	PD6
> 4	NRST <- (ICP header pin 3)
> 5	PA1
> 6	PA2
> 7	VSS	GND
> 8	VCAP 105 capacitor to GND
> 9	VDD
> 10	PA3	-> LT8205 N-Channel MOSFET pins 6 and 4 (gates) 
> 11	PB5	-> A LED
> 12	PB4	-> Ah LED
> 13	PC3	-> V LED
> 14	PC4/AIN4	<- V+ in 10k/51K ratio plus .06V
> 15	PC5	<- OK switch closes to GND 104 cap debounce
> 16	PC6	<- + switch closes to GND 104 cap debounce
> 17	PC7	<- - switch closes to GND 104 cap debounce
> 18	PD1/SWIM <-> (ICP header pin 2)
> 19	PD2/AIN3 <- LM 321 OpAmp
> 20	PD3/AIN4 -> TM1650 2 SCL
> ```
> 
> I numbered the head pins left to right while reference the board in its nor usable orientation.
> 
> ```
> 1 VCC (left pad only)
> 2 SWIM
> 3 NRST
> 4 GND (right pad only)
> ```
> 
> I also see that the version you received came with an ET6226M for the display driver instead the Titan Micro Electronics TM1650A, which has drivers in the Arduino environment.
> 
> I have to admit that the research I did on the Nuvoton showed qty 1000 pricing of .25 U S v s a r o u n d .75US so with that kind of cost difference we will see more and more of this transition. There are a few tutorials that discuss programming these, but I just don't have time to learn another CPU.

Give me full code please 🙏