TG9541/stm8ef #249: Errors with "codeload.py" preventing upload of source files

paulcrawford, 2019-01-23:
I am using an Odroid C2 with Ubuntu 16.04 to upload source code from W1209-0.2.1 to a W1209 board with codeload.py. The first error I found with "make load" was:

if [ ! -d "out" ]; then \
	curl -# -L -O https://github.com/TG9541/stm8ef/releases/download/2.2.20/stm8ef-bin.zip; \
	unzip -q -o stm8ef-bin.zip -x out/*; \
	unzip -q -o stm8ef-bin.zip out/W1209-FD/*; \
	rm stm8ef-bin.zip; \
fi
tools/codeload.py serial main.fs
Uploading main.fs
TX: \ W1209 basic data logging thermostat
TX: \ © 2017 TG9541, refer to https://github.com/TG9541/W1209/blob/master/LICENSE
TX: \ Note: not yet tested with real eggs :-)
TX: 
TX: NVM
TX: 
#require MARKER: skipped
TX: 
TX: RAM
TX: 
TX: : TARGET NVM ;
TX: 
TX: $4000 CONSTANT EE.SET    \ EEPROM cell "control set-point"
TX: $4002 CONSTANT EE.HYS    \ EEPROM cell "control hysteresis"
TX: $4004 CONSTANT EE.DEL    \ EEPROM cell "control delay"
TX: $4006 CONSTANT EE.COR    \ EEPROM cell "offset correction"
TX: $4008 CONSTANT EE.LOG    \ EEPROM cell "logger interval"
TX: 
TX: $8000 CONSTANT DEFAULT   \ Default value indicator (-32768)
TX: 
TX: TARGET
TX: 
TX: : init ( -- ) \ chained init - starting point
TX: 6 ADC!      \ W1209: use ADC channel 6
TX: ;
TX: 
TX: : .0 ( n -- )
TX: \ print fixed point number (n w/ factor 10x)
TX: DUP DEFAULT = IF
TX: DROP ."  DEF."  \ default display, e.g. sensor error
TX: ELSE
TX: \ formatted output for 3 digit 10x fixed point numbers
TX: DUP -99 < OVER 999 SWAP < OR IF
TX: \ number (and sign) too big for 3 7S-LED digits
TX: 5 + 10 / .    \ +0.5 before "floored division"
TX: ELSE
TX: \ -9.9 <= val <= 99.9
TX: SPACE DUP >R ABS <# # 46 HOLD #S R> SIGN #> TYPE
error SPACE DUP >R ABS <# # 46 HOLD S R> SIGN #> TYPE S?

Makefile:18: recipe for target 'load' failed

I commented out :.0 ...; in main.fs and tried again. This time the error was:

if [ ! -d "out" ]; then \
	curl -# -L -O https://github.com/TG9541/stm8ef/releases/download/2.2.20/stm8ef-bin.zip; \
	unzip -q -o stm8ef-bin.zip -x out/*; \
	unzip -q -o stm8ef-bin.zip out/W1209-FD/*; \
	rm stm8ef-bin.zip; \
fi
tools/codeload.py serial main.fs
Uploading main.fs
TX: \ W1209 basic data logging thermostat
TX: \ © 2017 TG9541, refer to https://github.com/TG9541/W1209/blob/master/LICENSE
TX: \ Note: not yet tested with real eggs :-)
TX: 
TX: NVM
TX: 
#require MARKER: skipped
TX: 
TX: RAM
TX: 
TX: : TARGET NVM ;
TX: 
TX: $4000 CONSTANT EE.SET    \ EEPROM cell "control set-point"
TX: $4002 CONSTANT EE.HYS    \ EEPROM cell "control hysteresis"
TX: $4004 CONSTANT EE.DEL    \ EEPROM cell "control delay"
TX: $4006 CONSTANT EE.COR    \ EEPROM cell "offset correction"
TX: $4008 CONSTANT EE.LOG    \ EEPROM cell "logger interval"
TX: 
TX: $8000 CONSTANT DEFAULT   \ Default value indicator (-32768)
TX: 
TX: TARGET
TX: 
TX: : init ( -- ) \ chained init - starting point
TX: 6 ADC!      \ W1209: use ADC channel 6
TX: ;
TX: 
TX: \  : .0 ( n -- )
TX: \    \ print fixed point number (n w/ factor 10x)
TX: \    DUP DEFAULT = IF
TX: \      DROP ."  DEF."  \ default display, e.g. sensor error
TX: \    ELSE
TX: \      \ formatted output for 3 digit 10x fixed point numbers
TX: \      DUP -99 < OVER 999 SWAP < OR IF
TX: \        \ number (and sign) too big for 3 7S-LED digits
TX: \        5 + 10 / .    \ +0.5 before "floored division"
TX: \      ELSE
TX: \        \ -9.9 <= val <= 99.9
TX: \        SPACE DUP >R ABS <# # 46 HOLD #S R> SIGN #> TYPE
TX: \      THEN
TX: \    THEN
TX: \  ;
TX: 
Uploading /home/daco/src/W1209-0.2.1/measure.fs
TX: \ W1209 temperature measurement with filter and noise suppression
TX: \ © 2017 TG9541, refer to https://github.com/TG9541/W1209/blob/master/LICENSE
TX: 
TX: \ Note: W1209 thermostats may require adjustment,
TX: \       especially when used outside the range of -5C to +20C
TX: \       Refer to https://github.com/TG9541/W1209/wiki/W1209-Sensor
TX: 
Uploading /home/daco/src/W1209-0.2.1/lib/STARTTEMP
TX: \ Scaffolding for temporary RAM usage during compilation
TX: \ refer to github.com/TG9541/W1209/blob/master/LICENSE.md
TX: 
TX: RAM
error RAM
Error file /home/daco/src/W1209-0.2.1/measure.fs line 8: could not upload file
>>>  #include STARTTEMP
Error file main.fs line 43: could not upload file
>>>  #include measure.fs
Makefile:18: recipe for target 'load' failed

Perhaps the second error is due to running the python code on an ARMv8 Cortex-A53 64bit Amlogic S905 processor rather than X86-64.

Can you please advise on what I need to do to make this work?

Thanks,

Paul Crawford

TG9541, 2019-01-23:
Hello Paul,

I think that in your setup codeload.py works as expected but there might be some problem with the serial interface.

In the first upload attempt there is a clear sign of a serial transmission problem:
```Forth
TX: SPACE DUP >R ABS <# # 46 HOLD #S R> SIGN #> TYPE
error SPACE DUP >R ABS <# # 46 HOLD S R> SIGN #> TYPE S?
``` 
`S` instead of `#S`, the `#` is missing.

In the second case I don't see what has happened - maybe the <LF> was dropped and there was a timeout? In any case the interpreter didn't throw a `?` at you.

I have no personal experience with Odroid - can it run Raspberry Pi binaries ? Maybe with the help of QEMU? In that case you might want to test your serial interface with e4thcom first. In fact, any terminal program will work, e.g. Picocom for which there is a howto in the [tools page](https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Programming-Tools).

I'd like to recommend a test of your tool chain with a plain STM8 board (you can also flash a MINDEV binary on your W1209 and connect [PD5-TxD/PD6-RxD](https://github.com/TG9541/stm8ef/wiki/Board-W1209#stm8s003f3) to your serial interface.

If all other things fail you use the binary created by [Travis-CI](https://travis-ci.org/TG9541/W1209/jobs/456282305#L1273).

Kind regards,
Thomas

TG9541, 2019-01-25:
Hello Paul,

thanks for the praise for this STM8 eForth - the biggest part goes to people who contributed ideas, know-how and code, and I happily accept it in the name of this small community :-)

I think that your specific use case can be easiest solved in the following way:

* try to extract the ihx file from the binary dump in the Travis-CI link above (that's indeed something I made for your use case)
* we work together in creating a linearization table for your HW variant using ADC readouts and the Forth console, and put it in a W1209 GitHub branch
* Travis-CI builds your new binary and you can move on to doing more interesting things, e.g. find a less frustrating option for interactive development

I think that we should continue this solution branch in a GitHub issue in the W1209 project repository and continue here with W1209/Odroid related trouble-shooting as needed.

Kind regards,
Thomas

paulcrawford, 2019-01-25:
Hello Thomas,

I have started a new issue at TG9541/W1209 to initiate the required source code changes for my application.

I did successfully get the cli for Travis CI to work on my headless Odroid c2 via ssh but I could not get stm8flash to make so I will use my Mac which does have stm8flash installed and tested successfully. The problem with the C2 and stm8flash was related to libusb-1.0. Perhaps we could look at that issue although it is not directly related at all to the W1209 or stm8ef.

Thanks,

Paul

BigVulcanDeal, 2019-01-27:
Thanks for the ringside seat guys. I found it relatively easy to adjust the
Forth program to make the device work as an Edge Device for my application.
Basically I had to:

- Adjust the lookup table for a different RTD
- Eliminate the user interface (there was some weird race condition that
occasionally caused the setpoint for the device to reset to an unwanted
default)
- Tinker with the serial interface to report deltas in measured
temperature, and to accept comands for setpoint and hysteresis parameters.
This serial I/O was designed to makenit easy to interact with Node-red

Overall, it was relatively easy, but I will confess that there are parts of
the program that used some interesting indirection mechanisms that I never
quite deciphered.

I never bothered making a hex image .. just uploaded the altered Forth
script and committed it to nonvolatile memory, since there were only a
handful of systems that used this W1209 setup for temperature control.

TG9541, 2019-01-27:
@BigVulcanDeal thanks joining the discussion, and for pointing out that there is a flaw in the menu implementation that will have a bad impact on applications that use the key GPIOs to send data frequently. 
I mentioned this issue here https://github.com/TG9541/W1209/issues/22 . I guess that a "ringside seat" there won't be without a certain entertainment value :-)

closed 2019-01-27 by TG9541