TG9541/stm8ef #253: W1209 wiki: LED segments B and F reversed. (B=A2, F=A1)

unfrozen, 2019-02-24:
Thank you for your information on the W1209 and STM8S in general.

I just created my library for the w1209 board, and found that your table of LED segments has B and F reversed. On my board, B is A2 and F is A1.
[](https://github.com/TG9541/stm8ef/wiki/Board-W1209)

I invite you to look at my libraries and test/example code for the STM8S103, -105, and now -003:
https://github.com/unfrozen/stm8_libs/wiki
https://github.com/unfrozen/stm8_tests/wiki
Thanks,
-Richard

TG9541, 2019-02-25:
Hi @unfrozen 
Thanks for the hint and for the links - I'll review them and reply soon!

TG9541, 2019-10-24:
I referenced the finding [in the Wiki](https://github.com/TG9541/stm8ef/wiki/W1209-Identifying-'compliant'-boards).

closed 2019-10-24 by TG9541