TG9541/stm8ef #254: Incorrect comparison?

NieDzejkob, 2019-03-06:
I was investigating various Forth implementations and noticed that [`<` is implemented](https://github.com/TG9541/stm8ef/blob/master/barebones/barebones.f#L212) as
```
: < - 0< ;
```

While I couldn't get this Forth to run in an emulator, as far as I can see this implementation does not fare well with overflow. Consider `-128 1 <` (or whatever your cell size is) - this will evaluate as false, even though both numbers are in range of the cell and the former is clearly smaller than the latter.

TG9541, 2019-03-07:
That's normal behavior of an "mod 2^n computation system", one with two's complement, I suppose.
Cell size is 16:
```Forth
-128 1 < . -1 ok
-1 1 < . -1 ok
-32768 1 < . 0 ok
: < - 0< ; reDef < ok
-128 1 < . -1 ok
-1 1 < . -1 ok
-32768 1 < . 0 ok
```
The wrap-around can be avoided by using 32bit arithmetics. Implementing floored (clipping) arithmetics would also be possible, but that's quite costly on an 8bit µC without an instruction set designed for that purpose.

NieDzejkob, 2019-03-07:
I thought about it for a while, and this should work for all numbers while still being quite efficient. Using this in my Forth does not exhibit any problems and passes the ANS test suite:
```
: < $8000 + >r $8000 + r> u< ;
```

TG9541, 2019-03-08:
Very nice, this sure doesn't look obvious :-) 

Changing the default behavior of signed number operators is a breaking change, though. I see 3 options: community vote, build option or a library function such as as `S<`.

VK6TT, 2019-03-08:
I never knew there was an issue with "<" and I've used it a lot. Better
minds than mine have clearly looked closely at this but I'm struggling to
see how CPW at the heart of < generates incorrect results. 
 
So, at the risk of again looking like someone who hasn't grasped the
blinding obvious, are we fixing something that isn't actually broken?

  _____  

<https://github.com/TG9541/stm8ef/issues/254#issuecomment-470819608> it on
GitHub, or mute
61OuJwOks5vUgErgaJpZM4bh40V> the thread.
<https://github.com/notifications/beacon/AT7yaxBlY8eQh8zsibxspyB26hO1reNgks5
vUgErgaJpZM4bh40V.gif>

NieDzejkob, 2019-03-08:
Right now, `<` uses a subtraction, which behaves incorrectly on underflow. For example, `-32768 32767 <`, or even `-30000 3000 <`. These should clearly return true (cf. the [ANS Forth](http://forth-standard.org/standard/core/less) specification for this word, especially the test suite), but with the current implementation, false will be returned.

I believe that no programmer would intentionally rely on this inaccuracy, but this change, should it happen, should still definitely be mentioned in the changelog.

As to my proposed implementation, I see two ways to explain it.

Imagine a ring (*not* the abstract algebra one) of numbers from 0 to 65535 (-1). Because of the modulo wrapping, there's not really an entirely natural way of defining an ordering. However, we can put a separator between 0 and 65535, and unroll the ring into a truncated number line. This is what U< does. Notice how adding a constant to both parts of the comparison moves this separator around (or, the ring rotates and separator is immobile).

Alternatively, notice how U< compares two negative numbers correctly. This is related to the fact that addition and subtraction don't require separate signed and unsigned versions for two's complement arithmetic. The trick here is to flip the sign of all numbers, making all positive numbers be greater (unsigned) than the negative numbers.

TG9541, 2019-03-09:
@NieDzejkob wrote:
> I believe that no programmer would intentionally rely on this inaccuracy, but this change, should it happen, should still definitely be mentioned in the changelog.

I used to rely on such properties for control applications with modulo arithmetics (e.g. with encoders). This isn't a major concern, though. It's more that I don't know now which assumptions words that use `<` make about the properties. I'll try it as a build option and do some tests. As a library function, an "extended range `<`" is obviously useful.

@VK6TT wrote:
> So, at the risk of again looking like someone who hasn't grasped the
blinding obvious, are we fixing something that isn't actually broken?

I think that it's worth checking how ANSI C does it.

TG9541, 2019-03-09:
I did a test with C:
```C
int a = 0x1234;
int b = -1;

void main(void)
{
  if (a < b) {
    a = 0; 
  }
}
```

SDCC (in this case old stable 3.6.0) turns that into:
```Asm
_main:
;       test.c: 7: if (a < b) {
        ldw     x, _b+0
        cpw     x, _a+0
        jrsgt   00109$
        ret
00109$:
;       test.c: 8: a = 0; 
        clrw    x
        ldw     _a+0, x
        ret
```
I don't mean to say that there is nothing to improve there, but most people, including the designers of 16bit instruction sets, do it the easy way.

closed 2019-05-27 by TG9541