TG9541/stm8ef #259: Apply workaround for ASxxxx macro bug

TG9541, 2019-08-10:
@kenyapcomau [found a workaround](https://hackaday.io/project/166954-eforthz80-modifications/log/166720-using-a-z80-assembler) for the bug in ASxxxx (`sdasstm8`) macro processing described in the [SDCC tracker issue 2626](https://sourceforge.net/p/sdcc/bugs/2626/): 

using octal escapes for characters like `,` and `;` prevents the assembler crash. 

Now all headers in `forth.asm` can use the `HEADER` macro!

kenyapcomau, 2019-08-10:
I have sent a link to this page to Alan Baldwin.

kenyapcomau, 2019-08-11:
Reply from Alan:

Ken,

  I see how the argument being passed is causing the problem.
A macro expects arguments to be separated by white space or the
delimiter ','.  So if the macro is passed an argument as -

   HEADER "C"  which has no delimiters in the argument "C"
               the "C" is passed as the argument.

If the macro is passed an argument as -

   HEADER ","  which has a delimiter, then the macro believes
               there are two arguments " and ".

If the argument contains a ";" then the ; signifies and end of code
and the argument is a single ".

The macro documentation specifies how to pass arguments containing
delimiters as ^/arg/.  So the method to pass string arguments with
(or without) delimiters is -

   HEADER  ^/","/

or with any argument containing white space.
Note that the / character can be any character
not in the string to be passed.  The arguments
are just a series of characters and the "
characters have no special meaning to the macro
parser.

I hope this clarifies the problem.

Alan

p.s.:
From the assembler code it appears my checking
of the macro argument count is in the wrong
place and misses detecting the extra arguments
trashing the macro structure, this I will need
to fix.  I have attached MT.asm which calls
the macro HEADER over 500 times. Uncommenting
the line containing HEADER a,b will cause a
seg fault.

[MT.zip](https://github.com/TG9541/stm8ef/files/3489426/MT.zip)

TG9541, 2019-08-11:
Great info, thanks Ken!
 
Using the notation `^/"..."/` mostly works, but it leaves the special case of `HEADER  STRCQ ^/"$,""/` which results in the following error message:
```
forth.asm:3410: Error: <q> missing or improper operators, terminators, or delimiters
```
Encoding `$,"` as `'$\054"'` works.

PR #260 (2019-08-10): Macro
Fixes #259 "Apply workaround for ASxxxx macro bug", also opens the 2.2.24 development line.
commit bf903c86 (2019-08-11): fixes #259 apply Alan Baldwin\'s advice
closed 2019-08-11 by TG9541