TG9541/stm8ef #267: STM8L-DISCOVERY

goog, 2019-08-20:
segment LCD ,

VK6TT, 2019-08-20:
Do you mean reduce power consumed by LCD or power consumed by STM8?

goog, 2019-08-20:
reduce power consumed by LCD

VK6TT, 2019-08-20:
I don't think there is a way to achieve that. Did you try to google "reduce power consumed by LCD"? That might help.

TG9541, 2019-08-20:
@goog: maybe this answers your question: https://community.st.com/s/question/0D50X00009XkaMLSAZ/stm8l-power-current-consumption-problem

closed 2019-08-20 by TG9541

goog, 2019-08-21:
@TG9541  thanks!

TG9541, 2019-08-21:
@goog you're welcome. 

If you find a solution to your problem don't forget to write about it, either here (if it has anything to do with STM8 eForth) or e.g. on the ST-Community site. A good description of your application would also help (e.g. tool chain, target application).