TG9541/stm8ef #275: W1209-CA-V2: another variation of the Common Anode boards

TG9541, 2019-10-28:
Hi, it looks like I've found another variation of the Common Anode boards (bought on Ebay in 2016) 
 which has totally different pin mappings, and a different layout please see the
 Readme.txt in the attached zipfile.

[W1209-CA-V2.zip](https://github.com/TG9541/stm8ef/files/3775299/W1209-CA-V2.zip)

_Originally posted by @g7bbg in https://github.com/TG9541/stm8ef/issues/153#issuecomment-546643256_

TG9541, 2019-10-28:
@g7bbg that's a nice find!

This variant looks actually better than most of the W1209 designs I've seen. Maybe it could be a starting point a "W1209 NG" design with direct access to a serial interface?

Would you care about following the "Fork - Pull-Request" process? If you're new to Git or GitHub I can help you getting started.

g7bbg, 2019-10-28:
Yes I would be pleased to go through the Fork/Pull process,  as you correctly surmised I am new to GitHub and would certainly apreciate the help to get it started :-)

TG9541, 2019-10-28:
@g7bbg that a lot easier to do than getting the STM8 eForth tool-chain up and running :-)
* prerequisite: [Git client installed](https://help.github.com/en/github/getting-started-with-github/set-up-git) and [SSH configured in GitHub](https://help.github.com/en/github/authenticating-to-github/connecting-to-github-with-ssh) 
* in the repo click on the `Fork` button to the right [image]
* in your fork of stm8ef (i.e. `g7bbg/stm8ef`) click `Clone or download`, `Use SSH` and "copy to clipboard" (icon) 
[image]
* in your usual  "source" folder run `git clone git@github.com:g7bbg/stm8ef.git` (i.e. the repository URL in you clipboard)
* cd into the sandbox and make a branch for the changes and change into it: `git checkout -b W1209-CA-V2`
* copy the W1209-CA-V2 board config folder and files from your previous source directory into your sandbox (just the `Readme.Txt` renamed to `README.md` and the files `globconfig.inc` and `boardcore.inc` - the photos and the hex files will go somewhere else later on)
* add the folder to your branch `git add W1209-CA-V2/` and commit it `git commit -a -m "new board W1209-CA-V2"`
* push your branch to your fork: `git push origin -u W1209-CA-V2`
* change to GitHub and click on `New pull request`: 
[image]
* select your new branch and `TG9541/stm8ef` as the target
* write `fixes #275` in the comment and create the request

The process above contains steps that you only need to do once or that are part of a normal development cycle with Git anyway. If you have questions (or if something doesn't work) please let me know it!

g7bbg, 2019-10-30:
Thomas,
 Hi I've updated the stm8 stuff as per your instructions and HOPEFULLY got it right.

Regards
 Brian

TG9541, 2019-10-30:
@g7bbg 

Hi Brian, yes, you got it right! Congrats! 

I'll do some additional changes, like formatting of the `README.md`.

closed 2019-10-30 by TG9541

TG9541, 2019-10-30:
@g7bbg I merged the new board now:
* I added `target.inc` which I forgot to mention
* the DOS line endings in your files have been removed
* the `README.md` contains the pictures above
* the `Makefile` includes your build
The next (pre-) release will contain the binary.

Please note that I'm thinking of moving most board definitions to "board repository" (see #272 and [STM8EF-MODBUS](https://github.com/TG9541/stm8ef-modbus)).

PR #277 (2019-10-30): W1209 ca v2
Fixes #275
commit c8d366b6 (2019-10-30): fixes #275 docs and editing
commit 6df71a96 (2019-10-30): fixes #275 docs and editing (#278)

* fixes #275 docs and editing

* fixes #275 docs and editing
commit 2154eceb (2019-12-13): fixes #275 docs and editing