TG9541/stm8ef #287: Can STM8S parts run on 2.8v, a flat LiPo cell voltage?

VK6TT, 2020-01-22:
I know this is general to all STM8 software, not just STM8eForth, but no reports could be found of people trying to run them below 2.95v. 

I

sbridger, 2020-01-22:
I would look at VCAP the 1.8V regulator pin, and reduce supply voltage until it drops.

Also 
VIT  Brown-out reset threshold - 2.5 2.65 2.8

Is BOR optional?

TG9541, 2020-01-22:
Hi everyone, sorry for letting you wait :-)
The STM8S core specifies a minimal supply voltage of just above 3V. There is no way to work around the low voltage detection at 2.8V - there will be a reset, no matter what you do.

There is a [nice reset related thread in the ST community forum](https://community.st.com/s/question/0D50X00009XkfXRSAZ/stm8af-output-nrst-pin-signal) that gave me a bit to think about.

sbridger, 2020-01-22:
Back to your title, a flat LIPO at modest current drains is well above 3.2V. 
2.8V is the permanent-damage-protection voltage - the cell is effectively exhausted well above this when you look at the curves.

VK6TT, 2020-01-24:
Hi everyone,

thank you for your advice on this matter. I was interested in exploring 
that region of the graph below 2.95v where "functionality is not 
guaranteed". ST did not label the voltage graph below 2.95V which is 
what piqued my interest.

SBridger's comment regarding the discharge curves put the question in 
perspective. Since my device when running, is discharging at <0.05C, the 
battery is effectively flat when the terminal voltage reaches 3.6V. I 
didn't know that at such low discharge currents the voltage falls off a 
cliff edge at 3.6v. So my quest now is not to turn everything off at 
<3.0v but at <3.6v.

I didn't know, until Thomas's email prompted me to check, that the Brown 
out reset cannot be disabled. An oversight on my part since in some 
families it can be. So the lower voltage level is somewhere between 2.5 
and 2.8V in the device I looked at, the STM8S003F3.

Again, thank you everyone. I'll close the issue, plan on not having to 
rise early to pursue this and ponder on my new challenge.

Kind regards

Richard

ps Some background to my thinking follows. It is only of academic 
interest for now but might be a useful starting point to anyone wanting 
to test the region from 2.95v down to the Brown out reset activating.

My application is a device running from a LiPo. I discarded running on a 
rechargeable coin cell early on because I didn't believe I could get the 
battery life I wanted. Now that I have optimised a few things the 
rechargeable coin cell might have been a viable power source.

A peripheral operates down to 2.7V. I wanted to be sure I could turn it 
off as the LiPo went flat because the LiPo cell I was going to use was a 
raw cell recovered from an old MAC battery pack. With no visible 
protection circuit I was worried about it being flattened and the danger 
that might pose.

My reasoning was that the core probably operates quite happily down to 
around 2.0v, since it runs at 1.8V and the inbuilt regulator has perhaps 
100mV of drop-out. But below the specified 2.95V the output pins of the 
micro could probably not source sufficient current to meet a logic high 
specification. However, I am using bit banged I2C, which sinks current 
and that could work at a slightly lower voltage than 2.95V since we are 
pulling an I/O line to ground that has a pull-up resistor in the order 
of 20-40k. Even a weak sink will allow the device to talk to the peripheral.

Here in Australia it is the middle of summer and my non-air conditioned 
workshop is unhealthy except for early mornings. I'll try to rise early 
tomorrow to test the low voltage limits of the STM8S part between 2.5 
and 2.95V. A few resistors for testing sink and source capabilities and 
some code that after a 3 second delay starts to toggle them on and off.  
The delay allows the brown out reset to be noticed while the toggling 
indicates how the source/sink for output pins suffers as the voltage drops.

closed 2020-01-24 by VK6TT

TG9541, 2020-01-24:
Hi Richard, maybe you should consider using an STM8L chip - there it's much easier to control low voltage protection feature. Just a hint: better don't use "value line" data sheets - there ST sometimes relaxes the specs to the point where contradictions arise. I've seen cases where the same chip (e.g. medium density) with "better specs" (i.e. operation range) also has the "better data sheet" (i.e. conscise description).

VK6TT, 2020-01-24:
Hi Thomas, thanks for the tip.

sbridger, 2020-01-27:
STM8L has too low an absolute maximum rating of VDD (4V) to take max possible 4.3v charging voltage of the lithium.

One trick that appears possible is to use a diode as a dropper to VDD when VBatt is too high for the 4V rating. You can then use a 5V tolerant high current pin (pA7) (5.2Vmax) as a pfet to connect VBatt (on PA7) to VDD (internally) when voltage is low enough / too low. 

Annoying these pins seem to be on lqfp48 only.
Also not driven by any pwm hardware, so the potential for a dirty buck smps vanishes.

---

Another idea is to just charge to 4.0V. You still get [75% of lipo capacity](https://www.powerstream.com/lithium-ion-charge-voltage.htm)

TG9541, 2020-01-28:
Your idea is to "vent" or "shunt" power using a 5V tolerant pin? This might work, but how is PA7 different from the FT pins available in LQFP32 or TSSOP20 packages? At least in LQFP32 some pins have timer functions.

Why not use a low power LDO? For example the TPS79733-EP consumes about 1 µA. Some charger/battery management chips with LDO even have lower quiescent current (e.g. the BQ25150).

Controlling the charge of a battery is, of course, another option.

VK6TT, 2020-01-28:
This is the path I am going down. The CN61CC33 flagging when the voltage 
drops below 3.3V so I can shut everything off and call for a charging 
cycle, and either a IMP8909J or HX61CN4002MR to terminate the charging 
cycle at 4.0V. I'd rather have a lot more cycles than a few extra joules 
in the battery.

I was surprised to learn that parts like this draw 4ua typically, orders 
of magnitude less than the resistor string I was going to use with the 
inbuilt ADC.

sbridger, 2020-01-28:
@VK6TT 

>orders of magnitude less than the resistor string I was going to use with the inbuilt ADC.

You don't do this. I am pretty sure STM8L can measure its own supply voltage. (use VDDA as ADC reference, and measure internal reference) . 

>The CN61CC33 flagging when the voltage drops below 3.3V

PVD does this at 3.05V for you.

>The device features an embedded programmable voltage detector (PVD) that monitors the VDD/VDDA power supply and compares it to ...  7 different levels between 1.85 V and 3.05 V ...  An interrupt can be generated ...

sbridger, 2020-01-28:
@TG9541 
> but how is PA7 different from the FT pins available in LQFP32 or TSSOP20 packages?

In the stm8L151 datasheet PA7 and PE0 are the only five volt tolerant pins I see, except for I2C pins. Does another STM8L have them in small packages?

>Why not use a low power LDO?

A better question is why use it when you can just use a diode, transistor or fet ?  Really I just like dirty deeds done dirt cheap.
I have done a few things using PICs where the quite grunty ports let you make self powering boost convertors with nothing extra than an inductor. 

Richard doesn't say what he is doing, but if this is a very low average drain, outside application, then I use a software boost convertor from a [small ($0.43) 1V, 80mA solar cell](https://www.aliexpress.com/item/33047290959.html) to extend battery life at negligible cost. On PIC this is a single inductor (fet and diode are internal). On STM8L it might be the high current pin PA0+schottky.

TG9541, 2020-01-29:
@sbridger I agree, I also like "sufficient" solutions, especially if they're reliable or have that little "systems quirk", like replacing hardware by a smart software solution :-)

You're right about the limitations of the STM8L151 regarding FT pins - I checked the STM8L152 datasheet which has more options. Maybe you should look for a STM8L052 device.

sbridger, 2020-01-30:
L052 are pin compatible with L151, so same i/o pins