TG9541/stm8ef #296: 4 digit 7-Segment STM8S103F3 (with serial and input)

andrewsclapp, 2020-04-10:
Hello.  I do not know if this is the right approach, I've tried several and this is the closest yet.  I'm still learning some of this stuff.  I am driving a 4-dig 7-seg bar (common anode), and I have a single analog input on pin 20, for brightness or data.  You can see my pinout in the changes below.

I started with the W1209-CA-V2 because it has the 7-segment code, and cloned it to a new directory, modified the files, added that as a build the the Makefile and created an output target to flash the board.

When I run this forth code, I get only the middle two digits (driven by PA1 and PA2), and they are missing the C (PD2) and G (PC7) segments.  I've tried so many things trying to figure out why this is the way it is and I'm finally just stuck.

nvm ok
: timer TIM 60 / . ; ok
: startup [ ' timer ] LITERAL BG ! HI ; ok
' startup 'BOOT ! ok
ram ok
cold

Since I am using the STM8S103F3, I copied the target.inc from MINDEV.

I made changes to boardcore.inc for the segments:

        ; A=PD1, B=PC6, C=PD2, D=PC5, E=PC4, F=PB4, G=PC7, Dec=PC3
        MOV     PA_DDR,#0b00000111 ; A1 A2 A3
        MOV     PA_CR1,#0b00000111
        MOV     PB_DDR,#0b00011000 ; B4 B5
        MOV     PB_CR1,#0b00011000
        MOV     PC_DDR,#0b01111100 ; C3 C4 C5 C6 C7
        MOV     PC_CR1,#0b01111100
        MOV     PD_DDR,#0b00001011 ; D1 D2 D3(input) D4
        MOV     PD_CR1,#0b00001111 ; (5,6 reserved for serial)

        RRC     A               ; RRC - rotate right logical thru carry
        BCCM    PD_ODR,#1       ; A - BCCM - copy carry bit to memory
        RRC     A
        BCCM    PC_ODR,#6       ; B
        RRC     A
        BCCM    PD_ODR,#2       ; C
        RRC     A
        BCCM    PC_ODR,#5       ; D
        RRC     A
        BCCM    PC_ODR,#4       ; E
        RRC     A
        BCCM    PB_ODR,#4       ; F
        RRC     A
        BCCM    PC_ODR,#7       ; G
        RRC     A
        BCCM    PC_ODR,#3       ; P

And for the digits (mine were reverse order from the existing ones). 

        ;BRES    PA_ODR,#4       ; Digit 4...  ???
        BRES    PA_ODR,#3       ; Digit .3..
        BRES    PA_ODR,#2       ; Digit ..2.
        BRES    PA_ODR,#1       ; Digit ...1

And globconf.inc are these (diff -c):

!         HAS_TXUART       = 1    ; No UART TXD, word TX!
!         HAS_RXUART       = 1    ; No UART RXD, word ?RX
!         HAS_TXSIM        = 0    ; Enable TxD via GPIO/TIM4, word TXGP!
!         HAS_RXSIM        = 0    ; Enable RxD via GPIO/TIM4, word ?RXGP
!         ;PSIM     = PORTD        ; Port for UART simulation
!         ;PNRX             = 4    ; Port GPIO# for HAS_RXDSIM
!         ;PNTX             = 5    ; Port GPIO# for HAS_TXDSIM
  
          EMIT_BG  = EMIT7S       ; 7S-LED background EMIT vector
!         ;QKEY_BG  = QKEYB        ; Board keys background QKEY vector
  
          HAS_LED7SEG      = 1    ; yes, 1*3 dig. 7-seg LED on module
  
!         ;HAS_KEYS         = 3    ; yes, 3 keys on module

TG9541, 2020-04-11:
I took the freedom to rearrange the GPIOs a bit:

* as a rule PD1 should be only an input most of the time
* port A has limited sink/source capabilities - it works better for segments

```
        ; Board I/O initialization
        ; STM8S103F3 init GPIO
        MOV     PA_DDR,#0b00001110 ; G,D,E
        MOV     PA_CR1,#0b00001110
        MOV     PB_DDR,#0b00110000 ; A,P
        MOV     PB_CR1,#0b00110000
        MOV     PC_DDR,#0b11111000 ; d3,d2,F,C,B
        MOV     PC_CR1,#0b11111000
        MOV     PD_DDR,#0b00010100 ; d4,d1
        MOV     PD_CR1,#0b00010100
```
I was too lazy to use segment resistors but the digit high side drivers limit the current to 20mA anyway. The result looks like this:

[image]

Here is the mapping routine. Note the offset in the `LED7LAST` indexed addressing:

```
; Common Anode display:
;   Digits - active high
;   Segments - active low

LED_MPX:
        BRES    PD_ODR,#2       ; Digit 4... d1
        BRES    PC_ODR,#7       ; Digit .3.. d2
        BRES    PC_ODR,#6       ; Digit ..2. d3
        BRES    PD_ODR,#4       ; Digit ...1 d4

        LD      A,TICKCNT+1
        AND     A,#3            ; 4 digits MPX

        JRNE    0$
        BSET    PD_ODR,#2       ; digit 4...
        JRA     3$
0$:     CP      A,#1
        JRNE    1$
        BSET    PC_ODR,#7       ; digit .3..
        JRA     3$
1$:     CP      A,#2
        JRNE    2$
        BSET    PC_ODR,#6       ; digit ..2.
        JRA     3$

2$:     BSET    PD_ODR,#4       ; digit ...1
        ; fall through

3$:     CLRW    X
        LD      XL,A
        LD      A,(LED7LAST-3,X)

        CPL     A               ; invert bits for active low
        ; 7S LED display row
        ; bit 76543210 in A
        ;  PA ....GDE.
        ;  PB ..AP....
        ;  PC ..FCB...
        RRC     A               ; A
        BCCM    PB_ODR,#5       
        RRC     A               ; B
        BCCM    PC_ODR,#3      
        RRC     A               ; C
        BCCM    PC_ODR,#4     
        RRC     A               ; D
        BCCM    PA_ODR,#2  
        RRC     A               ; E
        BCCM    PA_ODR,#1   
        RRC     A               ; F
        BCCM    PC_ODR,#5    
        RRC     A               ; G
        BCCM    PA_ODR,#3  
        RRC     A               ; P
        BCCM    PB_ODR,#4 
4$:     RET
```
You can of course rearrange the GPIO mapping, I just played around a bit and this worked for me. 

The analog input at PD3 is AIN4. The "program" I typed the following:
 ```Forth
: show ( -- ) 4 ADC! ADC@ . ; 
' show BG !
```

Here is my board folder:
[fourdig.zip](https://github.com/TG9541/stm8ef/files/4464433/fourdig.zip)

I guess it would be easiest to create a GitHub repository with a Makefile like the one in STM8EF-MODBUS, so that you can pull STM8 eForth as a dependency.

Hope this helps :-)

andrewsclapp, 2020-04-11:
Well, that works!  Thank you!  I must now try to understand why it works and what I did does not.

I guess you are saying it's mostly just pin choices and current limits implied?

TG9541, 2020-04-11:
It would work with PA1..PA3 for CA but the brightness is much reduced.

I had a look at your code snippets:

> When I run this forth code, I get only the middle two digits (driven by PA1 and PA2), and they are missing the C (PD2) and G (PC7) segments. I've tried so many things trying to figure out why this is the way it is and I'm finally just stuck.

It's not obvious but there could have been an PA0, PC0, etc, so the index of your GPIO config is off-by-one. 

This would have been better:
```
    MOV     PA_DDR,#0b00001110 ; A1 A2 A3
    MOV     PA_CR1,#0b00001110
    MOV     PB_DDR,#0b00110000 ; B4 B5
    MOV     PB_CR1,#0b00110000
    MOV     PC_DDR,#0b11111000 ; C3 C4 C5 C6 C7
    MOV     PC_CR1,#0b11111000
    MOV     PD_DDR,#0b00010110 ; D1 D2 D3(input) D4
    MOV     PD_CR1,#0b00010110 ; (5,6 reserved for serial)
```
I also removed the pull-up from PD3 (it's usually not needed for Ain).

There are some details in the `globconf.inc` I made, e.g. `ZERO` is the input device, 4 x 7S-LED digits. The offset in `(LED7LAST-offset,X)` in the LED_MUX code is a bit that I first overlooked.

andrewsclapp, 2020-04-11:
Thanks again.  Now I have to figure out the next steps.  I need to statically light the decimal point on digit 2, or I guess convert 7000 into the string "700.0" when it's output.  I with my forth was better.  I also need to slow down the refresh rate as it's just jumping all over.  And I need to dig in to the modbus example you pointed me at.  Thank you so much for all your time!

TG9541, 2020-04-12:
You're welcome - the hard part is getting started.

Forth has a set of [number formatting words](http://www.jimbrooks.org/archive/programming/forth/forthPicturedNumericOutput.php). Something like `<# # 46 HOLD #S #>` will do the trick. Here is a formatting word I wrote for [signed output with 3 digit LED displays](https://github.com/TG9541/W1209/blob/master/.0).

I'm not sure what you mean with "jumping" - ADC noise? Curiously the noise I observed was surprisingly low but I think that I was lucky. You may want to have a look at my [scaling/filter/unchatter solution](https://github.com/TG9541/W1209/blob/master/measure.fs).

andrewsclapp, 2020-04-20:
Well, it took a bit of meditation, but I today now fully grok the decimal
placement.  And I just added 16x oversampling and it was easy.  I think it
is safe to say, though I am not yet a beginner forth programmer, I am now
more effective with forth than I am with C on this platform.  The
equivalent C in the Arduino IDE for the same functionality is about 150-200
lines of code.  I cannot thank you enough for the help along the path.  I
am eager to meditate upon your unchatter solution and will commence that as
time allows.

-ASC

TG9541, 2020-04-20:
The "oversampling" / "de-noise" / "piecewise-linear-function" / "smoothen" / "unchatter" chain contains some insights from decades of dealing with µCs in embedded control. Doing a statistical error analysis should be fun ;-)

closed 2020-07-26 by TG9541