TG9541/stm8ef #315: Standalone device

Rahul24-06, 2020-05-30:
Is there any reference schematic for a standalone STM8S001J3 microcontroller?

TG9541, 2020-05-30:
The schematics from [AN2752](https://www.st.com/resource/en/application_note/cd00194637-getting-started-with-stm8s-and-stm8af-microcontrollers--stmicroelectronics.pdf) below is the closest thing to a reference schematic for the STM8S001J3 that I'm aware of: 
[image]
Of course, this chip doesn't need other supplies than VSS, VDD and VCAP, so most likely the following applies: 
[image]
The chip is rather robust - the reference design from ST doesn't take the layout recommendations too literally:
[image]
Some more information is [here in the Wiki](https://github.com/TG9541/stm8ef/wiki/Breakout-Boards#homemade-stm8s001j3m3-breakout-board).

Rahul24-06, 2020-05-30:
That's interesting. When it comes to programming this controller, how is the connection made with ST-Link / JTAG?

TG9541, 2020-05-31:
It's actually very simple: make sure that [all of the GPIOs internally bonded to pin 8](https://github.com/TG9541/stm8ef/wiki/STM8-Low-Density-Devices#stm8s001j3) are in input state at least some time (and not permanently driven by an external circuit). This means a half-duplex serial connection for a console, that's normally in RX mode, fits the bill. 

The [stm8s001rs485](https://github.com/TG9541/stm8s001rs485) is an example:
[image]

Here, PD1/SWIM is J2.2, the same as RX/TX of the half-duplex RS485 interface.

Rahul24-06, 2020-05-31:
Thanks for sharing.

closed 2020-05-31 by Rahul24-06