TG9541/stm8ef #358: STM8L152R8: add experimental support for USART2 and USART3

TG9541, 2020-09-26:
I'm sorry to say that, but experimental support means that someone else will have to experiment since I don't have a board with an STM8L Medium+ or High Density device like the NUCLEO-8L152R8 (yet). 

See notes in the [STM8L151K4](https://github.com/TG9541/stm8ef/tree/master/STM8L151K4) board folder.

4yerba, 2020-09-30:
Hi Thomas,
I happen to have the nucleo-8l152R8board, just short on free time during the week. I'll try to give it a shot  and play with your code over the weekend, if time permits. I'm kind of stuck with personal matters but maybe I get lucky this weekend. No firm promises though.

TG9541, 2020-09-30:
Thanks @4yerba! 

I have a bare STM8L052R8 chip and a LQFP64 breakout board in the basement - one of those with a bad solder resist mask that makes it hard to avoid shorts. Also the Vss, Vdd are all over the place. Not the ideal chip for a quick test. Maybe I'll try soldering it anyway.

TG9541, 2020-10-08:
So, while there's code that may work for the STM8L052R8 in the repo (but only when using USART3) the code most certainly broke access to USART2 and ... the USART1 console of STM8L Medium Density devices like STM8L151K4.

TG9541, 2020-10-08:
Bug fixed and tested with @Eelkhoorn 's board.

This doesn't mean that USART2 and USART3 console options work ... that still remains to be tested and this ticket stays open.

TG9541, 2020-10-08:
I finally finished soldering (0.5mm pitch LQFP64 wasn't that bad - soldering tiny wires for Vss, Vdd was worse and the cheap Chinese double-row headers were the wurst. This is for tomorrow:

[image]

Pin|GPIO|Function
-|-|-
1|PA0|SWIM
2|PA1|NRST
3|PA2|[USART1_TX]
4|PA3|[USART1_RX]
14|PG0|USART3_RX
15|PG1|USART3_TX
22|PE3|USART2_RX
23|PE4|USART2_TX
39|PF0|[USART3_TX]
40|PF1|[USART3_RX]
58|PC3|USART1_TX
57|PC2|USART1_RX

4yerba, 2020-10-09:
Sorry, I couldn't get to my board last weekend so no progress on my side. 
Will definitely give it a go this time. 
Actually, soldering the LQFP 64 is much easier than those 0.4 mm pitch qfn packages TI or SiLabs makes with the central solder pad. Never mind. I'm glad you managed to fix the bug so the testing will go much smoother. Will report the results when done.

TG9541, 2020-10-09:
That's great, thanks! I just gave my breakout board a try - there is a short between Vdd and Vss somewhere....

TG9541, 2020-10-09:
@4yerba I added an STM8L152R8 folder - now it's `make BOARD=STM8L152R8`, a bit less hacky, I hope.

TG9541, 2020-10-10:
I've removed the shorts and now I get a console on USART1.
No luck so far with USART2 and USART3. It might be a configuration problem or due to the fact that right now Vdd2, Vdd3, Vss2 and Vss3 are NC.

TG9541, 2020-10-10:
@4yerba: I've got USART2 and USART3 working! 

I'll make configuration options for USART1 and USART3 GPIO assignment and the push the code.

TG9541, 2020-10-10:
There will be a new option for controlling alternative port assignments of USART1:

If the default USART is used:
* `ALT_USART_STM8L = 0`: USART1_TX on PC3 and USART1_RX on PC2
* `ALT_USART_STM8L = 1`: USART1_TX on PA2 and USART1_RX on PA3
* `ALT_USART_STM8L = 2`: USART1_TX on PC5 and USART1_RX on PC6

If USART3 is used (`USE_UART3 = 1`):

* `ALT_USART_STM8L = 0`: USART3_TX on PG1 and USART3_RX on PG0
* `ALT_USART_STM8L = 1`: USART3_TX on PF0 and USART3_RX on PF1

If USART2 is used (`USE_UART2 = 1`) then `ALT_USART_STM8L` has no effect.

TG9541, 2020-10-10:
Next step: take `ALT_USART_STM8L` over to STM8L051F3 (Low Density) and STM8L151K4 (Medium Density) .

closed 2020-10-10 by TG9541

TG9541, 2020-10-11:
@4yerba running a quick test would still be appreciated :-) 

I'm thinking of making this feature a bit more generic so that it can be used by other devices, like STM8L051F3. One option is turning some of the configuration into an assembler macro (no one wants to read `.ifne ... .else ... endif` 4 levels deep).

4yerba, 2020-10-11:
No, no. It is me who should be grateful, Thomas. 

The quick test turned out to be quite an experience. First, it took me close to half an hour to get to the board as it was buried deep under some other stuff. Then UART1 and UART3 didn't seem to work receiving some garbage data.Then I looked into the code and the board schematic and closely inspected the board itself. Nothing! Only then I realized the only culprit could be the cable which I made myself years ago and ...bingo ! As it turned out it was a bad joint on the cable connector. Took me two hours to prove to myself every little detail counts. In the end I'm happy it works.

TG9541, 2020-10-11:
That's good to hear! It's always easier if there is something that's known to be in good working condition - I should know that since the STM8L101 had played tricks on me for a very long time ;-)

The STM8L152R8 (or STM8L058R8) is a rather interesting chip since it's versatile and relatively cheap (from [$0.60 @ 20](https://www.aliexpress.com/item/32871063372.html) or [$0.90 @ 10](https://www.aliexpress.com/item/4000798043777.html)). I have long wanted to play with LCDs and this seems to be the right µC to do that.

4yerba, 2020-10-11:
Nice find. 
These [stm32l151c8](https://de.farnell.com/stmicroelectronics/stm8l151c8u6/mcu-8bit-stm8-16mhz-ufqfpn-48/dp/2770265?st=stm8l151c8) are also at a reasonable price (when you get enough stuff to qualify for a free shipping) but without the LCD controller. I'll be getting these when my cart is beyond that limit :). I have no experience with the aliexpress and the wait is a bit long for me.

TG9541, 2020-10-11:
I never ordered from any of these distributors - if I'd do hardware design for a living or if I'd work in a shop where several colleagues do that their shipping and handling fees would hardly be an issue.

About the LCD controller: I'm not certain that only the "L152" devices have that. Surprisingly, STM8L151 devices have Vlcd bonded out. I'm also suspicious about the amount of memory in STM8L Medium+ Density devices and the CAN controller in STM8S207/208 devices.

4yerba, 2020-10-11:
It's a hobby for me too. 
Come to think of it, I probably spend too much money on it but, on the other hand, it is not that difficult to aggregate what you need over some period of time and crossing that 50 EUR mark once or twice a year is not a cardinal sin. A medium pizza is ~10 EUR where I live and the fun lasts only a couple of minutes ...
About these suspicions of yours - for hobby use many hacks are possible to get a maximum out of this or that chip (even if stretching the limits) but for a production device there are many limitations and challenges to consider when doing so. In most cases it's a "no go".

TG9541, 2020-10-11:
I absolutely agree, a developer in a commercial setting seldom has the opportunity to max something out. Sometimes teams get pretty far (especially when a product reaches maturity at the end of a product life cycle). Often such highly optimized resist refactoring, simply because the people who created them are no longer there or because the set of problems that a solutions was designed to solve is not longer known (!). With the next product generation asset becomes legacy, and if not legacy then sedimented knowledge. Semantic without tokens, solutions without a rationale, or even a department with a budget.

commit 57a48a8e (2020-10-10): fixes #358 STM8L High Density