TG9541/stm8ef #372: Add experimental outer Interpreter for debug use-cases

TG9541, 2020-10-18:
On Page 55 of the [ROBOFORTH II tutorial](https://www.strobotics.com/manuals/tutorial7.pdf) the use of `OUTER` and `EXIT` for debugging is demonstrated. A quick test shows that this also works for STM8 eForth but `EXIT` needs to be worked on (EXIT is just a RET in an STC Forth).

TG9541, 2020-10-19:
For `EXIT` it boils down to this: in interpretation semantic one more return address needs to be removed from the R-stack than at the end of a word (in `DO` or `FOR` loops it's of course more tricky). [DPANS94 6.1.1380 EXIT](https://www.complang.tuwien.ac.at/forth/dpans-html/dpans6.htm#6.1.1380) says that in interpretation semantics it's undefined. 

Maybe it would be better to remove `EXIT` from the standard vocabulary (`REMOVE_EXIT = 1`) and load a specialized version for normal use cases (i.e. like here in [`CURRENT`](https://github.com/TG9541/stm8ef/blob/2f40855c6b501627eecbd7dc09e1f26279eb930f/lib/CURRENT#L73)) that either compiles a `RET` or tries to figure out if a "debugging interpreter" needs to be exited. Another option is to call such a word by a different name (e.g. `BYE` since `QUIT` and `LEAVE` are already taken ;-) ).

VK6TT, 2020-10-19:
Hi Thomas

you're busy as always. Fantastic that you keep working on this.

I suggest avoiding "load a specialized version for normal use cases" 
because it can be confusing. I expect I would use the normal version in 
normal circumstances, and load the special version for special cases.

So BYE works for me, as does EXIT_DBG, XDBG or even XDB.

Keep up the good work. I've not had to stray from the STM8S003 devices 
yet, but the day is getting closer where the extra memory will be required.

Kind regards

Richard

TG9541, 2020-10-19:
@VK6TT thanks for the input! I'll think about the options, especially regarding safe execution (e.g. clearing the data stack after a typo might not be such a smart idea ;-) ) 

By the way - this little Forth now feels much more at home in STM8L now - and working through the RM0031 manual doesn't hurt as much the 3rd time you do it as it does while looking for anything familiar the first time you open it. Using the [STM8L151K4](https://www.aliexpress.com/item/4001304905877.html) is actually fun, and it isn't more expensive than an [STM8S005K6T6C](https://www.aliexpress.com/item/1005001424935630.html)!

TG9541, 2020-10-23:
I guess that's what they call "lean and mean" but it can be improved later on.

```
STM8EF2.2.26.pre3 ok
#r BYE  Uploading: ./lib/BYE                                             
\ BYE - leave a debug console OUTER interpreter
#require OUTER  Uploading: ./target/OUTER
: OUTER [ $CC C, $8D0B , OVERT ok
Closing: ./target/OUTER  ok 

: BYE ( -- ) [ ok
  $9085 ,   ok
  $9085 ,   ok
  ] OK
; ok

Closing: ./lib/BYE  ok 
: test for i dup . 2 mod 0= if ."  debug " cr outer then next ."  done" ; ok
5 test 5 4 debug                                                         
bye 3 2 debug 
bye 1 0 debug 
.S 
 <sp  ok
bye done ok
```

closed 2020-10-23 by TG9541

TG9541, 2020-10-26:
Here is a better BYE:

```Forth
#require OUTER
: BYE ( -- ) [
  \ exit the interpreter on the condition that OUTER was called
  $1605 ,           \ LDW Y,(5,SP)
  $905A ,           \ DECW Y
  $905A ,           \ DECW Y
  $90FE ,           \ LDW  Y,(Y)
  $90A3 , ' OUTER , \ CPW Y,#OUTER
  $2604 ,           \ JRNE +4
  $9085 ,           \ POPW Y
  $9085 ,           \ POPW Y
  ]
;
```

commit 152df4c9 (2020-10-23): fixes #372 debug console with OUTER and BYE