TG9541/stm8ef #381: Nucleo-8s207k8 supported?

RandoSY, 2020-11-21:
[image]

Nucleo-8s207k8

Did you recently post that you've added support for this Arduino Nano footprint compatible Nucleo?

TG9541, 2020-11-21:
I haven't tried with a real STM8S207K8T6C chip but the [STM8S207RB](https://github.com/TG9541/stm8ef/tree/master/STM8S207RB) configuration and the binary in the release archive should work right away.

The serial interface should be pin30-TX and pin31-RX.

RandoSY, 2020-11-22:
Thanks, I keep you posted on progress...

closed 2020-11-22 by RandoSY

TG9541, 2020-11-22:
@RandoSY: In the [Nucleo-8S207K8 User Manual](https://www.st.com/resource/en/user_manual/dm00489875-stm8-nucleo32-board-mb1442-stmicroelectronics.pdf) I found the following about the COM port:

[image]

It would be interesting to know if the virtual COM port works (under Linux or Windows).