TG9541/stm8ef #382: IF .. ELSE .. THEN with relative addressing

TG9541, 2020-11-25:
Some applications, e.g. ISR handlers, Flash page write, or STM8L Low-Power-Run Mode require code that doesn't depend on the STM8 eForth core. While coding in assembler or C is possible, it would be great if Forth structures like `IF .. ELSE .. THEN` could be used with `JRxx` or `BTJx`.

TG9541, 2020-11-25:
An example is the improved [I2C driver](https://gist.github.com/TG9541/5c3405320794d91ef8129734a4bfc880#improving-the-solution).

VK6TT, 2020-11-27:
Hi Thomas

I wish I had a bit more time to look at this. While my own bit bang 
routines have proved reliable that might be because I haven't pushed the 
clock speed. When I get the need I'll be sure to dig through this but as 
always you are well ahead of me. Great job and all the best for the 
coming festive season!

Regards

Richard

TG9541, 2020-11-27:
Hi Richard, thanks, and greetings your way!

To tell the truth this is the first time I actually use an I2C peripheral. Bit-banging I2C is the way of gravity because since, to work with a device only needs to read one datasheet, not two. I'm now moving from an 8K EEPROM to a DS1621 to learn how the I2C API should look like. Obviously there needs to be error handling and a handshake with the application.

TG9541, 2020-12-02:
I used the code above as-is. #383 contains some examples for words one can use instead of `IF` (e.g. `]A<IF`). Other condition words can be made from any `JRxx` or `BTJx`.

Examples are in the improved I2C code [here](https://gist.github.com/TG9541/5c3405320794d91ef8129734a4bfc880#reusable-isr-code-for-i2c).

closed 2020-12-02 by TG9541